solectrus/solectrus #3031: Aktualisierung auch bei inaktivem Fenster ermöglichen

gereons, 2024-04-21:
Ich würde gerne SOLECTRUS in einem separaten Fenster laufen lassen und darin immer aktuelle Werte sehen. Leider funktioniert das im Moment nicht, da die Aktualisierungen angehalten werden sobald das Fenster inaktiv wird - für den Normalbetrieb, wenn man in einem Browsertab oder auf dem Handy nur mal eben schnell draufschaut ist das auch völlig OK.

Wäre es möglich irgendwie von aussen zu sagen, dass man ein ständig aktuelles Display haben will, egal ob das Fenster gerade Focus hat oder nicht? Sowas wie `https://solectrus.meine-domain.de?ignoreBlur=true` oder so?

ledermann, 2024-04-21:
Derzeitig ist es so, dass die laufende Aktualisierung in zwei Fällen gestoppt wird:

- [Window: blur event](https://developer.mozilla.org/en-US/docs/Web/API/Window/blur_event)  (Fenster verliert den Fokus)
- [Document: visibilitychange event](https://developer.mozilla.org/en-US/docs/Web/API/Document/visibilitychange_event) (Website wird unsichtbar, z.B. durch Tab-Wechsel oder Minimierung des Fensters etc.)

Dieses Verhalten gibt es schon ziemlich lange in SOLECTRUS, seit Version [0.7.4](https://github.com/solectrus/solectrus/releases/tag/v0.7.4). Ich frage mich gerade, ob ein solcher Sleep-Mode wirklich noch sinnvoll ist.

Eigentlich spricht ja nichts dagegen, die Aktualisierung auch bei inaktiven Fenstern weiterlaufen zu lassen. Nur wenn es wirklich unsichtbar ist, macht ein Weiterlaufen leinen Sinn ([Zitat](https://developer.mozilla.org/en-US/docs/Web/API/Document/visibilitychange_event#usage_notes): "The transition to hidden is also a good point at which pages can stop making UI updates and stop any tasks that the user doesn't want to have running in the background.")

Ich werde mal sehen, ob ich Handling von Blur/Focus nicht ganz herausnehmen kann, dann braucht es auch keinen zusätzlichen Parameter.

Das Refresh-Intervall ist übrigens nicht fix 5 Sekunden, sondern es gilt:

- Auf der Jetzt-Seite erfolgt die Aktualisierung gemäß dem `INFLUX_INTERVAL`, das in der `.env` eingestellt, standardmäßig also 5 Sekunden. Das kann aber auch anders eingestellt werden, was z.B. nützlich ist, wenn die Werte gar nicht so schnell hereinkommen.
- Für alle **aktuellen** Zeiträume (aktueller Tag, aktuelle Woche, ...) erfolgt eine Aktualisierung alle 5 Minuten
- In allen **vergangenen** Zeiträume (gestriger Tag, letzte Woche, ...) erfolgt gar keine Aktualisierung

Ich denke, das Intervall muss nicht weiter anpassbar gemacht werden.

ledermann, 2024-04-21:
Ich habe das jetzt geändert, d.h. die Aktualisierung läuft weiter, auch wenn das Fenster keinen Fokus mehr hat. Verfügbar im `develop`-Branch (und übernommen auch nach `separate_consumer` und `next`). Passt das für Dich, @gereons?

gereons, 2024-04-21:
Top! Danke für die schnelle Änderung!

gereons, 2024-04-21:
Ich fürchte bei dieser Änderung ist irgendwas kaputtgegangen, oder zumindestens fällt es mir erst jetzt auf: bei zwei Ansichten wird links Herkunft/Verwendung brav aktualisiert, aber rechts der Chart nicht weitergezeichnet. Diese Ansichten sind aktuell bei mir "Netz" und "Akku". Ich versuche gleich mal ein Screen-Video aufzuzeichnen. (Stand `next-248-g2901296a`)

gereons, 2024-04-21:
Siehe [Video1](https://www.dropbox.com/scl/fi/7codfmnt21oynpntb3lq6/Bildschirmaufnahme-2024-04-21-um-19.39.12.mov?rlkey=oauhea6kar01iabigd08x1vet&st=p9l1kv5n&dl=0) und  [Video2](https://www.dropbox.com/scl/fi/tstlb8v65qibkee2mas6q/Bildschirmaufnahme-2024-04-21-um-19.40.16.mov?rlkey=sllq9eq74rli6crn794mdsm2u&st=s8m4nyme&dl=0)

ledermann, 2024-04-21:
Du hast recht, bei "Netz" und "Akku" stimmt etwas nicht, die Charts werden nicht fortgeführt. Es sieht mir aber eher nicht so aus, als läge das an der heute vorgenommen Blur/Focus-Änderung, sondern trat schon vorher auf - und ist nur nicht aufgefallen. Das sind übrigens die einzigen beiden Charts, bei denen es **zwei** Kurven gibt. Gerade getestet: Ja, es tritt auch mit der aktuellen `0.14.4` auf. Ich kümmere mich darum, danke für den Hinweis!

closed 2024-04-21 by ledermann