solectrus/solectrus #3042: Anpassung Fullscreen an Hochformat Screens (nicht mobile Geräte)

StevieC121176, 2024-04-23:
Wäre es möglich den Fulscreen Modus von Solectrus auch an Hochformat Screens anzupassen?
Als Beispiel kann ich hier EVCC anführen, wo das gerade umgesetzt wurde und sehr gut funktioniert.
sh hier: https://github.com/evcc-io/evcc/pull/13508

hier mal ein Beispielbild Vor und Nachher

vorher:
[image]

nachher:
[image]

ledermann, 2024-04-23:
Du beziehst dich auf das Phänomen, was du [in #2511 beschrieben](https://github.com/solectrus/solectrus/issues/2511#issuecomment-2052028943) hattest, oder?

> die normale Kachelansicht wird korrekt dargestellt sh. Foto. Ich kann auch den vollbildmodus in SOLECTRUS aktivieren, nur da wird leider in das Querformat gewechselt und somit bleibt die untere Bildschirmhälfte leer. Das Display im Kühlschrank ist senkrecht.

> evtl. liegt es an der Auflösung. normal im Hochformat werden 982x1514 Pixel genutzt, beim Fullscreen dann 1080x1920px

Ich hatte das ursprünglich nicht verstanden und vermutet, dass es ein Problem mit dem Fullscreen-Mode bzw. dem Ausblenden des Browser UI ist. Das ist es aber gar nicht, es ist eine reine Frage der Breakpoints für das responsive Layout.

Es sieht bei dir also offenbar so aus:

| 986x1514 (OK)     | 1080x1920  (falsch) |
| ------------- | ------------- |
| [image] | [image]  |

Danke für den weiteren Hinweis, das lässt sich beheben, ich kümmere mich darum.

ledermann, 2024-04-23:
@StevieC121176: Ich habe das nun korrigiert, d.h. die Darstellung auf großen und gleichzeitig hohen Displays (also wie bei deinem Kühlschrank) erfolgt jetzt etwas anders. Die Änderung ist in den Branches `develop`, `separate_consumer` und `next`. Bin gespannt, ob es bei dir nun passt.

StevieC121176, 2024-04-23:
Sobald ich heute Abend wieder zu Hause bin Probier ich es gleich aus. Vielen Dank

StevieC121176, 2024-04-23:
@ledermann funktioniert perfekt, danke 

[image]

closed 2024-04-24 by ledermann