solectrus/solectrus #3051: Wärmepumpe mit MQTT konfigurieren

ledermann, 2024-04-27:
ich versuche gerade auch meine Wärmepumpe über MQTT in Solectrus zu bekommen.

den mqtt collector habe ich in docker-compose.yml eingefügt (Datei ist unten angefügt)
Dabei musste ich allerdings `service_unhealty` auskommentieren. 
Außerdem musste ich noch `env_file: .env` hinzufügen. ist das richtig so?

In der .env Datei habe ich die credentials gesetzt und nur das Topic angegeben.

Im MQTT Log erscheint jedoch dauernd:

`Error while pushing to InfluxDB: unable to parse 'heatpump_power=3i 1714204074': invalid field format`

Hat jemand eine Idee was nicht läuft?
Macht es Sinn den Konfigurator zu nutzen und die .env Datei sowie die docker-compose.yml komplett neu aufzubauen?

Grüße Michael

```
mqtt-collector:
    image: ghcr.io/solectrus/mqtt-collector:develop
    labels:
      - 'com.centurylinklabs.watchtower.scope=solectrus'
    depends_on:
      - influxdb
       # condition: service_healthy
    links:
      - influxdb
    environment:
      - INFLUX_HOST
      - INFLUX_SCHEMA
      - INFLUX_PORT
      - INFLUX_TOKEN=${INFLUX_TOKEN_WRITE}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_MEASUREMENT=${INFLUX_MEASUREMENT_PV}
      - MQTT_HOST
      - MQTT_PORT
      - MQTT_SSL
      - MQTT_USERNAME
      - MQTT_PASSWORD
#      - MQTT_TOPIC_HOUSE_POW
#      - MQTT_TOPIC_GRID_POW
#      - MQTT_TOPIC_BAT_CHARGE_CURRENT
#      - MQTT_TOPIC_BAT_FUEL_CHARGE
#      - MQTT_TOPIC_BAT_POWER
#      - MQTT_TOPIC_BAT_VOLTAGE
#      - MQTT_TOPIC_CASE_TEMP
#      - MQTT_TOPIC_CURRENT_STATE
#      - MQTT_TOPIC_MPP1_POWER
#      - MQTT_TOPIC_MPP2_POWER
#      - MQTT_TOPIC_MPP3_POWER
#      - MQTT_TOPIC_INVERTER_POWER
     # - MQTT_TOPIC_WALLBOX_CHARGE_POWER
#      - MQTT_FLIP_BAT_POWER
      - MQTT_HEATPUMP_POWER
    env_file: .env
    restart: always
    logging:
      options:
        max-size: '10m'
        max-file: '3'
```

```
# MQTT Collector

# MQTT Broker credentials
MQTT_HOST=192.168.178.53
MQTT_PORT=1883
MQTT_SSL=false
MQTT_USERNAME=test
MQTT_PASSWORD=test

# MQTT topics
# https://docs.evcc.io/docs/reference/api#mqtt-api
#MQTT_TOPIC_HOUSE_POW=evcc/site/homePower
#MQTT_TOPIC_GRID_POW=evcc/site/gridPower
#MQTT_TOPIC_BAT_FUEL_CHARGE=evcc/site/batterySoc
#MQTT_TOPIC_BAT_POWER=evcc/site/batteryPower
#MQTT_TOPIC_INVERTER_POWER=evcc/site/pvPower
#MQTT_TOPIC_WALLBOX_CHARGE_POWER=evcc/loadpoints/1/chargePower
MQTT_TOPIC_HEATPUMP_POWER=controller/ccp
# MQTT options
#MQTT_FLIP_BAT_POWER=true
```

_Originally posted by @halmich in https://github.com/solectrus/solectrus/discussions/2861#discussioncomment-9243848_

halmich, 2024-04-27:
Noch eine Frage:
Ich habe den Konfigurator benutzt und angegeben, dass ich einen v2.1 und zusätzlich die Wärmepumpe über MQTT auslese.
In der .env und in der .yml tauchen jedoch keine Parameter / Credentials zu MQTT auf.
Ist das ein Bug?

ledermann, 2024-04-27:
Hier gibt es einen ganzen Schwung verschiedener Probleme, die sich aber alle lösen lassen :-)

1) Wenn ein Container `unhealthy` ist, dann bitte unbedingt ins Log dieses Containers gucken, was da falsch läuft. Das geht z.B. mit `docker compose logs influxdb`. Je nachdem, was dort steht, muss dieses Problem behoben werden. Es ist **nicht** empfehlenswert, das `condition: service_healthy` auszubauen.

2) Wenn einer der Container redis, postgresql oder influxdb ungesund ist, ist das ein schlechtes Zeichen, das hat dann auch nichts mit SOLECTRUS zu tun. Vor allen anderen Schritte sollte das geklärt werden. Die drei Datenbank-Container müssen auf fehlerfrei laufen.

3) Zu PostgreSQL: Neuer Major-Versionen können nicht die Daten der Vorgängerversion lesen, die müssen migriert werden. Version 16 kann also nicht die Daten der Version 15 lesen. Da das Migrieren etwas fummelig ist, lass das besser erst mal weg. Wenn du also bislang `postgres:15-alpine` verwendet hast, dann darf nicht einfach auf `postgres:16-alpine` gewechselt werden.

4) An die Logs des MQTT-Collectors kommst du analog: mit `docker compose logs mqtt-collector`

5) Den Konfigurator lasse besser auch erst mal weg, der ist noch unfertig. Den beschriebenen Fehler, dass der MQTT-Collector nicht in .env und compose.yml auftaucht, kann ich zwar nicht reproduzieren, meinen Tests zufolge klappt das, aber etwaige Fehler im Konfigurator behandeln wir besser separat.

6) Der Eintrag `env_file: .env` sollte nicht erforderlich sein. Was passiert, wenn du das weglässt?

7) Die Fehlermeldung `heatpump_power=3i 1714204074': invalid field format` resultiert vermutlich aus einem Datentyp-Konflikt. Integer-Werte können nicht in ein Float-Field geschrieben werden.

8) Etwaige Fehler im Konfigurator behandeln wir besser separat, sonst ufert das hier wirklich aus :)

Um der Sache genauer auf den Grund zu gehen: Gib mir bitte mal die beiden vollständigen Dateien `.env` und `compose.yml` - Passwörter und andere heikle Angaben vorher anonymisieren. Dann werde ich das hier nachstellen.

halmich, 2024-04-27:
auch hier noch kurze als Antwort für Mitlesende:

zu 1/2: ohne Auskommentierung erscheint beim Startes der Container: `Dependency failed do start: container solectrus-influxdb-1 has no healtcheck configured`
Aus meiner Sicht ist die Datenbank aber intakt.

3/4: alles klar, bin wieder auf 15.

5: Zum Nachstellen. Konfigurator aufrufen.
Haken bei Solarwechselrichter, Speicher, Wärmepumpe
Stromspeicher: v3/2.1
Wärmepumpe über MQTT
Lokal
RaspberyPi
ip: egal
Ip: egal, alle Werte lassen
Ip egal, port 1883, benutzername: meinBenutzer: Passwort: MeinPasswort

MQTT Topics:
bei Wärmepumpe: controller/ccp  , Verbrauch ist im Hausverbrauch enthalten. ansonsrten alles leer.
Prognose: nein
AWS: nein
Inbetriebnahme und PW: egal

Abschließen.
In beiden Dateien ist kein Hinweis zu MQTT.

Zu 6: wenn der eintrag nicht gesetzt ist, fragt MQTT keine Werte ab.

Zu 7: ok, da muss ich schauen, ob ich das Lösen kann.

compose.yml
```
services:
  dashboard:
    image: ghcr.io/solectrus/solectrus:next
    depends_on:
      postgresql:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - postgresql
      - influxdb
      - redis
    ports:
      - '3000:3000'
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - WEB_CONCURRENCY
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - DB_HOST=postgresql
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_POLL_INTERVAL
      - INFLUX_SENSOR_INVERTER_POWER
      - INFLUX_SENSOR_HOUSE_POWER
      - INFLUX_SENSOR_GRID_IMPORT_POWER
      - INFLUX_SENSOR_GRID_EXPORT_POWER
      - INFLUX_SENSOR_BATTERY_CHARGING_POWER
[... 264 lines cut ...]
# Set these names before first run, they can't be changed later!
INFLUX_BUCKET=solectrus

# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=./influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=./postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=./redis
```

ledermann, 2024-04-28:
Zwischenzeitlich habe ich den Konfigurator korrigiert, sodass im Fall von Wallbox=`NEIN` und Wärmepumpe=`JA` korrekt eine Konfiguration für den MQTT-Collector generiert wird. 

Etwas knifflig wird es, wenn der Konfigurator benutzt wird, um eine bestehende SOLECTRUS-Instanz (also mit bereits gefüllter InfluxDB) zu modernisieren. Es ist dann u.U. erforderlich, die generierten Angaben für Bucket und Measurement anzupassen, sonst sucht SOLECTRUS die Messwerte an der falschen Stelle und zeigt rein gar nichts an.

Da die generierte Konfiguration nun funktioniert, schließe ich das Issue.

closed 2024-04-28 by ledermann