solectrus/solectrus #3154: Werte für Wärmepumpe werden falsch angezeigt

patricknitsch, 2024-06-08:
Hallo Georg,

ich habe mit dem Configurator die Anbindung der Wärmepumpe über MQTT integriert.
Die Werte werden richtig übergeben und angezeigt. Die Einbindung läuft seit Donnerstag. Heute ist mir aufgefallen, dass die Berechnung falsch ist. Wie du im Bild sehen kannst, passen die Gesamtwerte nicht zu den Einzelwerten. Auch wird mir heute kein Hausverbrauch sondern nur Verbrauch Wärmepumpe angezeigt. 

[image]

[image]

[image]

Für den gestrigen Tag passen die Werte in der Wochenansicht aber noch. 

Hast du hierzu eine Idee, woran das liegen könnte?

ledermann, 2024-06-08:
Was aus meiner Sicht falsch ist:

1) Hausverbrauch 40 kWh passt nicht zu den sechs Balken, die eher ca. 80 kWh darstellen
2) Wärmepumpe 62 kWh passt nicht zu den zwei Balken, die eher ca. 15 kWh darstellen
3) Hausverbrauch ist am Samstag gar nicht vorhanden. Wie sieht die Kurve des Hausverbrauchs vom Samstag aus?

Wird die Wärmepumpe bei dir vom Hausverbrauch abgezogen? Wie genau sieht die Konfiguration aus? Bitte mal die `.env` posten. Ich bin sehr interessiert, dem Problem auf die Spur zu kommen.

Was du prüfen solltest, ist zunächst die Herkunft der Werte. Woher kommen die Werte von Hausverbrauch und Wärmepumpe? Und wie oft wird abgefragt? Ich habe den Verdacht, dass die Werte nicht oft genug abgetastet werden. Dies ist aber wichtig. Wenn z.B. der Hausverbrauch im 5-Minuten-Takt kommt, der Wärmepumpen-Verbrauch aber alle paar Sekunden, dann geht die Differenzbildung schief. Am besten ist es, wenn beide Messreihen sehr häufig (z.B. alle 5 Sekunden) aktualisiert werden. 

Ansonsten gibt es noch ein bekanntes Problem (#2032), dass die Darstellung von **aktueller** Woche oder aktuellem Monat Abweichungen haben kann, die sich aber von selbst korrigieren, wenn der Zeitraum vollständig in die Vergangenheit gerutscht ist. Das spielt hier vielleicht mit rein, ist aber vermutlich nicht das einzige Problem.

Übrigens scheint ein Screenshot in deinem obigen Post zu fehlen.

patricknitsch, 2024-06-08:
Danke für deine schnelle Rückmeldung!

Ja den fehlenden Screenshot habe ich nochmal angehängt.

Anbei auch der Screenshot von Samstag mit dem Verlauf des Hausverbrauchs. Sollten heute ca. 22 kWh sein.

[image]

Ich habe angegeben, dass die Pumpe vom Hausverbrauch abgezogen werden soll. Die .env ist angehängt.

```
##################################################################
###                   Dashboard application                    ###
##################################################################

# Domain name or IP address of your host
APP_HOST=####

# Date of commissioning of your photovoltaic system
INSTALLATION_DATE=2022-03-16

# Password to login as administrator, required to manage settings like historical prices
ADMIN_PASSWORD=solectrus_20_12_15

# Interval in seconds for polling the InfluxDB
INFLUX_POLL_INTERVAL=5

# SSL redirect: Use "true" if you want to auto-redirect to https, but ensure that you have a valid SSL certificate
# and a reverse proxy in front of the app!
# In all other cases, the option must be "false"!
FORCE_SSL=false

# Secret token to secure cookies, 128 chars long hexadecimal encoded string
SECRET_KEY_BASE=231038b402ec1a4112bf39130e545a72fc53559d6822a10344728f267b78312826d2e2067215827b2a045a8df878a9c740e7dd1db3a4797787bf9ef05a934bb4

# Start web server Puma in single mode (recommended to reduce memory usage)
WEB_CONCURRENCY=0

# Optional: Allow iframe embedding
# FRAME_ANCESTORS=https://my-other-home-automation-software.com

# Password for the PostgreSQL database, used by the app to comunicate internally with the database
# Caution: Do not change this after the first run, otherwise the app will no longer be able to connect to the database!
POSTGRES_PASSWORD=my-secret-db-password

##################################################################
###                      Sensor mapping                        ###
##################################################################

# Where to find the data in the InfluxDB
# Each sensor must be mapped to a measurement and a field
# The format is: INFLUX_SENSOR_NAME=MEASUREMENT:FIELD
# Example: INFLUX_SENSOR_INVERTER_POWER=my_pv:inverter_power

INFLUX_SENSOR_INVERTER_POWER=SENEC:inverter_power
INFLUX_SENSOR_HOUSE_POWER=SENEC:house_power
INFLUX_SENSOR_GRID_IMPORT_POWER=SENEC:grid_power_plus
INFLUX_SENSOR_GRID_EXPORT_POWER=SENEC:grid_power_minus
INFLUX_SENSOR_BATTERY_CHARGING_POWER=SENEC:bat_power_plus
INFLUX_SENSOR_BATTERY_DISCHARGING_POWER=SENEC:bat_power_minus
INFLUX_SENSOR_BATTERY_SOC=SENEC:bat_fuel_charge
# INFLUX_SENSOR_WALLBOX_POWER=
INFLUX_SENSOR_CASE_TEMP=SENEC:case_temp
INFLUX_SENSOR_SYSTEM_STATUS=SENEC:current_state
INFLUX_SENSOR_SYSTEM_STATUS_OK=SENEC:current_state_ok
INFLUX_SENSOR_GRID_EXPORT_LIMIT=SENEC:power_ratio
INFLUX_SENSOR_INVERTER_POWER_FORECAST=forecast:watt
INFLUX_SENSOR_HEATPUMP_POWER=heatpump:power

# Optional: Exclude sensors from house power calculation
[... 168 lines cut ...]
# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=./influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=./postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=./redis

```

Der Hausverbrauch wird ja über die Senec Api alle 5 Sekunden aktualisiert. Welchen Intervall hat der MQTT Collector? Kann man diesen ändern?

ledermann, 2024-06-08:
Ok, danke. Prüfe mal (im ioBroker), wie oft der Wert von `iobroker/heisswasser/value` aktualisiert wird. Wenn du das Abfrageintervall reduzieren kannst (also öfter abfragen), dann versuch das mal. Liegen die Werte dort wirklich in Watt vor? Mir erscheinen die Zahlen zu groß.

Hast du irgendwo Referenzwerte zu deiner Wärmepumpe, z.B. in der App des Herstellers? Kannst du beurteilen, ob die von SOLECTRUS angezeigten Werte an irgendeiner Stelle stimmen können?

Ansonsten ist mir ein solches Phänomen noch nicht begegnet. Ich würde mir deine Influx-Daten gerne (morgen) genauer ansehen. Falls du die Möglichkeit hast, eine Sicherung zu erstellen (notfalls Zip-Archiv deines Influx-Ordners, vorher aber die Container herunterfahren mit `docker compose down`), dann schicke mir diese gerne per E-Mail.

Habe wg. Datenschutz deine `.env` etwas "geschwärzt".

patricknitsch, 2024-06-08:
Ich habe gerade mal nachgeschaut. Er hat die Werte nur gesendet, wenn diese geändert wurden. Ich habe dies mal geändert auf Senden bei Aktualisierung.

Ja die Werte passen. Es ist keine Wärmepumpe in diesem Sinne, aber ein Heisswasserspeicher mit einer Spitzenleistung von 4500 W.

Die Sicherung erstelle ich und schicke sie dir per Mail zu.

patricknitsch, 2024-06-08:
Was mir gerade aufgefallen ist. Nachdem die Uhrzeit 00:00 passiert hat, haben sich die Werte geändert.

[image]

So passen die Werte auch in Summe wieder

ledermann, 2024-06-09:
Ok, danke für die Daten. Ich bin hier nun dran, eine Verbesserung bzw. Korrektur zu finden. Die Problematik: Bei der Formulierung von InfluxDB-Abfragen gibt es einen Trade-off zwischen Performance und Genauigkeit.

Derzeitig mache ich es so, dass bei Abfragen, die einen aktuellen (d.h. unvollständigen) Zeitraum umfassen (z.B. aktueller Tag, aktuelle Woche etc.) die **Integral**-Funktion genutzt wird. Langsam, aber genauer. Bei vollständig in der Vergangenheit liegenden Abfragen (z.B. gestern, letzte Woche etc.) wird hingegen nicht das Integral berechnet, sondern die **Summe der Stunden-Mittelwerte**. Das ist der Grund, warum nach Mitternacht die Darstellung von Samstag korrekt wurde.

In deinem Fall fehlen beim Heißwasserspeicher viele Messwerte, wegen der Einstellung "Publish only on change" im ioBroker. Daran scheitert die Integral-Funktion. Wenn du diese Einstellung deaktivierst, wird dies vermutlich zu einer deutlichen Verbesserung führen. Bitte beobachte das mal in der nächsten Zeit.

patricknitsch, 2024-06-09:
Verstanden. Ich habe es gestern so abgeändert, dass alle 5 Sekunden ein Werte gesendet wird. Ich werde es beobachten.

Danke für deine Unterstützung und die super Arbeit!

ledermann, 2024-07-24:
Da ich davon ausgehe, dass das Problem mittlerweile nicht mehr auftritt, schließe ich dieses Issue.

Eine Lösung der Art, dass fehlende Werte automatisch als 0 behandelt werden, ist leider nicht in Sicht. InfluxDB bietet das zwar grundsätzlich an ([fill() function](https://docs.influxdata.com/flux/v0/stdlib/universe/fill/#value)), das geht aber auf Kosten der Performance. Ich verfolge das daher erst mal nicht weiter.

closed 2024-07-24 by ledermann

patricknitsch, 2024-07-24:
Jup funktioniert wunderbar. Danke!