solectrus/solectrus #3155: Rundungen ab 100 kWh

StevieC121176, 2024-06-09:
Wäre es möglich, zumindest für die Tagesansicht auf eine Stelle nach dem Komma zu runden? Aktuell wird ab 100 kwh auf ganze kwh gerundet. Ich hatte gestern 102,5 kWh Erzeugung, angezeigt wurden aber 103 kWh

[image]
[image]

AlpenFlizzer, 2024-06-09:
Ich weiß nicht ob es bereits eine Lösung in deinen Augen wäre, aber wenn du lange mit dem Finger auf den Wert der Erzeugung tippst, kommt in einem Tooltipp der exakte Wert mit drei Nachkommastellen.

StevieC121176, 2024-06-09:
nein leider nicht, ab 100 kWh wird auch im Tooltip nur 103 kWh angezeigt

AlpenFlizzer, 2024-06-09:
Georg wird mich korrigieren, wenn es falsch ist bitte, aber in der `number` Komponente werden die Zahlen gestyled. Dort findet sich folgendes Snippet:
```
  def formatted_number(value, max_precision:)
    return unless value

    # Some numbers don't need fractional digits
    need_fractional_digits =
      value.round(max_precision).nonzero? && value.abs < 100

    number_with_precision(
      value,
      precision: need_fractional_digits ? max_precision : 0,
      delimiter: I18n.t('number.format.delimiter'),
      separator: I18n.t('number.format.separator'),
    )
  end
```
Im Grunde wird hier festgelegt, dass Werte über 100 keine Nachkommastelle bekommen. Die Frage stellt sich jetzt, ob man entweder kWh Werte > 100 in MWh Werte Umwandelt und sie so wieder kleiner 100 bekommt, oder ob man defaultmäßig nicht auf wenigstens 1 Nachkommastelle rundet.

ledermann, 2024-06-09:
Grundsätzlich verfolge ich mit SOLECTRUS die Idee, unnötige Details möglichst zu verbergen. Das führt bei Zahlen zur Idee, Nachkommastellen nur bei kleinen Zahlen anzuzeigen und andernfalls zu runden. Nur was ist eine "kleine" Zahl? Wie @AlpenFlizzer richtig gefunden hat, ziehe ich diese Grenze bei 100. Nun sollte man auf Wunsch aber dennoch Nachkommastellen sehen können, nämlich per Tooltip. Dummerweise funktioniert das nicht gut, denn große Zahlen werden, wie @StevieC121176 richtig feststellt, auch im Tooltip ohne Nachkommastellen dargestellt. Das sehe ich als Fehler an. Bei Zahlen kleiner 100 funktioniert es hingegen (statt einer sieht man dann drei Nachkommastellen). Bei 1000 ist übrigens die nächste Schwelle, da werden aus kW dann MW.

So ganz leicht ist das nicht zu korrigieren. Zusätzliche Ausnahmen möchte ich eher nicht hinzufügen. Die [Number::Component](https://github.com/solectrus/solectrus/blob/develop/app/components/number/component.rb) könnte nämlich sowieso ein grundlegendes Refactoring vertragen, die ist historisch gewachsen, und zugegebenermaßen etwas krumm und schief geworden. Besser wäre es, die Komponente würde Zahl und Einheit erhalten und dann die formatierte Ausgabe über **eine** Methode erledigen (`to_html`), die über Parameter mitgeteilt bekommt, was man haben möchte (Anzahl Nachkommastellen, Einfärbung, Mega/Kilo etc). Dann könnte man die gleiche Zahl je nach Bedarf mal so oder so darstellen. Das hat aber weitreichende Auswirkungen, da in der gesamten App alles angepasst werden müsste, was Zahlen ausgibt.

Ich lasse das mal offen und kümmere mich später darum. So dringend ist es ja nicht, denke ich.

AlpenFlizzer, 2024-06-16:
Ich habe nochmal darüber nachgedacht.
Abgesehen von deinen Refactor Plänen fürs numbering wollte ich vorschlagen eventuell die mindestens angezeigten Nachkommastellen zumindest konfigurierbar zu machen.

Meine Gedanken dazu: ich sehe Solectrus als eine (einzige) Möglichkeit den limitierten und oft Cloud gebundenen Hersteller Apps zu entkommen. Dass du es simpel und übersichtlich halten willst verstehe ich. Jedoch sollten abweichende Bedürfnisse, wenn sie mit überschaubaren Aufwand umgesetzt werden können, Berücksichtigung finden. 

Mittels env könnte man das Nachkommaverhalten situativ oder global vermutlich relativ einfach überschreiben.

Was meinst du?

ledermann, 2024-06-18:
Grundsätzlich stimme ich dir zu, @AlpenFlizzer, jede Maßnahme zu ergreifen, sodass SOLECTRUS gegenüber den Hersteller-Apps noch besser wird. 

Aber ich denke, hier kommen wir ohne eine zusätzliche Konfiguration aus. Denn wir haben eine Inkonsistenz, die als Fehler anzusehen ist: Im Tooltip wird genauso gerundet wie im Bilanz-Segment selbst. Das macht keinen Sinn und daher habe ich das jetzt korrigiert (in `develop`).

Jetzt werden auch bei Werten über 100 im Tooltip die Nachkommastellen angezeigt:

[image]

Ich denke, das sollte das Problem lösen, oder?

AlpenFlizzer, 2024-06-18:
Ich denke schon. Wahrscheinlich muss man wirklich etwas konservativer sein, wenn es um individuelle Konfigurierbarkeit geht, da es sonst schnell mal ausartet. Ich bin da immer etwas euphorisch, weil es bei Solectrus zumindest theoretisch überhaupt die Möglichkeit gibt, etwas zu individualisieren... :-)

ledermann, 2024-06-18:
Jetzt möchte ich zum Abschluss doch noch mal ein Zitat meines alten Physiklehrers herauskramen ;-)

> "Suggeriere keine hohe Genauigkeit durch viele Nachkommastellen, wenn du gar nicht so präzise messen kannst!" 

Wie genau die Werte sind, die vom Wechselrichter gemeldet werden, wissen wir nicht. Da ist eher **kein Hochpräzisions-Messgerät** verbaut. Hinzu kommt: Die Messwerte, die SOLECTRUS verarbeitet, sind Momentaufnahmen, die bei den meisten Installationen im 5-Sekunden-Takt abgegriffen werden. Was zwischen zwei Abtastungen passiert, wird komplett **ignoriert**. In der Gesamtsicht passt dennoch alles ganz gut, weil es üblicherweise keine erheblichen Schwankungen gibt. Aber eine Watt-genaue Angabe halte ich für eigentlich nicht gerechtfertigt.

Ganz allgemein erhöht eine Konfigurierbarkeit den Entwicklungsaufwand und verkompliziert tendenziell das gesamte Tool. Dinge, die eher den persönlichen Geschmack betreffen (Anzahl der Nachkommastellen, Schriftart, Icons, Farbverlauf im Graphen usw.), die möchte ich sinnvoll festlegen (und gerne auch diskutieren), aber keine Veränderbarkeit ermöglichen.

Ich hoffe hier auf allseitiges Verständnis :-)

closed 2024-06-18 by ledermann