solectrus/solectrus #3156: Feature Request: E-Auto Laden und Autarkie - Anteil von PV- und Netzenergie

halmich, 2024-06-09:
Hallo Georg,
folgendes Feature ist bei Zappi implementiert und für E-Autofahrer sicherlich interessant.
Es ist zwar "nur eine Spielerei", aber dafür sind wir ja da :)

Es wäre schön zu wissen, welcher Anteil der Energiemenge, die fürs E-Autoladen benutzt wurde, aus der Photovoltaik Anlage und welcher Anteil aus dem Netz kommt. Ähnlich der Autarkie fürs gesamte Haus.

Umsetzung könnte wie folgt gehen:
In einer Variablen "EV_Grid" wird die Energie gespeichert, die das Auto aus dem Netz benötigt. 
Berechnung:
Falls Strom eingespeist wird
       EV_Grid = 0.
Falls Strom bezogen wird
       Falls Leistung Auto > Netzbezug
              EV_Grid = Integral Netzbezug
              # Ohne E-Eauto wäre Netzbezug = 0, daher mache ich das E-Auto für den Netzbezug "verantworlich"
       Falls Leistung Auto < Netzbezug
              EV_GRid = Integral Wallbox_Power
              # z.B. im Winter ist der Netzbezug sicherlich höher als Wallbox_Power.

Man berechnet nun:
- Man berechnet 1- (EV_Grid / Wallbox_Power) für den "grünen" Anteil, also die Energie direkt aus Sonnenenergie.

In einem weiteren Schritt wäre u.a. denkbar, damit sowohl einen Ersparnis gegenüber dem Benzin/Dieselpreis auszurechnen, CO2 Ersparnis, etc.
Aber das wären ebenfalls "Spielereien" :)

Grüße Michael

halmich, 2024-06-09:
Ich sehe gerade, dass du dieses Issue gernen verschieben / löschen kannst. 
In https://github.com/solectrus/solectrus/issues/2407 und https://github.com/solectrus/solectrus/issues/2436 wird beides schon angesprochen.

ledermann, 2024-06-09:
Genau, und zumindest die Deckung je Verbraucher (#2436) ist auch bereits in Arbeit :)

closed 2024-06-09 by ledermann