solectrus/solectrus #3157: CO2-Emissionsfaktor solle konfigurierbar sein

AlpenFlizzer, 2024-06-09:
In Österreich werden dreierlei Emissionsfaktoren auf der Homepage des Umweltbundesamtes für Strom angegeben.
Siehe: https://secure.umweltbundesamt.at/co2mon/co2mon.html

Je nachdem, welcher Strommix auf den eigenen Haushalt zutrifft, ist ein anderer Faktor anwendbar:

**Stromaufbringung   Österreich: 0,23kg/kWh**
_Bei   der  Stromaufbringung Österreich werden   die inländische Stromerzeugung und die Stromimporte berücksichtigt. Der   Emissionsfaktor für die inländische Stromerzeugung wird anhand des   Kraftwerkparks ermittelt. Für die Stromimporte werden die Emissionsfaktoren   der Stromerzeugung der Importländer herangezogen._

**Kraftwerkspark Österreich: 0,17kg/kWh**
_Es   wird der österreichische Kraftwerkspark d.h. sämtliche Stromerzeugungsanlagen   auf österreichischen Staatsgebiet berücksichtigt. Somit wird wie bei der   österreichischen Luftschadstoffinventur das Territorialprinzip angewendet._

**Umweltzeichen "Grüner   Strom": 0,01kg/kWh**
_Das   österreichische Umweltzeichen definiert in der Version 5.0 vom Jänner 2018   die Kriterien für die Stromerzeugung. Der Strom muss zur Gänze aus   erneuerbaren Energieträgern stammen und bestimmte Kriterien erfüllen. Diese   Kriterien sind in der Richtlinie "UZ 46 - Grüner Strom" definiert.   Der Emissionsfaktor wird basierend auf dem Stromerzeugungsmix der größten   Anbieter ermittelt._

Gemäß Prinzip: andere Länder andere Sitten, ist das vermutlich auch für andere Länder als DE und AT zutreffend.

Ich schlage vor, entweder den Faktor aus dem ENV zu fetchen oder
den Emissionsfaktor analog den Strompreisen im GUI konfigurierbar zu machen.

ledermann, 2024-06-09:
Interessanter Einblick, vielen Dank! 

Eine ENV-Variable einzuführen, wäre schnell gemacht. Was meiner Ansicht nach aber nicht berücksichtigt wird, ist der zeitliche Aspekt. Die derzeitig hartkodierten 401 g/kWh stammen ja nicht nur aus Deutschland, sondern gelten auch nur für 2019. 

Um es korrekt zu machen, müsste man also den Emissionsfaktor jährlich erfassbar machen. Also vergleichbar mit den Strompreisen. Denn auch wenn ich in zwei Jahren den exakt selben PV-Strom erzeugt habe, sinkt meine CO-Einsparung, wenn der Strommix in meinem Land grüner geworden ist.

Insofern stellt sich die Frage: Wollen wir das wirklich genau berechnen (und somit die Faktoren jahresbezogen per UI komfortabel erfassbar machen) oder genügt die grobe Methode (ein Faktor per ENV-Datei)?

spirit0607, 2024-06-13:
Wie wäre es denn mit der Möglichkeit "Electricity Maps" einzubinden? 

Diese stellt eine kostenfreie API mit dem aktuellen CO2 Mix für den privaten Gebrauch zur Verfügung.

https://www.electricitymaps.com/free-tier-api

Somit hätte man sogar einen minütlich korrekten Wert und eine wirklich genau Auskunft darüber, wie hoch die CO2 Ersparnis tatsächlich ist. Ich stelle mir die Integration in Solectrus analog dem WeatherForecast vor. 

Wenn sich jemand nicht registrieren möchte, oder den Dienst nicht nutzen mag, könnte ja trotzdem per env einen fixen Wert eingeben. 

Vll. ist das ja umsetzbar.

ledermann, 2024-06-13:
Das Angebot von @electricitymaps kannte ich bislang gar nicht, sehr cool, danke für den Hinweis, @spirit0607! Ich klicke mich gerade durch https://app.electricitymaps.com/zone/DE. Den bisher von SOLECTRUS verwendeten Emissionsfaktor von 401 g/kWh für DE finde ich dort zwar nicht (auch nicht bei 2019, wo er eigentlich herkommt), aber die Größenordnung stimmt. Ähnliches gilt für die oben von @AlpenFlizzer genannten Werte für AT.

Der korrekte API-Endpunkt wäre wohl dieser hier:
https://api.electricitymap.org/v3/carbon-intensity/latest?zone=DE

Dieser Wert könnte von einem neuen (noch zu erstellendem) Collector kontinuierlich abgefragt und in InfluxDB gespeichert werden. Bei der Darstellung in SOLECTRUS könnte er dann mit der PV-Erzeugung multipliziert werden. Das würde viel mehr Sinn ergeben als der jetzige konstante Faktor. 

Und ja, Fallback auf ENV-Variable wäre nützlich. Und da es eine [Integration für Home Assistant](https://www.home-assistant.io/integrations/co2signal/) gibt, könnte SOLECTRUS den Wert prinzipiell auch dort abholen, über den MQTT-Collector.

Historische Daten wären ideal, seit Inbetriebnahme der eigenen PV-Anlage. Aber die sind [im Free-Plan leider nicht enthalten](https://www.electricitymaps.com/pricing).

Insgesamt eine spannende Sache, wie sehen das die anderen?

AlpenFlizzer, 2024-06-13:
Für meinen Use Case und Anspruch wäre eine Integration in den MQTT Collector das Mittel der Wahl. Bei mir läuft im Hintergrund auch noch Home Assistant und EVCC - beide können die API bereits konsumieren.
EVCC darüber hinaus auch noch Grünstromindex für DE und National Grid ESO für UK, siehe https://docs.evcc.io/docs/devices/tariffs#co-vorhersage

Habe es auch mal schnell in EVCC für den Grünstromindex versucht. Über den MQTT Broker ist dann auch folgendes Topic befüllt:
evcc/site/tariffCo2 zB um 17:10 Uhr am 13.06.2024 tariffCo2=44 für PLZ 98708 in DE

Leider konsumiert EVCC von Electricitymap nicht den Wert des latest endpoint, sonder des forecast endpoint und dafür braucht man auch einen bezahlten Token. Also wäre an der Stelle Home Assistant die einfachere Wahl. 

Ein extra Collector macht Solectrus zwar von unabhängiger von umgebenden Services (HA, EVCC,...) aber erscheint aufgrund des soliden MQTT Collectors nicht unbedingt erforderlich.

PS: Coole Sache, dass @spirit0607 diese Services auf den Tisch gebracht hat. Ist natürlich viel sinnvoller als statische Werte...

StevieC121176, 2024-06-13:
Bei mir wäre es ähnlich wie bei @AlpenFlizzer . Ich nutze auch EVCC und das stellt bei mir den Stromindex über "Grünstromindex" bereit. Per MQTT Collector könnte der Wert dann in Solectrus genutzt werden.