solectrus/solectrus #3187: Werte in Herkunft/Verwendung und im Chart sind zeitversetzt

gereons, 2024-06-15:
Seit #3031 ist bei mir praktisch immer ein separates Solectrus-Fenster irgendwo offen, meistens in der "Aktuell"-Ansicht. Dabei ist mir vorhin aufgefallen, dass die Anzeige links bei Herkunft/Verwendung und die des letzten Wertes im Chart rechts zeitlich versetzt sind, und zwar hinkt der Chart um einen "Tick" hinterher. Besonders bei den Schwankungen in der Erzeugung kann man das recht gut beobachten.

Anders ausgedrückt: Mal angenommen, die Verbrauchswerte kommen in folgender Reihenfolge an: 1.1, 1.5, 2.0, 1.8. Dann zeigen Balken links und letzter Tick im Chart folgendes jeweils parallel an:

1.1 / \<der vorherige Wert>
1.5 / 1.1
2.0 / 1.5
1.8 / 2.0
\<nächster Wert> / 1.8
... usw.

ledermann, 2024-06-15:
Gut gesehen, danke! 👍 

Da gibt es - offenbar schon seit längerem - ein Timing-Problem, das ich nun behoben habe - in `develop` verfügbar.

closed 2024-06-15 by ledermann

gereons, 2024-06-16:
Danke, jetzt sieht's gut aus!