solectrus/solectrus #3203: Werte in der Wochenansicht passen nicht zu den Werten in der Tagesansicht

gereons, 2024-06-20:
Mir ist aufgefallen dass manchmal die Werte aus dem Balkendiagramm der Wochenansicht nicht zu denen in der Tagesansicht passen.

Hier ein Beispiel aus KW 24 wo es beim E-Auto sofort auffällt:

[image]

und der Tagesansicht vom 13.6.:

[image]

Laut Wochenansicht sollte ich für den 13.6. folgende Zahlen haben:

Netz: 35,358
Erzeugung: 44,496
Akku-Entnahme: 5,167

E-Auto: 43,612
Haus: 12,93
Einspeisung: 24,167
Akku-Ladung: 4,58

und laut Tagesansicht das hier:

Netz: 28,456
Erzeugung: 44,496
Akku-Entnahme: 4,832

E-Auto: 36,754
Haus: 12,585
Einspeisung: 24,162
Akku-Ladung: 4,58

also recht krasse (identische) Unterschiede vor allem bei E-Auto und Netzbezug.

Wie man sieht hat der erste Ladevorgang für's Auto bereits am 12.6. begonnen, und der dritte war auch erst wieder kurz vor Mitternacht fertig. Mein erster Verdacht dabei war daher sofort dass irgendwas mit den Zeitzonen zwischen meinem lokalen Raspi und meinem Server nicht zusammenpasst, aber da finde ich nix, beide stehen brav auf CEST. Was könnte da die Ursache sein?

ledermann, 2024-06-20:
Das scheint mir auch wirklich nicht korrekt zu sein. Interessant ist hier, dass es sich um einen vergangenen Zeitraum handelt. Wenn ein aktueller, noch laufender Zeitraum betroffen ist, dann ist es kniffliger, aber darum geht es hier ja nicht.

Bezüglich Zeitzone: Schau mal nach, ob beim Raspi alle involvierten Container die Variable TZ bekommen, die auf "Europe/Berlin" stehen sollte. Kann mir zwar nicht vorstellen, dass es so einfach ist, aber das wäre einfach abzuklären.

Ansonsten kann ich mir nur vorstellen, dass wir uns die Influx-Daten ansehen müssen. Aus Performance-Gründen wird intern nicht mit der Integral-Funktion gearbeitet, sondern mit der Summe der Durchschnitte. Ich habe auch kürzlich noch etwas daran herumfeilt, daher: Besteht das Problem schon länger (also auch in älteren Versionen) oder erst seit kurzem?

Kannst du mir vielleicht ein Backup deiner InfluxDB-Datenbank zukommen lassen? Möglichst ab dem 1. Juni würde genügen.

gereons, 2024-06-20:
> Besteht das Problem schon länger (also auch in älteren Versionen) oder erst seit kurzem?  

Kann ich nicht sagen, mit ist es heute zum ersten Mal aufgefallen - ich werde aber mal schauen ob ich andere Beispiele von vor längerer Zeit finde.

> Schau mal nach, ob beim Raspi alle involvierten Container die Variable TZ bekommen, die auf "Europe/Berlin" stehen sollte.

Das war tatsächlich nicht der Fall, TZ war leer (obwohl `date` CEST gemeldet hat) und auch in .env und im compose-File nicht erwähnt. Log-Einträge vom Collector sahen so aus:

`Got record #180062 at 2024-06-20 18:57:57 +0000 within 18 ms, ...` (um 20:57 Ortszeit)

Jetzt hab ich TZ explizit gesetzt, was zu Logs wie `Got record #43 at 2024-06-20 21:02:39 +0200 within 10 ms, ...` führt - das werde ich jetzt mal beobachten ob das schon die Ursache war. Der Collector-Neustart mit dieser Änderung hat aber keine Lücke oder sonstwie sichtbare Spuren im aktuellen Chart hinterlassen. Influx-Backup mach ich dann gerne wenn das Problem weiter auftritt.

ledermann, 2024-06-20:
Ok, hat sich denn dadurch die Darstellung vom 13. Juni und KW 24 geändert? Ich denke, das ist der relevante Testfall. Es ist nicht nötig, auf neue Daten zu warten.

Wenn das Problem sich weiterhin so darstellt, würde ich mir die InfluxDB-Daten gerne mal im Detail ansehen.

ledermann, 2024-06-20:
Sehe gerade, dass ich bei mir auch Abweichungen habe (bei der Wärmepumpe), sehr gering zwar, aber das darf auch nicht sein. Das untersuche ich noch genauer. Insofern ist es erst mal nicht nötig, mir ein Backup zu schicken.

gereons, 2024-06-20:
> Ok, hat sich denn dadurch die Darstellung vom 13. Juni und KW 24 geändert?  

Nur nach der Änderung von TZ auf dem Raspi nicht, nein. Ich habe aber gerade auch mal auf dem Server TZ explizit in der .env gesetzt und an alle Container exportiert, und damit und den geänderten Queries passt jetzt die Tagesanzeige vom 13. wieder zur (unveränderten) Woche:

[image]

Danke für die schnelle Hilfe!

ledermann, 2024-06-21:
Ich hatte gestern Abend noch einen kleinen Fix [committed](https://github.com/solectrus/solectrus/commit/7a9ef0c52c12edb124459e23c3d50dff798da512), der eine vermeintlich sinnvolle Änderung rückgängig gemacht hat, die ich vor 3 Wochen in `develop` vorgenommen hatte. Hintergrund war damals, die Flux-Query zu optimieren. Das führte aber zu einer Ungenauigkeit, die mir erst durch deinen Hinweis aufgefallen ist.

Seit `v0.14.5-222` werden die Queries wieder wie zuvor gebildet. Ich gehe davon aus, dass dies auch der Grund ist, warum deine Darstellung wieder korrekt ist. Mit `TZ` hat es eher nichts zu tun.

gereons, 2024-06-21:
Stimmt, TZ ist egal, ich hab diese Änderung auf meinem Server gerade wieder rückgängig gemacht und die Daten stimmen immer noch :-)

closed 2024-06-21 by gereons