solectrus/solectrus #3218: Mobilansicht: "Mouse-Over" im Chart löst Swipe-Events aus

gereons, 2024-06-25:
Dass man mit #3214 jetzt mobil swipen kann ist super!

Leider gibt's eine kleine Nebenwirkung die vermutlich so nicht beabsichtigt ist: Wenn man zB in einer Tagesansicht unten im Chart mit dem Finger hin- und herfährt um die Werte im Popover zu sehen, wird beim Hochheben des Fingers ein Swipe-Event gefeuert und die Anzeige wechselt auf den nächsten/vorherigen Tag, je nach Bewegungsrichtung.

Eine Swipe-Geste wird generell auch bei extrem kleinen und langsamen Fingerbewegungen erkannt - vielleicht hilft es schon wenn man hier die Geschwindigkeit der Bewegung und die Strecke mit in die Erkennung einfliessen lässt.

(getestet mit iOS 17 und iPadOS 17)

ledermann, 2024-06-25:
Hm, stimmt, das kann so nicht bleiben. Ich habe nun etwas nachgebessert - Geschwindigkeit und Strecke fließen jetzt mit in die Beurteilung ein, ob es eine Swipe-Geste ist oder nicht. Zu sehen in `develop`.

gereons, 2024-06-25:
Super, danke! Mit dieser Änderung fühlt sich's für mich gut an

closed 2024-06-25 by gereons