solectrus/solectrus #3219: Solectrus zeiugt Ständig "Keine Verbindung"  aber es werden trotzdem aktuelle Werte angezeigt

jacitty, 2024-06-25:
Seit ein paar Wochen sehe ich ständig "Keine Verbindung" aber es werden die aktuellen Werte angezeigt. 
[image]

ledermann, 2024-06-25:
Die Anzeige "Keine Verbindung" erfolgt dann, wenn der Zeitpunkt der letzten Messung zu lange her ist. Meist passiert das, wenn `INFLUX_POLL_INTERVAL` zu klein ist, insbesondere kleiner als das Intervall, mit dem die Daten hereinkommen.

Wenn das nicht hilft: Welche Version verwendest du? Wo kommen die Daten her (welche Kollektoren)? Wie sieht deine `.env` und `docker-compose.yml`? Was hast du geändert, bevor dieses Problem das erste mal auftrat?

jacitty, 2024-06-25:
Hi. Danke für die schnelle Antwort. 
Als Version habe ich die v0.14.5 im Einsatz, das wird jede Nacht bei Bedarf aktualisiert.
Ich vermute, es könnte sein dass der Fehler auftritt seit ich vor kurzem in der .env die Parameter für die Anlagenausrichtung angepasst habe. Ich kann zumindest keinen Parameter in der .env namerns INFLUX_POLL_INTERVAL finden. Merkwürdig, da ich dort nix geändert habe. Ich füge jetzt ,al wieder INFLUX_POLL_INTERVAL=5 ein und starte neu.

ledermann, 2024-06-25:
`INFLUX_POLL_INTERVAL` steht hier:
https://github.com/solectrus/hosting/blob/main/guide/raspberry-pi/.env#L60-L61

und hier (nicht vergessen!):
https://github.com/solectrus/hosting/blob/main/guide/raspberry-pi/docker-compose.yml#L38

Wenn der Wert nicht gesetzt ist, wird ein Default von `5` angenommen.

Bei weiter bestehenden Problemen bitte unbedingt `.env` und `docker-compose.yml` posten, sonst ist das alles nur ein Raten.

jacitty, 2024-06-25:
Das ist ja nun doch etwas merkwürdig, da ich an der docker-compose.yml nichts geändert habe. Trotzdem fehlen dort einige Zeilen.
**.env:**
[env.txt](https://github.com/user-attachments/files/15972703/env.txt)

**docker-compose.yml:**
[docker-compose.yml.txt](https://github.com/user-attachments/files/15972716/docker-compose.yml.txt)

jacitty, 2024-06-25:
Nachdem ich die docker-compose.yml korrigiert habe startet docker zum einen viel schneller und der Raspberrry fühlt sich auch zum andern wieder viel schneller an. Starten/Stoppen von docker dauert nicht mehr "eine Ewigkeit", die Startzeiten sind von  ca. 80 sec runter auf 20 sec. Vermutlich war der Raspi so dermaßen ausgebremst dass die Daten vom senec nicht mehr rechtzeitig abgeholt werden konnten. Ob das nur alleine an der defekten yml-File lag kann ich nicht sagen, ich hatte aber jeweils mit defekter als auch korrigierter Datei jeweils 1x komplett den raspi neu gestartet.

ledermann, 2024-06-25:
Es ist so, dass ich die Anleitung mit der Konfiguration im Laufe der Zeit oftmals verbessert und ausgebaut habe. `INFLUX_POLL_INTERVAL` ist beispielsweise vor 9 Monaten dazugekommen, das war vorher fest auf 5 eingestellt, nun ist es konfigurierbar. Wenn du es jetzt selbst manuell auf 5 stellst, sollte sich eigentlich nichts ändern.

Dass der ganze Raspi jetzt viel schneller ist, wundert mich sehr. Das klingt nach etwas ganz anderen Problem, was du möglicherweise nebenbei behoben hast. Ich würde mal prüfen, welche Docker-Container laufen, ob möglicherweise die Docker-Logs vollgelaufen sind, was `docker stats` sagt, ob `apt update && apt upgrade` vielleicht schon lange nicht mehr gemacht wurde usw.

Verstehe ich dich denn richtig, dass jetzt alles wieder korrekt ist, die Anzeige "Keine Verbindung" also nicht mehr erscheint?

jacitty, 2024-06-26:
So, ich konnte mir das ganze jetzt mal in Ruhe anschauen. 
Ja, apt update und apt upgrade liefen schon lange nicht mehr und da lag einiges im Argen. Ist jetzt behoben. 
Jetzt hab ich mir die .yml mal genauer angeschaut. Die Aktuelle hier aus github hat doch ein paar neuere Versionen genannt (influx 2.7, bisher 2.6, postgres 16, bisher 15). So 1:1 kann ich die nicht einfach austauschen, da bekomme ich doch andere Probleme, z. B. wird mir Influx als "uinhealthy" gemeldet. 
Gibt es eine Anleitung, wie ich die Versionen innerhalb des Containers aktualisieren kann? Wie mache ich eigentlich Sicherungen der DB? Oder einfach immer das ganze solectrus-Verzeichnis sichern? Da bekomme ich dann Probleme mit den Berechtiugungen, da einige Dateien für den User nicht lesbar sind (.../solectrus/pgsql und .../solectrus/redis):

```
> pi@raspberrypi:~/solectrus $ ls -al
> insgesamt 84
> drwxr-xr-x  6 pi   pi    4096 25. Jun 17:08 .
> drwxr-xr-x 19 pi   pi   12288 25. Jun 17:08 ..
> drwxr-xr-x  2 pi   pi   12288 18. Jan 14:37 csv
> -rw-r--r--  1 pi   pi    2552 25. Jun 17:20 docker-compose.yml
> -rw-r--r--  1 pi   pi    2552 25. Jun 16:11 docker-compose.yml.old
> -rw-r--r--  1 pi   pi    5928 25. Jun 16:38 docker-compose.yml.orig
> -rw-r--r--  1 pi   pi    5301 25. Jun 17:08 .env
> -rw-r--r--  1 pi   pi    5301 25. Jun 17:01 env.new
> drwx------  4 pi   pi    4096 14. Dez 2023  influxdb
> drwx------ 19   70 pi    4096 26. Jun 00:05 postgresql
> drwxr-xr-x  2  999 pi    4096 26. Jun 06:48 redis
```

ledermann, 2024-06-26:
Zu den Versionen:
- InfluxDB kann meiner Erfahrung nach problemlos von 2.6 auf 2.7 eingestellt werden. Das würde ich auch empfehlen
- PostgreSQL darf **nicht** einfach so von 15 auf 16 eingestellt werden, da dies ein Backup/Restore erfordert. Dann kann man bei Interesse machen, aber es ist auch problemlos möglich, bei 15 zu bleiben. Jedes PostgreSQL-Release wird 5 Jahre mit Updates versorgt.

Alle anderen Services (also die von SOLECTRUS selbst) in der `docker-compose.yml` sollten das `latest`-Tag verwenden. Richtet man dann noch WatchTower ein (siehe https://github.com/solectrus/hosting/blob/main/guide/raspberry-pi/docker-compose.yml#L215-L226), dann kommen Updates automatisch, was ich **sehr empfehle**.

Backups siehe https://github.com/orgs/solectrus/discussions/3014. Manuelles Kopieren kann man machen (mit `sudo` wegen der Rechte), aber dazu müssen vorher alle Container beendet werden. Das ist daher nur eine Notlösung.

Ist der ursprüngliche Fehler mit "Keine Verbindung" denn jetzt beseitigt?

jacitty, 2024-06-26:
> Ist der ursprüngliche Fehler mit "Keine Verbindung" denn jetzt beseitigt?

So wie es aussieht ja, ich vermute durch zu hohe Last auf dem Raspi (warum auch immer) wurden die Daten nicht immer rechtzeitig empfangen.

Den Rest überarbeite ich mal.

jacitty, 2024-06-26:
So, jetzt hatte ich dann mal alles überarbeitet, die docker-compose.yml und .env sind hier im Anhang. 

[docker-compose.yml.txt](https://github.com/user-attachments/files/15983687/docker-compose.yml.txt)
[env.txt](https://github.com/user-attachments/files/15983689/env.txt)

Leider tritt das Problem mit der Verbindung immer noch auf, wie ein Blinker (keine Verbindung -> Verbunden -> keine Verbindung) Hover-Text: "letzte Messung vor xxx" wobei ich XXX jetzt schon als "halbe Minute" und auch als "vor weniger als 20 Sekunden" gesehen habe. Keine Ahnung warum, Auslastung des Systems ist ok.

ledermann, 2024-06-26:
Ich gehe davon aus, dass du nach einer Änderung von `.env` oder `docker-compose.yml` die Container neu erstellt hast, also `docker compose up -d` ausgeführt hast. Wenn nicht, dann bitte nachholen.

Laut der Config ist `INFLUX_POLL_INTERVAL=5` und `SENEC_INTERVAL=5`. Das passt: Alle fünf Sekunden wird ein Messwert abgeholt und das UI aktualisiert ebenfalls alle 5 Sekunden. Wenn der SENEC-Speicher korrekt liefert, dann sollte es keine Fehlermeldung geben. Die Meldung "keine Verbindung" erscheint übrigens, wenn der **jüngste** Messwert älter ist als `2 * INFLUX_POLL_INTERVAL`, in deinem Fall also 10 Sekunden. Es ist nun die Frage, warum das bei dir passiert.

Bitte prüfe, ob die Messwerte wirklich korrekt abgeholt und in die Datenbank gespeichert werden und führe das hier aus: `docker compose logs senec-collector`. Wie sieht der Output aus? Bitte poste mal die neuesten Zeilen. Am Zeitstempel lässt sich erkennen, ob es wirklich im 5-Sekunden-Turnus passiert.

Hat es einen Grund, dass du auf InfluxDB 2.6 geblieben bist? Es gab viele Bugfixes in der 2.7, das Update ist empfehlenswert. Das würde ich installieren, um weitere Fehlerquellen auszuschließen. Oben hattest du von einem "unhealthy" Status geschrieben, aber da vermute ich eine Verwechselung mit PostgreSQL. Wenn InfluxDB wirklich kein Upgrade auf 2.7 zulässt, liegt ein ernsteres Problem vor. Dann bitte prüfen, ob es im Log des InfluxDB-Containers Hinweise auf eine Ursache gibt.

jacitty, 2024-06-26:
Hallo.
Mensch bist du schnell. Ich komme ja kaum damit nach alles noch zu prüfen und Watchtower noch einzubinden. Das Update von InfluxDB 2.6  auf 2.7 wollte ich später machen, hab das aber jetzt gemacht. 

Also:
docker compose down
<Anpassung an docker_compose.yml>
docher copmpose pull
docker compose up -d 

Im Moment sehe ich mal wieder keine Aussetzer, aber die waren auch nicht immer zu sehen.
Trotzdem hier die letzten Zeilen von `docker compose logs senec-collector`:
Ups, jetzt hab ich dort doch Fehler drin stehen (trotz InfluxDB 2.7):

```
senec-collector-1  | Successfully pushed record #116 to InfluxDB
senec-collector-1  | Successfully pushed record #117 to InfluxDB
senec-collector-1  | 
senec-collector-1  | Got record #118 at 2024-06-26 09:35:12 +0000 within 9 ms, LADEN, Inverter 4170 W, House 3447 W, Wallbox 0 W
senec-collector-1  | 
senec-collector-1  | Got record #119 at 2024-06-26 09:35:17 +0000 within 10 ms, LADEN, Inverter 4170 W, House 3446 W, Wallbox 0 W
senec-collector-1  | Error while pushing record #118 to InfluxDB: Net::ReadTimeout with #<TCPSocket:(closed)>
senec-collector-1  | The record has been queued. Will retry to push 2 records later.
senec-collector-1  | 
senec-collector-1  | Got record #120 at 2024-06-26 09:35:22 +0000 within 8 ms, LADEN, Inverter 4174 W, House 3443 W, Wallbox 0 W
senec-collector-1  | 
senec-collector-1  | Got record #121 at 2024-06-26 09:35:27 +0000 within 11 ms, LADEN, Inverter 4177 W, House 3453 W, Wallbox 0 W
senec-collector-1  | Successfully pushed record #119 to InfluxDB
senec-collector-1  | 
senec-collector-1  | Got record #122 at 2024-06-26 09:35:32 +0000 within 10 ms, LADEN, Inverter 4178 W, House 3445 W, Wallbox 0 W
senec-collector-1  | 
senec-collector-1  | Got record #123 at 2024-06-26 09:35:37 +0000 within 9 ms, LADEN, Inverter 4179 W, House 3452 W, Wallbox 0 W
senec-collector-1  | Error while pushing record #118 to InfluxDB: Net::ReadTimeout with #<TCPSocket:(closed)>
senec-collector-1  | The record has been queued. Will retry to push 5 records later.
senec-collector-1  | 
senec-collector-1  | Got record #124 at 2024-06-26 09:35:42 +0000 within 9 ms, LADEN, Inverter 4177 W, House 3437 W, Wallbox 0 W
senec-collector-1  | 
senec-collector-1  | Got record #125 at 2024-06-26 09:35:47 +0000 within 11 ms, LADEN, Inverter 4180 W, House 3452 W, Wallbox 0 W
senec-collector-1  | 
senec-collector-1  | Got record #126 at 2024-06-26 09:35:52 +0000 within 8 ms, LADEN, Inverter 4176 W, House 3451 W, Wallbox 0 W
senec-collector-1  | Error while pushing record #120 to InfluxDB: Net::ReadTimeout with #<TCPSocket:(closed)>
senec-collector-1  | The record has been queued. Will retry to push 8 records later.
senec-collector-1  | 
senec-collector-1  | Got record #127 at 2024-06-26 09:35:57 +0000 within 15 ms, LADEN, Inverter 4064 W, House 3319 W, Wallbox 0 W
senec-collector-1  | Successfully pushed record #121 to InfluxDB
senec-collector-1  | Successfully pushed record #122 to InfluxDB
senec-collector-1  | Successfully pushed record #123 to InfluxDB
senec-collector-1  | Successfully pushed record #118 to InfluxDB
senec-collector-1  | 
senec-collector-1  | Got record #128 at 2024-06-26 09:36:02 +0000 within 9 ms, LADEN, Inverter 4181 W, House 3457 W, Wallbox 0 W
senec-collector-1  | Successfully pushed record #124 to InfluxDB
senec-collector-1  | 
senec-collector-1  | Got record #129 at 2024-06-26 09:36:07 +0000 within 177 ms, LADEN, Inverter 4182 W, House 3457 W, Wallbox 0 W
senec-collector-1  | Successfully pushed record #125 to InfluxDB

```

ledermann, 2024-06-26:
Interessant, die Fehlermeldung `Net::ReadTimeout` scheint die Ursache zu sein. Der SENEC-Collector kann manchmal die Messwerte nicht bei InfluxDB abladen.  Das ist ungewöhnlich. Es ist nun herauszufinden, warum das geschieht.

Eine mögliche Erklärung wäre ein Defekt der SD-Karte, das hatten wir bereits mehrfach hier. 

Ansonsten:
- Du könntest mal in die Logs von InfluxDB gucken: `docker compose logs influxdb`
- Ansonsten mal die Auslastung des Raspis prüfen mit `top`. Bei mir nahe `0`.
- Auch die Ausgabe von `docker stats` könnte hilfreich sein. Eigentlich sollten sich die Container allesamt nur langweilen und eine CPU-Auslastung nahe 0 % haben. Bei mir springt die Auslastung des InfluxDB-Containers alle paar Sekunden auf höchstens 10%. Wenn das bei dir anders sein sollte, müsstest du deinen Raspi genauer untersuchen.