solectrus/solectrus #3224: docker-compose.yml: `version` is obsolete

Zeppelin-PV, 2024-06-27:
hallo Georg
ich habe beim Abholen der develop Version diese Meldung bekommen
was hat dies zu bedeuten ?
muß ich da was tun ?

admin@raspberrypi:/home/pi/solectrus $ sudo docker compose pull && sudo docker compose up -d
WARN[0000] /home/pi/solectrus/docker-compose.yml: `version` is obsolete 
[+] Pulling 17/17
 ✔ db Pulled                                                                                                             1.3s 
 ✔ forecast-collector Pulled                                                                                             0.7s 
 ✔ app Pulled                                                                                                           11.9s 
   ✔ a258b2a6b59a Already exists                                                                                         0.0s 
   ✔ 43f659b8850f Already exists                                                                                         0.0s 
   ✔ 3180c4dd2ce0 Already exists                                                                                         0.0s 
   ✔ 7de5b45e5159 Already exists                                                                                         0.0s 
   ✔ cdba618368b9 Already exists                                                                                         0.0s 
   ✔ 890697fcbee2 Pull complete                                                                                          1.8s 
   ✔ beebeb81d1c6 Pull complete                                                                                          1.8s 
   ✔ e044c98ea70b Pull complete                                                                                          1.8s 
   ✔ ffec25c7ebff Pull complete                                                                                          6.8s 
   ✔ c8243b7bee1d Pull complete                                                                                         10.8s 
   ✔ 244113790ae5 Pull complete                                                                                         10.8s 
 ✔ redis Pulled                                                                                                          1.3s 
 ✔ influxdb Pulled                                                                                                       1.2s 
 ✔ senec-collector Pulled                                                                                                0.7s 
WARN[0000] /home/pi/solectrus/docker-compose.yml: `version` is obsolete 
[+] Running 6/6
 ✔ Container solectrus-redis-1               Running                                                                     0.0s 
 ✔ Container solectrus-influxdb-1            Running                                                                     0.0s 
 ✔ Container solectrus-senec-collector-1     Running                                                                     0.0s 
 ✔ Container solectrus-db-1                  Running                                                                     0.0s 
 ✔ Container solectrus-forecast-collector-1  Running                                                                     0.0s 
 ✔ Container solectrus-app-1                 Started

ledermann, 2024-06-27:
Bei neueren Versionen von Docker Compose (seit 1.27+) ist die Angabe von `version` in der `docker-compose.yml` überflüssig geworden. Die Zeile kannst du gefahrlos entfernen. Oder lass sie drin, abgesehen von der Warnmeldung passiert nichts weiter.

closed 2024-06-27 by ledermann

Zeppelin-PV, 2024-06-27:
ist das diese erste Zeile vor services ?

version: '3.7'

services:

ledermann, 2024-06-28:
Ja, genau um diese Zeile mit `version: '3.7'` geht es. Die kann weg, sodass die Datei mit `services:` beginnt.