solectrus/solectrus #3232: Tagesansicht: Wert in "Verwendung" findet sich nicht im Chart

gereons, 2024-07-01:
Sorry, ich schon wieder mit einer Inkonsistenz in den Charts...

Für den 28.6. sagt mir die "Verwendung", dass ich ca. 2,9 kWh in mein Auto geladen habe, aber im Chart taucht nichts davon auf:

[image]

Gegen 22h am 27.6. wurde noch ein Ladevorgang gestartet, der Tag sieht so aus:

[image]

Im Influx-Explorer sehen beide Tage zusammen so aus:

[image]

Hier fällt die Kurve nach der 3. Ladung um 2024-06-28-00:07:00 wieder auf 0. In den 7 Minuten können bei dem Ladestrom aber keine 2,9 kWh zusammengekommen sein...

Das scheint mir noch irgendwie mit #3203 und den Messungen über Tagesgrenzen hinweg zusammenzuhängen.

Backups meiner Influxdb stelle ich bei Bedarf gerne zur Verfügung.

ledermann, 2024-07-01:
Danke für deine nicht nachlassende Unterstützung :)

Anhand der Beschreibung konnte ich einen Fehler beheben, der seit Version 0.11 enthalten ist und dazu führte, dass bei der Summenberechnung (also Herkunft+Verwendung) fälschlicherweise immer eine Stunde des vorherigen Zeitraums mit einbezogen wurde. Statt 24 wurden also 25 Stunden aufsummiert.

Die Korrektur ist in `develop` enthalten (ab `v0.14.5-294`). Nach Installation des Updates sollte die Darstellung des 28. Juni korrekt sein, d.h. die E-Auto-Beladung sollte dann einen sehr geringen Wert enthalten (etwa 0,1 kWh schätze ich). Sämtliche andere Zeiträume werden sich ebenfalls (geringfügig) ändern.

gereons, 2024-07-01:
Wunderbar, mit der Version sieht's jetzt passend aus. Danke!

closed 2024-07-01 by gereons