solectrus/solectrus #3241: Inkonsistente Aggregation von Tageswerten

ledermann, 2024-07-03:
An verschiedenen Stellen gibt es seit längerem kleinere Ungenauigkeiten bei der Aggregation von Messwerten:

- Top10: Ein Balken der Top10 weicht vom tatsächlichen Wert bei Tag/Woche/Monat ab
- Diagramme nach Woche/Monat: Höhe des Balkens weicht vom tatsächlichen Wert eines Tages ab

Die Abweichungen sind meist minimal und resultieren daraus, dass die Zeitzone nicht korrekt berücksichtigt wird. Die Tage werden nicht exakt von 0 bis 24 Uhr gebildet, sondern um ein oder zwei Stunden versetzt (je nach Sommer oder Winter). Hat man aber z.B. um Mitternacht herum hohe Verbrauchsspitzen, dann fällt diese Ungenauigkeit deutlicher ins Gewicht.

Ähnliche Probleme hatten wir bereits in #3232 und #3203.

Unabhängig davon sind #1894 und #2032, bei denen es um eine Ungenauigkeit durch einen noch nicht abgeschlossenen Zeitraum geht. Das ist ein anderes Problem, das ich auch separat lösen möchte.

ledermann, 2024-07-03:
Die Korrekturen befinden sich jetzt in `develop`. In der Live-Demo kann ich dadurch keine Abweichungen mehr entdecken. Mag das jemand mit seinen Daten ausprobieren? @gereons vielleicht?

gereons, 2024-07-03:
LGTM! Ich hab mal ein wenig in meiner Instanz herumgeklickt und besonders auf die geänderten Stellen geachtet, da ist mir auch nichts mehr aufgefallen.

closed 2024-07-07 by ledermann