solectrus/solectrus #3281: The data directory was initialized by PostgreSQL version 15, which is not compatible with this version 16.3

Oggy512, 2024-07-21:
### Description (in German or English)

Hallo Georg,
ich habe eben ein Update auf v0.15 gemacht, das läuft soweit auch problemlos.
Ich habe im Anschluss die .env und docker-compose.yml gegen die aktuellste Version (aus dem Online Konfigurator) getauscht und die alten Werte für Pwds, Hosts etc pp 1:1 übernommen.
Nun erhalte ich beim Start den Fehler, dass die PSQL Version nicht mehr kompatibel ist.
Du hast doch sicher eine Lösung, wie man vorgehen sollte, wenn man von einer alten Solectrus Version kommt und auf aktuelle Configs wechseln will/soll?

Besten Dank!

### What version of SOLECTRUS are you using?

0.15.0

### What machine are you running SOLECTRUS on?

Debian 12 VM

### If applicable (e.g. in case of error 500), please provide logs

```text
~/solectrus$ docker logs solectrus-postgresql-1

PostgreSQL Database directory appears to contain a database; Skipping initialization

2024-07-21 10:58:04.406 UTC [1] FATAL:  database files are incompatible with server
2024-07-21 10:58:04.406 UTC [1] DETAIL:  The data directory was initialized by PostgreSQL version 15, which is not compatible with this version 16.3.

PostgreSQL Database directory appears to contain a database; Skipping initialization

2024-07-21 10:58:04.816 UTC [1] FATAL:  database files are incompatible with server
2024-07-21 10:58:04.816 UTC [1] DETAIL:  The data directory was initialized by PostgreSQL version 15, which is not compatible with this version 16.3.

PostgreSQL Database directory appears to contain a database; Skipping initialization

2024-07-21 10:58:05.372 UTC [1] FATAL:  database files are incompatible with server
2024-07-21 10:58:05.372 UTC [1] DETAIL:  The data directory was initialized by PostgreSQL version 15, which is not compatible with this version 16.3.

PostgreSQL Database directory appears to contain a database; Skipping initialization

2024-07-21 10:58:06.284 UTC [1] FATAL:  database files are incompatible with server
2024-07-21 10:58:06.284 UTC [1] DETAIL:  The data directory was initialized by PostgreSQL version 15, which is not compatible with this version 16.3.
```

### Please provide your `docker-compose.yml` file

```yaml
# version: '3.7'

services:
  app:
    image: ghcr.io/solectrus/solectrus:latest
    depends_on:
      db:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - db
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - INSTALLATION_DATE
[... 332 lines cut ...]
# Set these names before first run, they can't be changed later!
INFLUX_BUCKET=solectrus

# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=xxx
INFLUX_TOKEN_READ=xxx

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=/home/pi/solectrus/influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=/home/ladmin/solectrus/postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=/home/ladmin/solectrus/redis
```

ledermann, 2024-07-21:
Die Fehlermeldung bei PostgreSQL besagt, dass PostgreSQL 16 läuft, die mit den Daten von PostgreSQL 15 aber nicht zurechtkommt. Das ist korrekt so, ein Datenbank-Upgrade ist leider umständlicher, man muss ein Backup/Restore durchführen, was aber umständlich ist und sich auch nicht lohnt, bleibe einfach bei PostgreSQL 15.

Allerdings stehe in deiner compose.yml bereits die korrekte Zeile:

```yml
image: postgres:15-alpine
```

Bist du sicher, dass das Fehler-Log wirklich mit dieser `compose.yml` auftritt? Ich habe eher die Vermutung, dass die von dir gepostete `docker.yml` die alte ist, nicht die neue generierte. Der Inhalt jedenfalls enthält die ALTE Konfiguration.

Nimmt also einfach die **neu generierte** `compose.yml`, ändere darin den Eintrag `postgres:16-alpine` zu `postgres:15-alpine` und fertig. 

Übrigens: `compose.yml` ist der zeitgemäße Dateiname, `docker-compose.yml` ist eine alte Schreibweise. Beide Namen sind gültig, aber sollte nicht beide Dateien gleichzeitig haben.

Oggy512, 2024-07-21:
Danke für den Tipp,  das hat geklappt.
Allerdings klappt die Abfrage des Shellys nicht, das log sagt
```

shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error
shelly-collector-1    | Error getting data from Shelly at http://10.10.13.16: uninitialized constant ShellyAdapter::Error

```
Die IP stimmt aber auf jeden Fall. Was mir auffällt: die Abfragt geht als http raus, der Shelly antwortet aber auf https. Der Shelly macht aber ja selbst ne redirection nach https, sodass es hier eigentlich kein Problem geben sollte.
Muss ich am Shelly selbst noch etwas einstellen?

ledermann, 2024-07-21:
Ok, prima. Das Shelly-Problem klären wir in solectrus/shelly-collector#89

closed 2024-07-21 by ledermann