solectrus/solectrus #3301: Diagramm-Skalierung nicht immer optimal

ledermann, 2024-07-26:
Auf der Seite AKTUELL wird der Verbrauch, der sich prinzipiell nur zwischen 0% und 100% bewegen kann, auf 150% skaliert. Das ist ungünstig und sollte korrigiert werden.

[image]

Gemeldet von @WolfImBusch in https://github.com/orgs/solectrus/discussions/3299

WolfImBusch, 2024-07-26:
.. kleine Ergänzung noch ... betrifft nicht nur die Seite "AKTUELL", sondern auch "TAG"

ledermann, 2024-07-26:
Ist jetzt korrigiert, für AKTUELL und TAG. Bei Verbrauch und Autarkie reicht das Diagramm jetzt immer von 0% bis 100%.

Zu sehen in der Live-Demo, verfügbar im nächsten regulären Update.

closed 2024-07-26 by ledermann

WolfImBusch, 2024-07-26:
Suuuuuuuuuper! Deine Geschwindigkeit macht mir Angst! :-)