solectrus/solectrus #3309: Keine Aktualisierung der Reiter "Woche", "Monat", "Jahr" für Erzeugung

StevieC121176, 2024-07-30:
### Description (in German or English)

Mit Watchdog wird bei mir die Version aktuell gehalten (develop) seit gestern wird bei mir die Erzeugung für den Monat nicht mehr aktualisiert und seit heute zusätzlich für Woche und Jahr nicht.
Die Heute Ansicht wird korrekt dargestellt und aktualisiert. Ich habe mal ein paar Screenshots gemacht.

[image]
[image]
[image]
[image]
[image]

### What version of SOLECTRUS are you using?

v0.15.0-71

### What machine are you running SOLECTRUS on?

Raspberry Pi 4 8GB

StevieC121176, 2024-07-30:
Ich sehe gerade, auf demo.solectrus.de kann man das auch nachvollziehen. Ab Sonntag Abend stimmen da die Daten auch nicht mehr in der Monatsansicht und ab Montag bei den anderen Ansichten auch nicht mehr. (Außer Tagesansicht)

ledermann, 2024-07-30:
Danke für den Hinweis. Habe kürzlich das Caching "optimiert", das hatte wohl Nebenwirkungen. Ich kümmere mich darum.

ledermann, 2024-07-30:
Die vor zwei Tagen von mir vorgenommene fehlerhafte Verbesserung führte bei der Monatsanzeige zu einer zu langen Caching-Dauer. Ich habe das rückgängig gemacht. 

Wer die fehlerhafte Entwickler-Version zwischendurch im Einsatz hatte und das Problem immer noch beobachtet, muss den Cache löschen:

```
docker exec -it solectrus-redis-1 redis-cli FLUSHALL
```

Sorry, Leute!

closed 2024-07-30 by ledermann

StevieC121176, 2024-07-30:
Nach Löschen des Caches funktioniert wieder alles bestens.