solectrus/solectrus #3311: Nach Update sind Wochen-/Monats-Diagramme durcheinander

frizzle0815, 2024-07-31:
### Description (in German or English)

Solectrus hat sich am 30.07. auf die neuste develop-Version aktualisiert. Redis wurde auch aktualisiert.
Seitdem ist irgendetwas mit den Wochen- und Monats-Übersicht durcheinander. Für die Jahresübersicht kann ich es schlecht verifzieren. Die Tagesansicht funktioniert anscheinend ganz normal.

"Erzeugung":
- Werte bis 29.07. sind konsistent und vollständig
- Werte ab 30.07. sind in der Wochen- und Monats-Ansicht leer (bzw. 30.07. zeigt einen minimalen Wert an, vielleicht bis zum Zeitpunkt des Updates)

"Netz"
- Werte bis 29.07. sind konsistent und vollständig
- Werte ab 30.07. werden in der Wochen-Ansicht angezeigt, in der Monats-Ansicht aber gar nicht.

"Haus-Verbrauch"
- Werte bis 29.07. sind konsistent und vollständig
- Werte ab 30.07. scheinen auch zu passen
- Werte am 31.07. (heute) Tag zeigt 6,8 kWh, Woche 7,251 kWh, Monat 7,298 kWh (immerhin ist hier für Monat aber ein Wert vorhanden)

"Akku"
- Werte bis 29.07. sind konsistent und vollständig
- Werte am 30.07. Tag zeigt Entl. 7,7+Bel. 7,9, passt aber nicht zu: Woche+Monat: Bel. 7,9+Entl 5,092
- Werte am 31.07. (heute) werden in Woche/Monat gar nicht angezeigt

"Co2 Reduktion"
Keine Werte in Wochen-/Monats-Ansicht ab 30.07.

Normal zu funktionieren scheint "Ladestand", "Autarkie", "Verbrauch". Temperatur erfasse ich nicht und das E-Auto wurde seit 30.07. nicht geladen, weshalb ich dazu auch nichts sagen kann...

### What version of SOLECTRUS are you using?

v0.15.0-71-g9d346cdg

### What machine are you running SOLECTRUS on?

Synology

### Please provide your `docker-compose.yml` or `compose.yml` file

```yaml
services:
  dashboard:
    image: ghcr.io/solectrus/solectrus:develop
    depends_on:
      postgresql:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - postgresql
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - WEB_CONCURRENCY
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - DB_HOST=postgresql
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
[... 289 lines cut ...]
FORECAST_0_AZIMUTH=20
FORECAST_0_KWP=5.34
FORECAST_1_DECLINATION=18
FORECAST_1_AZIMUTH=110
FORECAST_1_KWP=6.23
FORECAST_2_DECLINATION=15
FORECAST_2_AZIMUTH=20
FORECAST_2_KWP=2.67
FORECAST_INTERVAL=2700
FORECAST_INFLUX_MEASUREMENT=forecast
FORECAST_INVERTER=10.0
INFLUX_HOST=influxdb
INFLUX_SCHEMA=http
INFLUX_PORT=8086
INFLUX_ORG=solectrus
INFLUX_USERNAME=admin
INFLUX_PASSWORD=ExAmPl3PA55W0rD
INFLUX_ADMIN_TOKEN=my-super-secret-admin-token
INFLUX_BUCKET=solectrus
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token
INFLUX_VOLUME_PATH=/volume1/docker/solectrus/influxdb
DB_VOLUME_PATH=/volume1/docker/solectrus/postgresql
REDIS_VOLUME_PATH=/volume1/docker/solectrus/redis
```

ledermann, 2024-07-31:
Danke für deine sehr präzise Beschreibung, das Problem ist bereits in `v0.15.0-72` behoben, siehe #3309 (nach dem Update auf diese Version Cache löschen. Wie das geht steht auch dort). Ich schließe dein Issue (da Duplikat).

Solche Fehler sind leider die kleinen Nachteile, wenn man die Developer-Version verwendet. Wer das vermeiden möchte, kann auch auf die stabile Version wechseln (Docker-Tag `latest` statt `develop`)

closed 2024-07-31 by ledermann

frizzle0815, 2024-07-31:
Vielen Dank! Ich bin auch erst einmal wieder auf `latest` gewechselt und nach dem Leeren des Cache klappt wieder alles.

Ich nehme für mich mit, dass ich vor dem nächsten posten auch die bereits geschlossenen Issues lesen sollte; dann hätte sich meine Frage schon geklärt. Du bist einfach unfassbar schnell im Lösen von Problemen! 👍

ledermann, 2024-07-31:
Ja, Suchfunktion verwenden oder Durchgucken vorhandener Issues ist sinnvoll, bevor man ein neues anlegt.

Ansonsten: Ich bin übrigens auch super gut darin, neue Probleme zu erschaffen ;-)