solectrus/solectrus #3319: Nach Update auf MQTT und neuere docker-compose, sind all historischen Daten verschwunden

ronzle, 2024-08-02:
### Description (in German or English)

Ich habe heute auf mqtt-collector umgestellt und dafuer dann eben auch die neuste version der docker-compose.yml benutzt. 
Meine .env habe ich einfach nur um die fehlenden variablen erweitert. 
Nun sehe ich grad das keine Daten aelter seit  heute angezeigt werden. 
Mein Installationsdatum is Okt 2022
Sieht so aus als ob entweder ein neues Influxdb Bucket benutzt wird oder die Daten beim Update der influxdb von v2.4 auf v2.7 verschwunden sind 

### What version of SOLECTRUS are you using?

v0.15.0

### What machine are you running SOLECTRUS on?

Virtual Machine

### Please provide your `docker-compose.yml` or `compose.yml` file

```yaml
version: '3.7'

services:
  app:
    image: ghcr.io/solectrus/solectrus:latest
    labels:
      - 'com.centurylinklabs.watchtower.scope=solectrus'
    depends_on:
      - db
      - influxdb
      - redis
    links:
      - db
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - DB_HOST=db
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_MEASUREMENT_PV
      - INFLUX_MEASUREMENT_FORECAST
      - INFLUX_POLL_INTERVAL
      - WEB_CONCURRENCY
    healthcheck:
      test: ['CMD-SHELL', 'nc -z 127.0.0.1 3000 || exit 1']
    restart: always
    logging:
      options:
        max-size: '10m'
        max-file: '3'

  influxdb:
    image: influxdb:2.7-alpine
    labels:
      - 'com.centurylinklabs.watchtower.scope=solectrus'
    volumes:
      - ${INFLUX_VOLUME_PATH}:/var/lib/influxdb2
    environment:
[... 281 lines cut ...]
# MQTT options
MQTT_FLIP_BAT_POWER=true

##################################################################
# Solar forecasting with https://forecast.solar
# API docs: https://doc.forecast.solar/doku.php?id=api:estimate
#
# Latitude of the plant location
FORECAST_LATITUDE=48.9140506
#
# Longitude of the plant location
FORECAST_LONGITUDE=8.2468035
#
# Plane declination: 0 (horizontal) - 90 (vertical)
FORECAST_DECLINATION=30
#
# Plane azimuth: -180 ... 180 (-180 = north, -90 = east, 0 = south, 90 = west, 180 = north)
FORECAST_AZIMUTH=20
#
# Installed modules power in kilowatt peak (kWp)
FORECAST_KWP=12.5
#
# Update interval in seconds, 900s = 15 minutes, the public (and free) API allows a minimum of 900 seconds
FORECAST_INTERVAL=900
```

ronzle, 2024-08-02:
when i query the influxdb i can still see data from the past 

influx query 'from(bucket:"senec") |> range(start:-1y)' --raw

,,1,2023-08-03T09:40:25.25062489Z,2024-08-02T15:40:25.25062489Z,2023-12-20T03:36:38Z,0826,application_version,SENEC

ronzle, 2024-08-02:
Ok ich habs gefixt, hier mal fuer alle die in das selbe problem laufen ....
Bevor es die option 

INFLUX_MEASUREMENT_PV=

wurden alle time series daten als SENEC in der influxdb gespeichert. 
Ich habe als ich die variable INFLUX_MEASUREMENT_PV in meine .env uebernommen habe, den Wert belassen und ab dem Zeitpunkt wurden alle neuen daten in my-pv-measurement gespeichert. 

So also 

1. docker-compose down
2. die .env editieren und my-pv-measurement in SENEC aendern
3. docker compose up -d 

alle historischen Daten sind wieder da.

So nun noch die Daten, die in der Zwischenzeit falsch gespeichert wurden umkopieren.

1. docker exec -ti solectrus_influxdb_1 bash
2. influx client config erstellen

   ```
 influx config create --config-name test \
  --host-url http://127.0.0.1:8086 \
  --org senec \
  --token <write token aus der .env> \
  --active
```

3. query zum kopieren 

` influx query 'from(bucket: "senec") |> range(start: -inf) |> filter(fn: (r) => r._measurement == "my-pv-measurement") |> map(fn: (r) => ({ r with _measurement: "SENEC" })) |> to(bucket: "senec")'`
    
    
dann sollte wieder alles passen

ledermann, 2024-08-02:
Die Ursache ist hier das veränderte Measurement. Bislang hat dein MQTT-Collector ins Measurement `SENEC` geschrieben, nun wird ins Measurement `my-pv-measurement` geschrieben. Ok, das hast du gerade selbst herausgefunden :)

Grundsätzlich noch zur neuen Sensor-Kofiguration:
https://github.com/solectrus/solectrus/wiki/Umstellung-auf-neue-Sensor%E2%80%90Konfiguration

Das Wichtigste ist, das Container-Log (von MQTT-Collector sowie Dashboard) zu beobachten. Da stehen Warnungen beim Start des Container, wenn eine neue Version mit alter Konfiguration betrieben wird. Und es stehen exakte Hinweise darin, wie man diese Warnung abstellen kann.

closed 2024-08-06 by ledermann