solectrus/solectrus #3329: after update to 0.15.0 no collector data

dellwo17, 2024-08-06:
### Description (in German or English)

Updating to 15-version and the erro occurs.
Trying with a complete new installation, but the same.

### What version of SOLECTRUS are you using?

v0.15.0

### What machine are you running SOLECTRUS on?

rasperry pi 

### If applicable (e.g. in case of error 500), please provide logs

```text
senec-collector-1  | Getting state names (language: de) from SENEC by parsing source code...
senec-collector-1  | Error getting data from SENEC at https://192.168.178.55: undefined method `[]' for nil
```

### Please provide your `docker-compose.yml` or `compose.yml` file

```yaml
services:
  dashboard:
    image: ghcr.io/solectrus/solectrus:latest
    depends_on:
      postgresql:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - postgresql
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - TZ
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - WEB_CONCURRENCY
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - DB_HOST=postgresql
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_POLL_INTERVAL
      - INFLUX_SENSOR_INVERTER_POWER
      - INFLUX_SENSOR_HOUSE_POWER
      - INFLUX_SENSOR_GRID_IMPORT_POWER
      - INFLUX_SENSOR_GRID_EXPORT_POWER
      - INFLUX_SENSOR_BATTERY_CHARGING_POWER
      - INFLUX_SENSOR_BATTERY_DISCHARGING_POWER
      - INFLUX_SENSOR_BATTERY_SOC
      - INFLUX_SENSOR_WALLBOX_POWER
      - INFLUX_SENSOR_CASE_TEMP
      - INFLUX_SENSOR_INVERTER_POWER_FORECAST
      - INFLUX_SENSOR_SYSTEM_STATUS
      - INFLUX_SENSOR_SYSTEM_STATUS_OK
      - INFLUX_SENSOR_GRID_EXPORT_LIMIT
[... 275 lines cut ...]
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=./influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=./postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=./redis
```

### Anything else?

LFP-Tausch kurz vorher, danach lief es aber noch.

ledermann, 2024-08-06:
Danke für den Hinweis. Du hast genau den passenden Ausschnitt aus den Logs gepostet:

```
senec-collector-1  | Getting state names (language: de) from SENEC by parsing source code...
senec-collector-1  | Error getting data from SENEC at https://192.168.178.55: undefined method `[]' for nil
```

Ursache ist also das Auslesen der Status-Codes. Ich habe da kürzlich einen Hinweis bekommen, dass SENEC nach dem LFP-Tausch etwas (per Firmware) geändert hat. 

Ich habe bereits einen Fix vornehmen können, der aber noch nicht vollständig getestet ist. Aber ich gehe davon aus, dass er bei dir helfen wird. Wechsele dazu bitte in der `docker-compose.yml` den SENEC-Collecetor von `latest` auf `develop`. Danach `docker compose pull && docker compose up -d`.

Bitte gib mir Bescheid, ob das hilft.

ledermann, 2024-08-06:
Sehr gut, danke für die Bestätigung. Ich werde in Kürze ein Update für den SENEC-Collector veröffentlichen, dann kannst du wieder auf `latest` wechseln.