solectrus/solectrus #3334: Beta Test Power Splitter alte -> neue Konfiguration

StevieC121176, 2024-08-08:
Ich habe mal testweise die MAPPING_0 Variable in der .env und docker-compose.yml ergänzt. Nun kommt ein neuer Fehler:

Docker-Compose.yml Auszug

```
    environment:
      - INFLUX_HOST
      - INFLUX_SCHEMA
      - INFLUX_PORT
      - INFLUX_TOKEN=${INFLUX_TOKEN_WRITE}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_MEASUREMENT=${INFLUX_MEASUREMENT_PV}
      - MQTT_HOST
      - MQTT_PORT
      - MQTT_SSL
      - MQTT_USERNAME
      - MQTT_PASSWORD
      - MQTT_TOPIC_HOUSE_POW
      - MQTT_TOPIC_GRID_POW
      - MQTT_TOPIC_BAT_CHARGE_CURRENT
      - MQTT_TOPIC_BAT_FUEL_CHARGE
      - MQTT_TOPIC_BAT_POWER
      - MQTT_TOPIC_BAT_VOLTAGE
      - MQTT_TOPIC_CASE_TEMP
      - MQTT_TOPIC_CURRENT_STATE
      - MQTT_TOPIC_MPP1_POWER
      - MQTT_TOPIC_MPP2_POWER
      - MQTT_TOPIC_MPP3_POWER
      - MQTT_TOPIC_INVERTER_POWER
      - MQTT_TOPIC_WALLBOX_CHARGE_POWER
      - MQTT_FLIP_BAT_POWER
      - MAPPING_0_TOPIC
      - MAPPING_0
      - MAPPING_0_MEASUREMENT
      - MAPPING_0_TYPE
```

.env Auszug

```
# MQTT Collector

# MQTT Broker credentials
MQTT_HOST=192.168.178.80
MQTT_PORT=1883
MQTT_SSL=false
MQTT_USERNAME=xxxxx
MQTT_PASSWORD=xxxxx

# MQTT topics (this is only an example, change to your needs!)
# This example is for using https://github.com/nobl/ioBroker.senec (which in most cases is not necessary, because there is a SENEC collector)
#MQTT_TOPIC_HOUSE_POW=evcc/site/homePower
MAPPING_0_TOPIC=evcc/site/homePower
MAPPING_0_FIELD=house_power
MAPPING_0_MEASUREMENT=my-pv-measurement
MAPPING_0_TYPE=integer
```
Fehler:
```

mqtt-collector-1      | 
mqtt-collector-1      | bundler: failed to load command: app.rb (app.rb)
mqtt-collector-1      | /mqtt-collector/lib/config.rb:192:in `block in validate_mappings!': Missing required keys: [:type, :topic, :measurement] (ArgumentError)
mqtt-collector-1      | 
mqtt-collector-1      |         raise ArgumentError, "Missing required keys: #{value.keys}"
mqtt-collector-1      |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
mqtt-collector-1      | 	from /mqtt-collector/lib/config.rb:178:in `each'
mqtt-collector-1      | 	from /mqtt-collector/lib/config.rb:178:in `validate_mappings!'
mqtt-collector-1      | 	from /mqtt-collector/lib/config.rb:62:in `initialize'
mqtt-collector-1      | 	from app.rb:22:in `new'
mqtt-collector-1      | 	from app.rb:22:in `<top (required)>'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/cli/exec.rb:58:in `load'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/cli/exec.rb:58:in `kernel_load'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/cli/exec.rb:23:in `run'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/cli.rb:455:in `exec'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/vendor/thor/lib/thor/command.rb:28:in `run'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/vendor/thor/lib/thor/invocation.rb:127:in `invoke_command'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/vendor/thor/lib/thor.rb:527:in `dispatch'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/cli.rb:35:in `dispatch'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/vendor/thor/lib/thor/base.rb:584:in `start'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/cli.rb:29:in `start'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/exe/bundle:28:in `block in <top (required)>'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/lib/bundler/friendly_errors.rb:117:in `with_friendly_errors'
mqtt-collector-1      | 	from /usr/local/bundle/gems/bundler-2.5.16/exe/bundle:20:in `<top (required)>'
mqtt-collector-1      | 	from /usr/local/bundle/bin/bundle:25:in `load'
mqtt-collector-1      | 	from /usr/local/bundle/bin/bundle:25:in `<main>'
forecast-collector-1  |   1: https://api.forecast.solar/estimate/51.798715/12.247921/8/40/9.72?damping=0,0&time=seconds ... OK
forecast-collector-1  |   Pushing forecast to InfluxDB ... OK
forecast-collector-1  |   Sleeping for 3600 seconds (until 2024-08-08 07:31:28 +0000) ...
mqtt-collector-1 exited with code 0

```

ledermann, 2024-08-08:
Ok, das klären wir. Danke fürs Auslagern in eine separate Diskussion.

Bei Docker-Compose ist es so, dass in `.env` eine globale Liste der Variablen geführt wird. In der `docker-compose.yml` wird dann für jeden Service festgelegt, auf welche Variable zugegriffen werden darf. Und da liegt bei dir der Fehler:

```yaml
- MAPPING_0_TOPIC
- MAPPING_0
- MAPPING_0_MEASUREMENT
- MAPPING_0_TYPE
```

Die Variable `MAPPING_0` gibt es nicht. Und es fehlt die Variable `MAPPING_0_FIELD`. Korrekt wäre es also so:

```yaml
- MAPPING_0_TOPIC
- MAPPING_0_FIELD
- MAPPING_0_MEASUREMENT
- MAPPING_0_TYPE
```

StevieC121176, 2024-08-08:
Oh, copy&Paste Fehler meinerseits. Danke für den Hinweis.

StevieC121176, 2024-08-08:
ich schließe das Issue, läuft jetzt problemlos.

closed 2024-08-08 by StevieC121176