solectrus/solectrus #3362: Opera Browser nicht (mehr) erlaubt

halmich, 2024-08-16:
### Description (in German or English)

Ich habe gesehen, dass die Änderung eingefügt worden ist.
[image]
Damit wird jedoch mein Browser nicht mehr erlaubt. 
Ich benutze https://play.google.com/store/apps/details?id=com.opera.browser in der neuesten Version, basierend auf Chromium 126.
Solectrus erkennt dies nicht und sperrt - zumindest in der "mobile" Ansicht. Wähle ich "Desktop Layout" gehts.

### What version of SOLECTRUS are you using?

power-splitter-176

### What machine are you running SOLECTRUS on?

Synology Nas

ledermann, 2024-08-16:
Hm, das ist natürlich nicht gut. Ich wollte alte Browser aussperren, bei denen die Darstellung von SOLECTRUS nicht optimal war. Ein aktueller Opera-Browser sollte natürlich nicht betroffen sein. Die Änderung habe ich gerade rückgängig gemacht, es ist also jetzt wieder wie vorher. Das aktualisierte Docker-Image ist in ein paar Minuten verfügbar.

Aus Interesse: Könntest du mir den exakten User-Agent-String nennen, den dein Opera-Browser hat? Also am einfachsten das hier aufrufen und den dort angezeigten String hier posten:
https://useragentstring.com/

closed 2024-08-16 by ledermann

halmich, 2024-08-16:
Danke!

Hier der User Agent String

Mozilla/5.0 (Linux; Android 10; K) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/126.0.0.0 Mobile Safari/537.36 OPR/83.0.0.0

ledermann, 2024-08-17:
Dein Browser meldet sich als "Opera 83", was der Grund für die Abweisung war. Seltsam, aber ich verfolge das nicht weiter.