solectrus/solectrus #3378: Einbau einer neuen NPU in der Senec Steuerung (Typ: Senec.Home V3 hybrid duo) --> keine Verbindung

mmuegge, 2024-08-22:
### Beschreibung (auf Deutsch oder Englisch)

Hallo, heute wurde bei uns die NPU (Senec) von EKD getauscht (Fehler NPU). Nach dem Neustart wurde eine neue IP-Adresse von unserem Router vergeben. Diese IP-Adresse habe ich in der Fritzbox wieder auf die alte IP-Adresse eingestellt (.83).
Seit dem Tausch der NPU kann Solectrus nicht mehr auf die aktuellen Daten zugreifen. Die Daten in der DB sind aber weiterhin sichtbar, nur akt. Daten sind nicht auslesbar. Die Solectrus-App zeigt "keine Verbindung".
Der Webserver von Senec ist über wieder über die 192.168.178.83 erreichbar.
Vor dem Tausch der NPU funktionierte alles zufriedenstellend.
Senec:
Revision MCU: 0827
Revision MCU-BL: 2307
Revision MPU-REGS: 09
Revision NPU-Image: 2011
Revision GUI: 0967

Was kann ich machen, um wieder die akt. Daten zu sehen

### Welche Version von SOLECTRUS verwendest du?

V0.15.1

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Hardware: Dell Wyse Proxmox Server , läuft im Linux-Container

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
Error getting data from SENEC at https://192.168.178.83: Failed to open TCP connection to 192.168.178.83:443 (Host is unreachable)
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
version: '3.7'

services:
  app:
    image: ghcr.io/solectrus/solectrus:latest
    labels:
      - 'com.centurylinklabs.watchtower.scope=solectrus'
    depends_on:
      db:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - db
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - DB_HOST=db
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_MEASUREMENT_PV
      - INFLUX_MEASUREMENT_FORECAST
      - INFLUX_POLL_INTERVAL
[... 361 lines cut ...]
      - FORECAST_3_DECLINATION
      - FORECAST_3_AZIMUTH
      - FORECAST_3_KWP
      - FORECAST_3_DAMPING_MORNING
      - FORECAST_3_DAMPING_EVENING
      - FORECAST_INTERVAL
      - FORECAST_SOLAR_APIKEY
    restart: always
    logging:
      options:
        max-size: '10m'
        max-file: '3'

  watchtower:
    image: containrrr/watchtower
    labels:
      - 'com.centurylinklabs.watchtower.scope=solectrus'
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
    command: --scope solectrus --cleanup
    logging:
      options:
        max-size: '10m'
        max-file: '3'
```

Zeppelin-PV, 2024-08-22:
[image]

mit genau denselben Revisionen ist unser Speicher mit dem http Protokoll gelaufen, nicht mit https
versuche es mal mit http den Speicher aufzurufen, müsste funktionieren

mmuegge, 2024-08-22:
Hallo Ernst, danke für den Hinweis. 
Mit http:/ip ist der Senec Webserver erreichbar.
Vielleicht wird ja das NPU Image ja noch von Senec auf einen neuen Stand gebracht.
Wie stelle ich diese URL in Solectrus ein?

Zeppelin-PV, 2024-08-22:
im Senec Collector in der .env
SENEC_SCHEMA=https
auf http stellen

btw
ich hatte genau daselbe, NPU Kartentausch am 1.3.24 - musste auch von https auf http umstellen, keine Probleme mehr,
jetzt nach dem LFP Tausch läuft dieselbe NPU mit neuer Software und nun auch wieder im https Modus,
... aber - just zu diesem Zeitpunkt waren die NPU-Fehler wieder da,

mmuegge, 2024-08-22:
Hallo Ernst, du machst uns ja Mut ;-)
Ich habe in der .env Datei unter der Sektion "Senec Collector" den Eintrag SENEC_SCHEMA=http eintragen (vorher stand https:).
Nach einem Reboot gibt es trotzdem über Solectrus "keine Verbindung" zu den akt. Daten.

Zeppelin-PV, 2024-08-22:
aus dem Solectrus Wiki

... eine manuelle Änderung in der compose-Datei (oder .env) nicht automatisch zu einem Update der Container. Dazu muss explizit der Befehl docker compose up -d ausgeführt werden.

mmuegge, 2024-08-22:
Hallo Ernst, danke für den Hinweis nach Ausführen von compose gibt es wieder eine Verbindung zu Solectrus.
Vielen Dank für deine Hilfe.