solectrus/solectrus #3389: Darstellung im Graph fehlerhaft

AlpenFlizzer, 2024-08-27:
### Beschreibung (auf Deutsch oder Englisch)

Ich versuche gerade eine neue Solectrus Instanz aufzusetzen. 
Diese Instanz bekommt die Inverter Daten via MQTT. Diese Daten werden sekünglich von einem Shelly Pro 3EM auf folgendes MQTT Topic gepublished: home/inverter/power, ein entsprechendes Sensor Mapping in der ENV und im MQTT Collector existiert.

Nach einer Weile (ca. 10 Sek) beginnt der Graph scheinbar zwischen jedem Messpunkt einmal auf die Nullinie zu hüpfen. Wenn ich den Graph neu lade (zB Timeframe hin und her wechseln oder F5 drücken oder Browser Tabs wechsle) wird er richtig gezeichnet, bis er nach einigen Momenten wieder mit den Null-Sprüngen beginnt.

[image]

Gibt es dafür einen Grund? Die MQTT Messages werden direkt am Shelly mittels Script gesendet - das Intervall kann ich beliebig wählen.

Offtopic: Der Shelly Connector ist ja nur für die Verbruachermessung vorgesehen wie es scheint, nicht für die Inverter Power Messung, daher der Umweg über den MQTT Collector und das Shelly Script...

### Welche Version von SOLECTRUS verwendest du?

0.15.1

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Synology NAS

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
services:
  dashboard:
    image: ghcr.io/solectrus/solectrus:latest
    depends_on:
      postgresql:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - postgresql
      - influxdb
      - redis
    ports:
      - 8080:3000
    environment:
      - TZ
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - WEB_CONCURRENCY
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - CO2_EMISSION_FACTOR
      - DB_HOST=postgresql
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_POLL_INTERVAL
      - INFLUX_SENSOR_INVERTER_POWER
      - INFLUX_SENSOR_HOUSE_POWER
      - INFLUX_SENSOR_GRID_IMPORT_POWER
      - INFLUX_SENSOR_GRID_EXPORT_POWER
      - INFLUX_SENSOR_BATTERY_CHARGING_POWER
      - INFLUX_SENSOR_BATTERY_DISCHARGING_POWER
      - INFLUX_SENSOR_BATTERY_SOC
      - INFLUX_SENSOR_WALLBOX_POWER
      - INFLUX_SENSOR_CASE_TEMP
      - INFLUX_SENSOR_INVERTER_POWER_FORECAST
      - INFLUX_SENSOR_SYSTEM_STATUS
[... 335 lines cut ...]
    powerData *= -1;

    // Ensure the value is non-negative
    if (powerData < 0) {
        powerData = 0;
    }

    return powerData;
}

// Function to publish the processed power data to the MQTT topic
function publishPowerData() {
    let powerValue = getTotalActivePower();
    
    // Convert the power value to a JSON string and publish to the MQTT topic
    MQTT.publish(mqttTopic, JSON.stringify(powerValue), 0, false);
    
    // Optional: Log the published value for debugging purposes
    // print("Published power value: ", powerValue);
}

// Timer to periodically publish the power data
let notifyTimer = Timer.set(publishInterval, true, function() {
    publishPowerData();
});

ledermann, 2024-08-27:
Der Grund für das "Hüpfen" ist vermutlich, dass immer wieder tatsächlich der Wert `0` geschrieben wird, verursacht vermutlich durch diese Zeile in deinem Script:

```javascript
// Ensure the value is non-negative
if (powerData < 0) {
  powerData = 0;
}
```

Dass dies erst nach einigen Sekunden sichtbar wird, liegt daran, dass bei JETZT zunächst eine gemittelte Darstellung der letzten Stunde erfolgt, dann aber im 5-Sekunden-Takt fortgeschrieben wird.

Insofern denke ich, dass eine Anpassung deines Scripts helfen sollte. Es darf keine `0` geschrieben werden, wenn nicht die Messung wirklich `0` ist, also in deinem Fall in der Nacht.

Zum Shelly-Collector noch: Ja, der ist eigentlich dafür gedacht, den Verbrauch zu messen. Aber wenn der Shelly bei dir so angeschlossen ist, dass er die Erzeugung misst, dann spricht eigentlich nichts dagegen, den Collector dennoch zu verwenden. Er schreibt in ein Measurement deiner Wahl (z.B. `inverter`) und dort ins Field `power`.  Du musst dann das Dashboard nur so konfigurieren, dass es die Erzeugung von dort gelesen wird. Beispiel:

Shelly-Collector:

```
INFLUX_MEASUREMENT=inverter
```

Dashboard:

```
INFLUX_SENSOR_INVERTER_POWER=inverter:power
```

AlpenFlizzer, 2024-08-27:
Habe es korrigiert und du hattest Recht. Danke für den Hinweis! 
Empfiehlst du eher den Shelly Connector für meinen Use Case?
Das Problem mit dem Shelly wäre doch, dass ich den Verbrauch des Wechselrichters (bzw. sind es mehrere Per-Panel Wechselrichter) dann in meinem PV Inverter Power Measurement bekomme oder täusche ich mich? Ich müsste jedenfalls garantieren, dass negative werte als 0 interpretiert werden.

ledermann, 2024-08-27:
Unabhängig von deiner elektrischen Installation arbeitet der Shelly-Collector so:

Im 5-Sekunden-Takt (konfigurierbar) holt der sich über die HTTP-Schnittstelle des Shelly (`http://[ip-adresse]/rpc/Shelly.GetStatus`) die aktuellen Messwerte (insbesondere `total_act_power`) und legt diese in InfluxDB ab - im selbst festzulegenden Measurement.

Wie dieser Messwert interpretiert wird (Verbrauch? Oder doch Erzeugung?) ist Sache des Dashboards. Mit dem Eintrag `INFLUX_SENSOR_INVERTER_POWER=inverter:power` erfolgt eine Behandlung als Erzeugung. 

Eine Spezialbehandlung von negativen Werten erfolgt aber nicht. Wie können die denn überhaupt auftreten? Ich könnte im Shelly-Collector eine Werte-Korrektur ergänzen, sodass immer positiv (oder 0) geschrieben wird.

Oder du bleibst beim MQTT-Collector.

AlpenFlizzer, 2024-08-27:
Ich habe einen Pro 3 EM an der Wurzel der PV Anlage (3 Phasen) und einen Pro 3 EM an der Wurzel beim Netzanschluss.
Wenn die PV erzeugt kommen positive Werte, wenn nicht, messe ich den Verbrauch der Wechselrichter und bekomme negative Werte. Zumindest theoretisch...

ledermann, 2024-08-27:
Ich vermute, dieses Issue ist hier erledigt, ich schließe es somit. Diskussion über den Umgang mit zwei Shellys und die Errechnung des Hausverbrauchs in #3392.

closed 2024-08-27 by ledermann