solectrus/solectrus #3390: Strompreise bzw. Einspeisevergütung scheinen nicht 0-safe zu sein

AlpenFlizzer, 2024-08-27:
### Beschreibung (auf Deutsch oder Englisch)

Da ich erst nach einigen Monaten nach Errichtung eine Einspeisevergütung erhalten habe, dachte ich, ich gebe eine Einspeisevergütung von 0€ ab Errichtungsdatum ein, welcher gültig sein sollte bis dem Zeitpunkt ab dem ich eine echte Einspeisevergütung erhalten habe.

Nachdem ich Gültigkeitsdatum (ab Errichtung) und Wert (0€) eingetragen hatte und auf Speichern geklickt habe kan sofort folgender Screen, welcher jedes mal kommt wenn ich die Einspeisevergütung Maske aufrufe:
[image]

Wie in den Logs ersichtlich macht mich " FloatDomainError (Computation results in 'Infinity')" skeptisch.

Ich müsste mich wohl in PG Container einloggen und den Eintrag jetzt manuell aus der DB entfernen...

### Welche Version von SOLECTRUS verwendest du?

0.15.1

### Auf welchem Gerät läuft bei dir SOLECTRUS?

x86 VM

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
solectrus-app                 | I, [2024-08-27T09:26:03.081565 #1]  INFO -- : [10.10.20.1] method=GET path=/ format=html controller=HomeController action=index status=302 allocations=620 duration=1.13 view=0.00 db=0.00 location=https://solectrus.alpenflizzer.com/inverter_power/now
solectrus-app                 | I, [2024-08-27T09:26:03.109624 #1]  INFO -- : [10.10.20.1] method=GET path=/inverter_power/now format=html controller=HomeController action=index status=200 allocations=19519 duration=22.34 view=21.84 db=0.00
solectrus-app                 | I, [2024-08-27T09:26:03.161786 #1]  INFO -- : [10.10.20.1] method=GET path=/stats/inverter_power/now format=html controller=StatsController action=index status=200 allocations=21399 duration=27.51 view=27.22 db=0.00
solectrus-app                 | I, [2024-08-27T09:26:03.177051 #1]  INFO -- : [10.10.20.1] method=GET path=/charts/inverter_power/now format=html controller=ChartsController action=index status=200 allocations=34907 duration=44.31 view=44.10 db=0.00
solectrus-app                 | I, [2024-08-27T09:26:04.948742 #1]  INFO -- : [10.10.20.1] method=GET path=/settings format=html controller=SettingsController action=edit status=200 allocations=16469 duration=67.01 view=66.62 db=0.00
solectrus-app                 | I, [2024-08-27T09:26:05.291029 #1]  INFO -- : [10.10.20.1] method=GET path=/settings format=html controller=SettingsController action=edit status=200 allocations=16458 duration=19.13 view=18.73 db=0.00
solectrus-app                 | I, [2024-08-27T09:26:07.421468 #1]  INFO -- : [10.10.20.1] method=GET path=/settings/prices/feed_in format=html controller=PricesController action=index status=500 allocations=11282 duration=17.46 view=0.00 db=4.95
solectrus-app                 | E, [2024-08-27T09:26:07.422965 #1] ERROR -- : [10.10.20.1]
solectrus-app                 | [10.10.20.1] ActionView::Template::Error (Computation results in 'Infinity'):
solectrus-app                 | [10.10.20.1]
solectrus-app                 | Causes:
solectrus-app                 | [10.10.20.1] FloatDomainError (Computation results in 'Infinity')
solectrus-app                 | [10.10.20.1]     1: = render PriceList::Component.new(prices:, name:)
solectrus-app                 | [10.10.20.1]
solectrus-app                 | [10.10.20.1] app/components/price_list/component.rb:23:in `round'
solectrus-app                 | [10.10.20.1] app/components/price_list/component.rb:23:in `initialize'
solectrus-app                 | [10.10.20.1] app/components/price_list/component.rb:16:in `new'
solectrus-app                 | [10.10.20.1] app/components/price_list/component.rb:16:in `relative_change'
solectrus-app                 | [10.10.20.1] app/components/price_list/component.html.slim:23:in `block in call'
solectrus-app                 | [10.10.20.1] app/components/price_list/component.html.slim:13:in `each'
solectrus-app                 | [10.10.20.1] app/components/price_list/component.html.slim:13:in `each_with_index'
solectrus-app                 | [10.10.20.1] app/components/price_list/component.html.slim:13:in `call'
solectrus-app                 | [10.10.20.1] app/views/prices/_list.html.slim:1
solectrus-app                 | [10.10.20.1] app/views/prices/index.html.slim:27
solectrus-app                 | [10.10.20.1] app/views/prices/index.html.slim:26
solectrus-app                 | [10.10.20.1] app/controllers/concerns/auto_locale.rb:14:in `switch_locale'
solectrus-app                 | I, [2024-08-27T09:26:07.436139 #1]  INFO -- : [10.10.20.1] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=10422 duration=12.73 view=12.44 db=0.00
```

ledermann, 2024-08-27:
Du hast völlig recht, danke für den Hinweis! Die Berechnung der Änderung von einem Preis zum nächsten (in Prozent) kommt nicht mit `€0,00` klar. Ich habe das gerade korrigiert. Wenn du auf `develop` wechselt, sollte es mit den erfassten Werten funktionieren, du musst also nicht manuell die Datenbank editieren.

Und ja, wenn nach Inbetriebnahme zunächst keine Einspeisevergütung gezahlt wurde, ist das Festlegen eines Preises von `€0,00` für diesen Zeitraum sinnvoll.

closed 2024-08-27 by ledermann

ledermann, 2024-08-27:
Je genau, im nächsten regulären Update ist der Fix drin. 

Der Fehler ist ein reiner Darstellungsfehler der Preisliste, der in deinem Fall nur in der Einspeisevergütung (`/settings/prices/feed_in`) auftritt. Wenn du die Strompreise aufrufst (`/settings/prices/electricity`), wird das funktionieren.

Manuell die DB zu editieren (Zeile mit `0` löschen oder auf `0.01` ändern) kann man machen, ist aber mühsam und lohnt eher nicht.