solectrus/solectrus #3396: Definieren zusätzlicher Verbraucher

ledermann, 2024-08-28:
Fortführung aus #2239:

Bislang kennt SOLECTRUS folgende Verbraucher:

- Hausverbrauch
- E-Auto (Wallbox)
- Wärmepumpe

Als Teil des Hausverbrauchs sind aber auch weitere denkbar, z.B. Heizstab, Klimaanlage, Kühltruhe, Pool, Dampfbad, Heimkino, Aufzug, Sessellift, Springbrunnen oder was man sonst so hat ;-) Wenn man davon separate Messwerte vorliegen hat - ermittelt z.B. über jeweils einen Shelly - sollten diese in SOLECTRUS integriert werden können.

Hierfür könnten zusätzliche Sensoren definiert werden, idealerweise mit folgenden Angaben:

- Bezeichnung
- Icon 
- Darstellung als separates Segment in der Strombilanz ja/nein

Verbraucher, die nicht als eigenes Segment dargestellt werden sollen, könnten im Tooltip des Hausverbrauchs aufgelistet werden.

Außerdem wäre die Darstellung als Diagramm sinnvoll.

AlpenFlizzer, 2024-08-28:
Definitiv +1 von mir. Wäre spätestens in der Testphase mit von der Partie.

Zeppelin-PV, 2024-08-28:
... wäre auch für mich interessant, wir haben einen Hiro Behinderten-Aufzug, und auch die Gastherme wäre nicht schlecht zu visualisieren. Shelly's für die beiden sind kein Problem. Nur das "how to ..."

ledermann, 2024-08-28:
@Zeppelin-PV Der erste Schritt wäre, einen Stromverbrauchsmesser an den Verbraucher zu klemmen. Wenn dieser an einer normalen Steckdose hängt, ist ein [Shelly Plus Plug S](https://www.shelly.com/de/products/shop/shelly-plus-plug-s-1) eine gute Wahl, bei großen Verbrauchern käme ein [Shelly Pro 3EM](https://www.shelly.com/de/products/shop/shelly-pro-3-em-120-a-1) für den Zählerschrank infrage. Der Plug ist in wenigen Minuten per WLAN in Betrieb genommen, für den anderen ist ein Elektriker vonnöten. Ein Shelly schreibt seine Messwerte erst mal in die Shelly-Cloud, mit der zugehörigen App sieht man den Verbrauch. Mit dem [Shelly-Collector](https://github.com/solectrus/shelly-collector) bekommt man die Messwerte in die InfluxDB von SOLECTRUS. Ich habe (neben dem Pro 3M für die Wärmepumpe) etliche Plugs im Haus verteilt - die warten nur darauf, in SOLECTRUS angezeigt zu werden.

Alles Weitere wird man sehen, wenn dieses Feature implementiert ist. Wann das sein wird, kann ich aber noch nicht absehen.

Zeppelin-PV, 2024-08-28:
könnte man mit diesen Verbrauchern (Aufzug etc.) auch den Power-Splitter aktivieren ? - oder macht das keinen Sinn ?

ledermann, 2024-08-28:
@Zeppelin-PV Der Power-Splitter ist primär dafür gedacht, große Stromverbraucher zu berechnen, bei denen es finanziell auch einen großen Unterschied macht, ob der Strom aus dem Netz oder vom Dach kommt. Theoretisch kann das auch für kleine Verbraucher berechnet werden. Aber der Nutzen ist eher gering: 

Nehmen wir einen Kühlschrank mit einem Jahresverbrauch von 250 kWh. Wenn ich den Strom zu 100 % aus dem Netz beziehe, kostet das etwa €70. Bei 100 % PV hingegen wären es €20 (wegen der entgangenen Einspeisevergütung). Zwischen diesen beiden Extremen muss es also liegen. Wenn der Power-Splitter mir nun ausrechnen, dass es tatsächlich €35 waren, ist der Erkenntnisgewinn nicht so groß.

Bei einer Wärmepumpe oder einem E-Auto, die sich beide jährlich 2.000 - 3.000 kWh reinziehen, wird es interessanter, die genauen Stromverbrauchskosten zu kennen. Hier kommt hinzu, dass man die Kosten mit Alternativen vergleichen möchte (Verbrenner-Auto oder Gas-Heizung).

Bei einer Wärmepumpe auch noch relevant: Die braucht man primär dann, wenn wenig Strom vom Dach kommt. Für mich war es interessant, durch den Power-Splitter zu erfahren, dass die Hälfte des Bedarfs aus PV-Strom gedeckt werden kann.

StevieC121176, 2024-08-30:
Ich hätte hier evtl. noch einen zusätzlichen speziellen Anwendungsfall. Mit dem neuesten Update des Smart #3 meiner Frau ist nun V2H (Vehicle2Home) möglich. Könnte man das evtl. auch z.Bsp. mit einem Shelly plus Plug S visualisieren. Hierzu könnte man die Wallboxanzeige grün färben und der Strom müsste vom E-Auto wieder abgezogen werden und nicht von Grid,PV oder Hausakku.
So könnte man schön visualisieren wieviel Strom man aus dem Auto (Monats,Jahres oder Gesamt) genutzt hat.

Zeppelin-PV, 2024-09-02:
... habe fertig !
Shelly Pro 1pm für Aufzug installiert, und Shelly Plug S für die Gastherme. Kann ich jetzt schon die beiden Geräte vorbereiten für die Option "zusätzlicher Verbraucher"? MQTT muß aktiviert werden- logisch, aber was muss da alles eingetragen werden damit Solectrus die Daten empfangen kann ?

spirit0607, 2024-11-12:
> Fortführung aus #2239:
>
> Bislang kennt SOLECTRUS folgende Verbraucher:
>
> * Hausverbrauch
> * E-Auto (Wallbox)
> * Wärmepumpe
>
> Als Teil des Hausverbrauchs sind aber auch weitere denkbar, z.B. Heizstab, [...]

Mir fällt noch "Sauna" als zusätzlicher Verbraucher ein. Hier wäre eine Darstellung sicherlich auch sinnvoll.

gereons, 2024-11-26:
Nice. Ich glaub ich geh direkt mal ein paar Black-Friday-Shellys kaufen :-)

Zeppelin-PV, 2024-11-26:
Ist, oder wäre es auch denkbar bzw. machbar, wenn einzelne der "Klein"-Verbraucher schon in HomeAssistant vorhanden sind, diese auf den Solectrus PI zu bringen ???
Das wäre natürlich eine grandiose Sache !!!

StevieC121176, 2024-11-26:
> Nice. Ich glaub ich geh direkt mal ein paar Black-Friday-Shellys kaufen :-)

🤭den gleichen Gedanken hatte ich auch 😜

ledermann, 2024-11-26:
@Zeppelin-PV Die Messwerte für SOLECTRUS müssen zwingend in der eigenen InfluxDB stehen. Wie sie dahin kommen, ist egal. Wenn du also Werte in HomeAssistant hast, könntest du sie dort per MQTT bereitstellen und dann über den MQTT-Collector abholen. 

Wenn es sich um Messwerte eines Shellys handelt, kannst du sie aber auch über den Shelly-Collector abholen, selbst wenn dieser bereits vom HomeAssistant abgefragt wird (oder umgekehrt der Shelly selbst per MQTT die Messwerte dorthin schickt).

Empfehlen würde ich den zweiten Weg, weil er einfacher ist und unabhängig von HomeAsistant funktioniert.

Bei weiteren Fragen zum Einsammeln von Messwerten bitte ein separates Diskussionsthema aufmachen, das führt uns sonst hier vom Thema weg.

ronzle, 2024-11-29:
Upps das habe ich wohl uebersehen, danke fuer den Newsletter. 
Interessant waeren auch Tasmota Geraete. Die kann man schoen ueber die API abfragen. 
Kann dir gern ein paar Beispiele (zur API und den Objects) geben

ledermann, 2024-11-29:
@ronzle Es gibt recht viele Geräte auf dem Markt, die zur Verbrauchsmessung eingesetzt werden können. Ich selbst bin bei Shelly gelandet, aber es gibt zweifellos gute Alternativen. Ich habe den [Shelly-Collector](https://github.com/solectrus/shelly-collector) gebaut, was aufwendiger war, als ursprünglich gedacht - es gibt bei Shelly verschiedene Generationen von Geräten und früher oder später hakt es an kleinen Details, um die man sich kümmern muss. 

Der Shelly-Collector funktioniert mittlerweile gut, aber die individuelle Programmierung von Kollektoren für jeden Hersteller kann ich nicht leisten. Ich empfehle daher, auf MQTT zu setzen. Dazu braucht man zwar einen MQTT-Broker im Haus, der aber möglicherweise für andere Zwecke bereits vorhanden ist. Und dann kann man mit dem [MQTT-Collector](https://github.com/solectrus/mqtt-collector) die Werte dort abholen und zu SOLECTRUS bringen.

gereons, 2024-11-30:
Frage zum Setup: wenn ich mit einem Shelly Plug S (Gen 3) so einen Verbraucher messen will, reicht es dann aus wenn ich die Standard-MQTT-Konfiguration auf dem Shelly aktiviere und die Daten dann via `mqtt-collector` mit zB 

```
MAPPING_2_TOPIC=shelly-plug-1/status/switch:0
MAPPING_2_MEASUREMENT=shellies
MAPPING_2_FIELD=dryer_power
MAPPING_2_TYPE=float
MAPPING_2_JSON_KEY=apower
```

speichere? Wenn ich das richtig sehe schickt der Shelly ja immer nur dann Daten wenn sich am Verbrauch irgendetwas ändert, und das war ja an anderen Stellen immer mal wieder ein Problem. Oder muss man irgendwie dafür sorgen dass in regelmässigen Abständen Daten gesendet werden?

gereons, 2024-11-30:
Prima, danke für die Antwort. Da bei mir eh schon ein mosquitto läuft ist mir die Lösung via mqtt sowieso lieber, weil ich dann auf meinem Raspi nicht noch X weitere Container starten muss, sondern alle Verbraucher zentral über den einen mqtt-collector steuern kann.

q2doc, 2024-12-02:
more-sensors sieht schon richtig schick aus 😉 - freue mich dann mal zu testen wenn Du verrätst wie der pr heißt...

ledermann, 2024-12-03:
@q2doc In der Demo sind die neuen Features zwar schon zu sehen, aber ich möchte noch ein paar Dinge korrigieren und verbessern, bevor ich den Beta-Test starte. Bitte noch ein wenig Geduld :-)

unclej84, 2024-12-04:
Hallo, sind in den 10 möglichen Verbrauchern die Wallbox und Wärmepumpe bereits enthalten oder wären die zusätzlich dazu?

ledermann, 2024-12-04:
@unclej84 10 zusätzliche Verbraucher, also zusätzlich zur Wallbox und Wärmepumpe. Die Zahl 10 ist außerdem nicht in Stein gemeißelt.

PR #3747 (2025-02-22): Selbstdefinierte Verbraucher
Resolves #3396 

## Darstellung auf neuer Seite

[image]

Der [Power-Splitter](https://github.com/solectrus/power-splitter/) berechnet den PV/Netz-Anteil für jeden Verbraucher, im Diagramm wird das durch gestapelte Säulen dargestellt.

## Neue Umgebungsvariablen

[image]

- bis zu ~zehn~ 20 Sensoren sind möglich
- einzelne Sensoren können vom Hausverbrauch ausgenommen werden und erscheinen dann in der Strombilanz (sinnvoll nur für Großverbraucher)

# Neue Einstellungsseite

[image]

## Noch offen

- [x] Ermöglichen, dass ein Verbraucher direkt in der Strombilanz au [... cut]
closed 2025-02-22 by ledermann