solectrus/solectrus #3399: Werte für stimmen in den verschiedenen Timeframe Tabs nicht überein

AlpenFlizzer, 2024-08-28:
### Beschreibung (auf Deutsch oder Englisch)

Ich habe eine neue Solectrus Instanz die heute den ersten Tag speichert.
In allen Timeframe Tabs werden verschiedene Werte angezeigt. Ich verstehe, dass es an unvollständigen Tagen wegen der Flux zu ungenauem Verhalten kommt, aber bedeutet das, dass ich jetzt warten muss, bis diese Woche, Monat, Jahr, zukünftige Jahre immer erst vollständig sein müssen, damit sie korrekt dargestellt werden? Oder heißt das bei meiner neuen Instanz, dass ab morgen die Anzeigen korrekt sind, wenn der erste Tag im Kasten ist?

Anbei ein Bild das das Problem zeigt:
[image]

Wie schon erwähnt ist das eine Solectrus Instanz die ich seit heute frisch füttere. Nach einigen Setup Problemen habe ich den Influx Container und die Daten im Bind-Mount gelöscht und mit der Speicherung neu begonnen. Kann das ein Probem sein?
Das Bild ist demonstrativ, ich habe ähnliche Abweichungen bei den anderen Sensoren auch.

Die Daten kommen alle 5 Sekunden via MQTT daher. Ich hatte bei meiner ersten Instanz ebenfalls komische Probleme mit den historischen Daten aus Solaredge, damals lag es daran, dass die Datenpunkte wohl nicht zu einem erwarteten Zeitintervall gepasst haben.
[export(2).csv](https://github.com/user-attachments/files/16787948/export.2.csv)

### Welche Version von SOLECTRUS verwendest du?

0.15.1

### Auf welchem Gerät läuft bei dir SOLECTRUS?

x86 VM

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
Auszug aus dem MQTT Collector Log:

# Message from 2024-08-28 20:53:26 +0200
  topic = home/inverter/power
  message = 0
  => pv:inverter_power = 0.0
# Message from 2024-08-28 20:53:26 +0200
  topic = home/energy/house_power
  message = 1249.128
  => pv:house_power = 1249.128
# Message from 2024-08-28 20:53:27 +0200
  topic = home/grid/power
  message = 1249.128
  => pv:grid_export_power = 0.0
  => pv:grid_import_power = 1249.128
# Message from 2024-08-28 20:53:31 +0200
  topic = home/inverter/power
  message = 0
  => pv:inverter_power = 0.0
# Message from 2024-08-28 20:53:31 +0200
  topic = home/energy/house_power
  message = 1252.56
  => pv:house_power = 1252.56
# Message from 2024-08-28 20:53:32 +0200
  topic = home/grid/power
  message = 1252.56
  => pv:grid_export_power = 0.0
  => pv:grid_import_power = 1252.56
# Message from 2024-08-28 20:53:36 +0200
  topic = home/inverter/power
  message = 0
  => pv:inverter_power = 0.0
# Message from 2024-08-28 20:53:36 +0200
  topic = home/grid/power
  message = 1245.478
  => pv:grid_export_power = 0.0
  => pv:grid_import_power = 1245.478
# Message from 2024-08-28 20:53:37 +0200
  topic = home/energy/house_power
  message = 1245.478
  => pv:house_power = 1245.478
# Message from 2024-08-28 20:53:41 +0200
  topic = home/inverter/power
  message = 0
  => pv:inverter_power = 0.0
# Message from 2024-08-28 20:53:41 +0200
  topic = home/energy/house_power
  message = 1248.203
  => pv:house_power = 1248.203
# Message from 2024-08-28 20:53:46 +0200
  topic = home/inverter/power
  message = 0
# Message from 2024-08-28 20:53:47 +0200
  topic = home/grid/power
  message = 1250.508
  => pv:grid_import_power = 1250.508

```
```

### Gibt es sonst noch etwas?

Ein Datenexport des heutigen (ersten) Tages liegt bei.
Der Export ist mit der Query erfolgt:
```
from(bucket: "solectrus")
  |> range(start: 2024-08-28T00:00:00.000Z, stop: 2024-08-28T23:59:59.999Z)
  |> filter(fn: (r) => r["_measurement"] == "pv")
  |> filter(fn: (r) => r["_field"] == "grid_export_power" or r["_field"] == "grid_import_power" or r["_field"] == "house_power" or r["_field"] == "inverter_power")
  |> aggregateWindow(every: 5s, fn: mean, createEmpty: false)
  |> yield(name: "mean")

```

AlpenFlizzer, 2024-08-29:
Nachdem der erste Tag nun abgeschlossen ist, präsentieren sich die Daten für diesen ersten Tag so:
[image]

Tag, Woche und Monat (jeweils der Tagesbalken im Diagramm) stimmen für diesen abgeschlossenen Tag nun überein. 

Als Referenz dann noch die Bildstrecke für den heutigen Tag:
[image]

ledermann, 2024-08-29:
Danke für die präzise Beschreibung. Es kommen hier verschiedene Probleme zusammen:

1) Veralteter Cache: Wenn du in die InfluxDB manuell eingegriffen hast (z.B. durch nachträgliches Löschen von Messwerten) dann passen die Daten im Redis-Cache nicht mehr dazu. Dieser muss dann gelöscht werden: `docker exec -it solectrus-redis-1 redis-cli FLUSHALL`

2) Ungünstige Caching-Strategie: Der Query-Cache merkt sich Abfrageergebnisse abhängig von der Länge des Betrachtungszeitraums für eine unterschiedliche lange Zeit. Jahres-Ergebnisse werden länger zwischengespeichert als z.B. Tageswerte. Wie ich nun sehe, wird dabei aber das `INSTALLATION_DATE` nicht berücksichtigt. Das Jahr 2024 ist bei deiner Installation noch gar nicht 241 Tage alt, sondern erst 1 Tag. Das korrigiere ich noch. Ein Löschen des Caches (siehe 1) wird dir momentan also nur kurzfristig helfen, weil das nächste Caching dann erneut zu lange erfolgt.

3) Ungünstige Flux-Queries: Offene Zeiträume (aktueller Tag, aktuelle Woche, aktueller Monat usw.) werden mit Integral berechnet (langsam, aber genau). Für abgeschlossene Zeiträume wird eine andere Methode benutzt (Summe der 1h-Durchschnitte), die schneller ist, bei fehlenden Messwerten aber ungenau wird. An deinen Beispielen wird ein Fehler offensichtlich: Der erste Tag ist zwar abgeschlossen (der 28.08.2024 liegt vollständig in der Vergangenheit), aber er ist dennoch unvollständig, weil erst ab Tagesmitte Messwerte vorhanden sind. 

4) Balken in Diagrammen für noch nicht abgeschlossene Zeiträume sind ungenau (siehe #2032). Fix steht noch aus.

Einzelwerte beziehen ihre Daten aus der gleichen Quelle wie die Strombilanz. Dortige Fehler wirken sich also auch hier aus. Ziel ist natürlich, dass es **immer** stimmt, auch wenn die Angaben für "Heute", "Dieser Monat" und "Dieses Jahr" unterschiedlich lange gecacht werden.

Für 2) und 3) arbeite ich gerade an einem Fix, der dann ins `develop`-Image fließt. Ich gebe dann nochmal Bescheid.

AlpenFlizzer, 2024-08-29:
Danke für die Hilfe.

ad 1) überlebt der Redis Cache auch ein docker compose down und up? Ich habe nur die Daten im Bind Mount von Influx gelöscht, um quasi eine saubere Messwerte DB zu haben. Die Daten im Redis Bind Mount habe ich nicht verändert, genauso wie die für PostgreSQL.

ad 2) und 3) werde ich das dev image dann ziehen und überprüfen.

Mit freundlichen Grüßen,
Sascha Böck

ledermann, 2024-08-29:
Ja, obwohl Redis eine reine In-Memory-Datenbank ist, überdauert diese ein down/up. Redis speichert nämlich beim Herunterfahren des Containers seinen aktuellen Stand in einem Dump, der beim nächsten Start wieder gelesen wird (sofern vorhanden). Dieser Dump liegt in der Datei `dump.rdb`, die auch die einzige Datei im Redis-Volume ist. Das Leeren des Caches kann also auf zwei Wegen erfolgen:

- Wenn Redis läuft: `docker exec -it solectrus-redis-1 redis-cli FLUSHALL`
- Wenn Redis nicht läuft: Volume leeren, also `dump.rdb` löschen

AlpenFlizzer, 2024-08-29:
Danke für den Fix. Ich habe develop gepullt und den Stack aktualisiert. Den Redis Cache habe ich gecleart.

Für die beiden Balkendarstellungen scheint sich das Problem gelöst zu haben, die Einzelwerte Seite stimmt auch! Für die Graphen besteht nun glaube ich noch dein erwähntes Problem mit den unfertigen Zeiträumen - hier gibt es differenzen:
[image]
Summe der Balken im Graph für
Woche: 47,206
Monat: 47,321
Jahr: 47,303
Gesamt: 47,296
Korrekt wäre: 45,8

ledermann, 2024-08-29:
OK, prima! Damit sind wir in diesem Issue vermutlich durch.

Der verbleibende Defekt bei den Balken ist #2032. Es sieht aber schlimmer aus, als es ist: Betroffen sind nur Balken, bei denen die Messwerte unvollständig sind, also nicht durchgängig von 0:00 bis 24:00 vorhanden sind. Da deine Instanz momentan noch ganz am Anfang ist, gibt es oft nur einen Balken. Und der ist dann leider ungenau. Eine Lösung dafür habe ich in #3328 skizziert.

closed 2024-08-29 by AlpenFlizzer