solectrus/solectrus #3401: Query-Partitionierung für bessere Performance und Genauigkeit der Strombilanz

ledermann, 2024-08-29:
### Beschreibung (auf Deutsch oder Englisch)

Zur Ermittlung der Energiewerte für die Strombilanz eines Zeitraums (Tag/Woche/Monat/Jahr/Gesamt) ist eine Flux-Query erforderlich, die die Daten aus InfluxDB abruft. Die Ausführung kann für InfluxDB recht aufwändig sein, die Abfrage kann also länger dauern. Um dies zu verhindern, erfolgt eine Kombination aus Query-Optimierung und Caching-Strategie.

Grundsätzlich kann die Energie (in kWh) von InfluxDB auf zwei unterschiedliche Weisen erfolgen:

- Als Integral: Sehr präzise, kommt auch mit lückenhaften Messwerten klar. Bei längeren Zeiträumen aber langsam.
- Als Summe von 1h-Durchschnitten. Sehr schnell, aber ungenau, insbesondere bei Lücken in den Messwerten. Mit Lücke ist gemeint, dass ein Zeitraum nicht vollständig mit Messwerten ausgefüllt ist, also irgendwo in der Mitte, aber auch am Anfang oder am Ende Messpunkte fehlen.

SOLECTRUS entscheidet daher, bei kurzen Zeiträumen (bis zu einem Monat) oder wenn der Zeitraum noch nicht abgeschlossen ist, die Integral-Methode anzuwenden. In allen anderen Fällen kommt die Methode der 1h-Durchschnitten zum Einsatz. In jedem Fall wird das Ergebnis zwischengespeichert (in Redis), sodass eine erneute Ausführung direkt aus dem Cache bedient werden kann, also gar nicht von InfluxDB ausgeführt werden muss.

Die funktioniert recht gut, hat aber dennoch Nachteile:

- Aktuelle Zeiträume (z.B. das laufende Jahr) sind umso langsamer, je mehr sich das Jahr dem Ende neigt (z.B. Ausführung im November). Durch eine im gleichen Maße steigende Caching-Dauer wird das etwas abgefedert, aber nicht gelöst.
- Längere Zeiträume der Vergangenheit (z.B. einzelne Jahre) können ungenau werden, wenn Lücken in den Messwerten enthalten sind. Dies kann z.B. vorkommen, wenn ein MQTT-Broker im Spiel ist, der nur Daten sendet, wenn sich diese auch verändert haben. Oder wenn aufgrund von Ausfallzeiten Messwerte fehlen.

Hier ein Lösungsansatz, der ausschließlich mit Integral-Abfragen arbeitet, ohne dass es zu Performance-Einbußen kommt:

1) Eine Query wird, wenn sie einen Zeitraum von bis zu einem Monat umfasst, grundsätzlich mit der Integral-Methode ausgeführt. Ist der Monat abgeschlossen, landet das Ergebnis unbefristet im Cache. Ist der Monat noch nicht abgeschlossen, wird auch gecacht, aber nur für wenige Minuten oder Stunden. Also alles so, wie bereits jetzt implementiert.

2) Ist der Zeitraum länger, wird dieser in kleinere Abschnitte von der Länge eines Monats aufgeteilt. Dies gilt für einzelne Jahre sowie für GESAMT. Für jeden Monat wird dann eine einzelne Query ausgeführt und die Teilergebnisse dann aufsummiert. Dies wird zu einem exakt gleichen Gesamtergebnis führen. Jede der einzelnen Query-Ergebnisse wird im Cache landen und kann für die nächste Abfrage mitbenutzt werden. Während der Ausführung der Queries sollte eine Fortschrittsanzeige erscheinen (statt Warteanimation). Der Cache soll so angelegt werden, dass er auch für Diagramme genutzt werden kann (siehe #3328).

Die Methode der 1h-Durchschnitte würde dann gar nicht mehr verwendet werden.

Die Implementierung müsste die bisherige Methode [PowerSum#build_query](https://github.com/solectrus/solectrus/blob/v0.15.1/app/services/power_sum.rb#L79-L99) ersetzen.

Es kann angenommen werden, dass der Aufwand für das Lesen aus dem Cache und das Aufsummieren der Teilsummen zu vernachlässigen ist. Falls sich herausstellen sollte, dass dem nicht so ist, könnten zusätzliche auch Jahresergebnisse gecacht werden.

Im Normalfall wird sich durch die Bedienung von SOLECTRUS der Cache immer weiter füllen, sodass auch bei Abfrage längerer Zeiträume ein Großteil aus dem Cache geholt werden kann. Nur der jeweils aktuelle Monat muss dann tatsächlich berechnet werden. Sobald der Monat abgeschlossen ist, wird er dauerhaft gecacht.

Es gibt ein theoretisches Worst-Case-Szenario: Ein User installiert SOLECTRUS, importiert anschließend Messwerte aus 20 Jahren und öffnet dann die Strombilanz für GESAMT. Es werden dann hintereinander 20 × 12 = 240 Integral-Abfragen hintereinander ausgeführt, da noch kein Cache vorliegt. Das wird dann länger dauern. Ein solcher Fall dürfte aber eher selten sein.

unclej84, 2024-09-01:
Hallo, nur so als Gedanke - sofern vorhandene Smartmeter oder Smartgateways ausgelesen und auch in InfluxDB geloggt werden, sind ja die Werte von Netzbezug und Einspeisung zu Beginn und Ende eines jeden Zeitraums vorhanden - das wäre dann jeweils nur eine difference Abfrage für den Zeitraum (für jeden Zeitraum oder dann als Summe von Tag -> Woche -> Monat -> Jahr -> Gesamt).

ledermann, 2024-09-01:
@unclej84 Danke fürs Mitdenken :) Das wäre tatsächlich eine Möglichkeit, würde aber erfordern, dass jeder Messwert auch in einer kumulierten Variante vorliegt. Bei den allermeisten ist das aber nicht der Fall. 

Nehmen wir beispielhaft die Erzeugung: Der Inverter liefert üblicherweise nur die aktuelle Leistung (in W), nicht jedoch die kumulierte Energie (in Wh). Ok, jetzt wäre es denkbar, dass diese vom SENEC-Collector, Shelly-Collector oder MQTT-Collector jeweils errechnet wird. Beim Start könnte der Collector den letzten kumulierten Wert aus der InfluxDB holen und dann bei jedem neuen Messwerteingang erneut rechnen, also das Integral bilden. Beide Werte müssten dann geschrieben werden.

Das würde allerdings das Datenvolumen verdoppeln, weil jeder Wert zweimal gespeichert werden müsste (als aktuelle Leistung und als kumulierte Energie). Das gefällt mir momentan eher nicht so gut. Aber ich behalte das mal im Hinterkopf.

ledermann, 2024-10-11:
Wie schon #3328 schließe ich auch dieses Issue zugunsten von #3533. Vorberechnete Tageswerte in PostgreSQL zu halten, erscheint mir als die deutlich bessere Lösung.

closed 2024-10-11 by ledermann