solectrus/solectrus #3404: Unterstützung von Vehicle2Home

ledermann, 2024-08-30:
> Ich hätte hier evtl. noch einen zusätzlichen speziellen Anwendungsfall. Mit dem neuesten Update des Smart #3 meiner Frau ist nun V2H (Vehicle2Home) möglich. Könnte man das evtl. auch z.Bsp. mit einem Shelly plus Plug S visualisieren. Hierzu könnte man die Wallboxanzeige grün färben und der Strom müsste vom E-Auto wieder abgezogen werden und nicht von Grid,PV oder Hausakku.
So könnte man schön visualisieren wieviel Strom man aus dem Auto (Monats,Jahres oder Gesamt) genutzt hat.

Ein Anwendungsbeispiel bei uns ist die Gartenpumpe. Die zieht ordentlich Strom und bei den heißen Temperaturen kann nur abends wenn die Sonne weg ist gewässert werden. Somit kann der Hausakku für den Hausverbrauch genutzt werden (gerade wenn die Nächte jetzt wieder länger werden ) und das Auto liefert den Strom für die Gartenpumpe.

_Originally posted by @StevieC121176 in https://github.com/solectrus/solectrus/issues/3396#issuecomment-2320534586_

ledermann, 2024-08-30:
@StevieC121176 Ich habe das mal in ein separates Issue ausgelagert. In #3396 geht es um zusätzliche Verbraucher, den Autoakku würde ich nicht als einen solchen ansehen.

Grundsätzlich ist das ein spannendes Thema, wobei ich persönlich keine Erfahrung damit habe und in absehbarer Zeit vermutlich auch kein Equipment, um das auszuprobieren. 

Zu klären wäre zunächst, wo die Messwerte herkommen. Da die Wallbox bidirektionales Laden unterstützen muss, wird sie wohl auch die Entladeleistung (Auto → Haus) liefern können. Es wäre in SOLECTRUS ein zusätzlicher Sensor nötig, der das behandelt. 

Die Darstellung ist dann vermutlich einfach: Man könnte die Wallbox auch links in der Strombilanz (also bei Herkunft) darstellen, wenn Strom vom Auto ins Haus fließt. 

Wenn das bei dir in Betrieb ist und du an die Messwerte herankommst, können wir das gerne weiter abklären.

StevieC121176, 2024-08-30:
Ich hatte mich verschrieben, was ich meine ist V2L (vehicle2Load). Das funktioniert mittels Adapter für die typ2 Ladebuchse des Autos. An fem Adapter kann man dann elektrische Geräte mittels Schuko Stecker betreiben max. 3,8 kw. 
mit einem Shelly plug s oder ähnliche. Zwischensteckern mit messfunktion, kann man so den entnommenen Strom messen. Bereitstellen könnte man den per mqtt oder in dem fall mit dem shelly collector. 
Darstellung auf der linken Seite bei Herkunft würde ich auch sagen.
Im Endeffekt wäre es bei einer bidirektionalen Wallbox nicht anders, nur das man den Strom dann im ganzen Haus zur Verfügung hätte, nicht nur bei einzelnen Steckergebundenen Geräten.
Eine Besonderheit gäbe es, man könnte den Strom halt nur messen wenn das Auto mit dem Adapter verbunden ist bzw mit der bidirektionalen Wallbox.

ledermann, 2024-08-30:
Ah, jetzt verstehe ich auch den Hinweis auf den Shelly Plug. Ja, das wäre tatsächlich kein großer Unterschied. 

Der Shelly-Collector würde sich anbieten. Ob es Probleme gibt, wenn dieser nicht non-stop erreichbar ist, kann ich noch nicht absehen. Der Shelly-Collector würde es alle 5sec versuchen und müsste mit Verbindungsfehlern zurechtkommen.

Ein Detail fällt mir aber noch ein: Wenn man erst das Auto belädt, danach aber Strom über V2L entnimmt, dann muss man spätestens dann, wenn man die Autokosten berechnet (z.B. im Rahmen von #2407), den entnommenen Strom wieder herausrechnen.

StevieC121176, 2024-08-30:
Ja guter Punkt. Aus den Autokisten herausrechnen und bei den Ersparnissen durch den PV Strom dazu addieren.

StevieC121176, 2024-09-14:
Ich hätte das Equipment jetzt einsatzbereit.
sobald man am Auto das Entladen freigibt verbindet sich der Shelly Plug s mit dem Elan und liefert Stromwerte. Per MQTT diesen bereitzustellen ist mit dem ioBroker nicht ideal, da dieser im Shelly Adapter nur 1x alle 5min den Wert abfragt. Ich kann leider nicht finden wo ich da etwas anderes einstellen könnte.

ledermann, 2024-09-14:
Ok, dann könntest du es mit dem [Shelly-Collector](https://github.com/solectrus/shelly-collector) versuchen. Dieser fragt den Verbrauch alle 5 Sekunden ab (konfigurierbar) und schreibt den Messwert in die InfluxDB. 

Wenn keine Verbindung möglich ist, schreibt der Collector eine Fehlermeldung ins Log und versucht es 60s später (nicht konfigurierbar) erneut.

StevieC121176, 2024-09-15:
Ich habe es mit einem Shelly Plug S probiert. Bekomme aber folgende Fehlermeldung:
`shelly-collector-1    | Error getting data from Shelly at http://192.168.178.102: 404`

Wird der Shelly Plug S evtl. nicht unterstützt?

ledermann, 2024-09-16:
@StevieC121176 Ja, so ist es leider. Der Shelly Plug ohne PLUS ist Gen1, und die hat eine andere Schnittstelle. 

Ich werde das in solectrus/shelly-collector#90 angehen, was aber etwas knifflig ohne Gerät ist.

StevieC121176, 2024-09-17:
ah ok, ich hab mir jetzt mal die Plus Variante bestellt.

StevieC121176, 2024-09-18:
So mit dem neuen Shelly plus funktioniert es ohne Probleme. Gehe ich recht in der Annahme das, wenn es einen Sensor in Solectrus für V2L gibt, ich diesen mit dem Shelly verknüpfen kann und er die bereits aufgezeichneten Werte dann darstellt?

ledermann, 2024-09-19:
Genau, die bessere Erweiterbarkeit ist der Gedanke hinter der Sensor-Konfiguration. SOLECTRUS wird im Laufe der Zeit zusätzliche Sensoren unterstützen.

Schon jetzt können weitere Messwerte in das InfluxDB-Bucket geschrieben werden, auch wenn SOLECTRUS sie aktuell noch nicht verarbeitet. Diese Werte können in einem beliebigen Measurement und Field abgelegt werden.

Sobald SOLECTRUS später beispielsweise den Sensor `V2L` erkennt, setzt man einfach die Variable `INFLUX_SENSOR_V2L` auf `$Measurement:$Field` und erhält sofort Zugriff auf alle aufgezeichneten Daten.

Ich mache das hier auch so: Ich protokolliere etliches Zeugs in meine InfluxDB (z.B. Wärmepumpen-Details, Außentemperatur, Auto-Fahrleistung usw.), auch wenn ich noch gar nicht weiß, ob und wie ich das zukünftig in SOLECTRUS darstellen werde. Ich habe dann aber die Möglichkeit, mit dem InfluxDB-Admin-UI Diagramme zu erstellen und ein Gefühl dafür zu entwickeln, was genau interessant sein könnte.

StevieC121176, 2024-09-19:
Das klingt super.  👍