solectrus/solectrus #3418: Aktuellen Strompreis anzeigen

bennysweb, 2024-09-03:
### Beschreibung (auf Deutsch oder Englisch)

Man könnte (ähnlich wie beim Auto-SoC) den aktuellen Strompreis im Dashboard anzeigen. Bei einem dynamischen Tarif (Tibber o.ä.) ändert sich der Wert stündlich und wird sowieso vom Tibber-Collector eingefangen (oder kann vom MQTT-Collector bereitgestellt werden), beim statischen Stromtarif ändert sich der Wert nicht und wird manuell vom User in den Einstellungen hinterlegt.

ledermann, 2024-09-05:
Ja, das ist sinnvoll - aber ich meine, dass erst #2150 gemacht werden sollte. Andernfalls würde das zu Missverständnissen führen, da SOLECTRUS derzeitig die Kosten ausschließlich mit den statischen Preisen (die in den Einstellungen erfasst werden) errechnet.

bennysweb, 2026-11-09:
Das sich ja nun viel getan hat in den letzten Monaten würde ich das Feature gern nochmal hoch holen. Vielleicht bekommst du es im Zuge des Umbaus mit dynamischen Stromtarifen umgesetzt?

ledermann, 2026-11-10:
@bennysweb Das Feature bleibt auf dem Zettel, aber aktuell kann ich es nicht umsetzen. Die Anzeige des aktuellen Strompreises hängt an #2150. Und der Umbau für dynamische Tarife ist eine größere Nummer: Er betrifft die gesamte Kostenberechnung und ist nichts, was ich nebenbei erledigen kann. Ich sehe das aber als immens wichtig an, keine Sorge!

Mein Fokus liegt momentan darauf, die [Version 1.4 fertigzustellen](https://github.com/orgs/solectrus/discussions/5966). Dieses Release wird ohnehin schon sehr umfangreich. Danach wende ich mich neuen Themen zu :)