solectrus/solectrus #3419: neuer Tooltip - historische Daten

Zeppelin-PV, 2024-09-03:
### Beschreibung (auf Deutsch oder Englisch)

mir ist folgendes aufgefallen
im Januar und Februar war unser Speicher in der Fernabschaltung, somit mise Autarkie 29% und viel Netzbezug 266kWh
habe heute den Monat Januar angeschaut und festgestellt, dass im Hausverbrauch trotzdem 100% aus PV angezeigt wird,
was eigentlich nicht sein kann
ist das Tooltip für historische Daten ungeeignet ?

[image]

### Welche Version von SOLECTRUS verwendest du?

0.16.0

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspberry 4

Zeppelin-PV, 2024-09-03:
das Tooltip fehlte
[image]7

ledermann, 2024-09-03:
Du hast völlig recht, das ist ein Fehler, der unter folgenden Umständen auftritt:

- Kein Power-Splitter installiert, also in InfluxDB keine Netzbezugs-Anteile für die Verbraucher vorhanden
- Sponsoring aktiv, also Zugriff auf den Power-Splitter-Daten erlaubt

Es wird dann fälschlicherweise angenommen, dass der PV-Anteil 100 % ist, was natürlich **falsch** ist. Auch die 0 € sind verkehrt.

Der Fehler ist bereits korrigiert (in `develop`). Der PV-Anteil sowie die Kosten werden damit nur angezeigt, wenn der Power-Splitter auch aktiv ist.

closed 2024-09-03 by ledermann

ledermann, 2024-09-09:
Korrektur steht jetzt mit Version 0.16.1 bereit.