solectrus/solectrus #3426: Ergänzung Datum bei ""Seit Inbetriebnahme"

spirit0607, 2024-09-04:
### Beschreibung (auf Deutsch oder Englisch)

Hallo Georg,

nur ein kleiner Vorschlag oder Idee: Wie wäre es auf der "Gesamt" Ansicht bei "Seit Inbetriebnahme" das tatsächliche Datum der Inbetriebnahme zu ergänzen. Dies ist ja in der .env Datei hinterlegt. Bei mir könnte es dann z.B. lauten: "Seit Inbetriebnahme am 20.06.2022"

[image]

ledermann, 2024-09-05:
Ja, gute Idee. Ich habe das gerade ergänzt, mit einer kleinen Variation: Der Zeitraum seit Inbetriebnahme wird relativ angegeben, also z.B. "Seit Inbetriebnahme vor fast 4 Jahren". Mittels Tooltip wird dann das genaue Datum angezeigt.

[Zu sehen in der Live-Demo](https://demo.solectrus.de/inverter_power/all), wird im nächsten Update enthalten sein.

closed 2024-09-05 by ledermann

ledermann, 2024-09-09:
Die Änderung steht jetzt mit Version 0.16.1 bereit.