solectrus/solectrus #3437: Power-Splitter Berechnung

Wolff925, 2024-09-06:
### Beschreibung (auf Deutsch oder Englisch)

Heute wurden in meinem SENEC die LFP Module eingebaut. Strom wurde aus dem Netz bezogen. Power-Splitter zeigt trotzdem 100% aus Photovoltaik 
[image]

### Welche Version von SOLECTRUS verwendest du?

v.0.16.0

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspberry Pi 4

ledermann, 2024-09-06:
Hast du den Power-Splitter denn installiert?
https://github.com/solectrus/power-splitter/wiki/Installation

Ansonsten sieht es nach #3419 aus.

Wolff925, 2024-09-06:
Hier meine Konfig. Habe noch keine WALLBOX und HEATPUMP
INFLUX_SENSOR_INVERTER_POWER=SENEC:inverter_power
INFLUX_SENSOR_HOUSE_POWER=SENEC:house_power
INFLUX_SENSOR_GRID_IMPORT_POWER=SENEC:grid_power_plus
INFLUX_SENSOR_GRID_EXPORT_POWER=SENEC:grid_power_minus
INFLUX_SENSOR_BATTERY_CHARGING_POWER=SENEC:bat_power_plus
INFLUX_SENSOR_BATTERY_DISCHARGING_POWER=SENEC:bat_power_minus
INFLUX_SENSOR_BATTERY_SOC=SENEC:bat_fuel_charge
_# INFLUX_SENSOR_WALLBOX_POWER=_
INFLUX_SENSOR_CASE_TEMP=SENEC:case_temp
INFLUX_SENSOR_SYSTEM_STATUS=SENEC:current_state
INFLUX_SENSOR_SYSTEM_STATUS_OK=SENEC:current_state_ok
INFLUX_SENSOR_GRID_EXPORT_LIMIT=SENEC:power_ratio
INFLUX_SENSOR_INVERTER_POWER_FORECAST=forecast:watt
_# INFLUX_SENSOR_HEATPUMP_POWER=_

Wir dann auch #3419 sein

ledermann, 2024-09-06:
Ja, ohne E-Auto und Wärmepumpe ergibt der Power-Splitter keinen Sinn, da es nichts aufzuteilen gibt. Die Autarkie des Hausverbrauchs entspricht dann der Gesamtautarkie, die bereits angezeigt wird.

Die Korrektur von #3419 entfernt nur die fehlerhafte 100%-Anzeige, sonst nichts.

closed 2024-09-06 by ledermann