solectrus/solectrus #3452: Update auf ver 0.16.1

TheBrain8788, 2024-09-11:
### Beschreibung (auf Deutsch oder Englisch)

Nach Update von 0.16.0 auf 0.16.1 folgende Fehlermeldung beim Starten des Containers:

### Welche Version von SOLECTRUS verwendest du?

v.0.16.0

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Pi 4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
frank@pi64:/home/pi/solectrus $ docker compose up -d
WARN[0000] /home/pi/solectrus/docker-compose.yml: the attribute `version` is obsolete, it will be ignored, please remove it to avoid potential confusion
[+] Running 2/0
 ⠋ Container fd4be9ece6e2_solectrus-redis-1     Recreate                                                           0.0s
 ✔ Container ed00bd18ed7f_solectrus-influxdb-1  Running                                                            0.0s
 ⠋ Container 3c7a1eb1556d_solectrus-db-1        Recreate                                                           0.0s
 ✔ Container solectrus-forecast-collector-1     Running                                                            0.0s
 ⠋ Container solectrus-senec-collector-1        Recreate                                                           0.0s
Error response from daemon: Conflict. The container name "/fd4be9ece6e2_solectrus-redis-1" is already in use by container "fd4be9ece6e2fc78af738bea2174f4a9541b4b036ff1d88bb411261c7104fc10". You have to remove (or rename) that container to be able to reuse that name.
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
version: '3.7'

services:
  app:
    image: ghcr.io/solectrus/solectrus:latest
    depends_on:
      - db
      - influxdb
      - redis
    links:
      - db
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - INSTALLATION_DATE
      - ELECTRICITY_PRICE
      - FEED_IN_TARIFF
      - ADMIN_PASSWORD
      - DB_HOST=db
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
 - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
    healthcheck:
      test: ['CMD-SHELL', 'nc -z 127.0.0.1 3000 || exit 1']
    restart: always

  influxdb:
    image: influxdb:2.6-alpine
    volumes:
      - ${INFLUX_VOLUME_PATH}:/var/lib/influxdb2
    environment:
      - DOCKER_INFLUXDB_INIT_MODE=setup
      - DOCKER_INFLUXDB_INIT_USERNAME=${INFLUX_USERNAME}
      - DOCKER_INFLUXDB_INIT_PASSWORD=${INFLUX_PASSWORD}
      - DOCKER_INFLUXDB_INIT_ORG=${INFLUX_ORG}
      - DOCKER_INFLUXDB_INIT_BUCKET=${INFLUX_BUCKET}
      - DOCKER_INFLUXDB_INIT_ADMIN_TOKEN=${INFLUX_ADMIN_TOKEN}
    command: influxd run --bolt-path /var/lib/influxdb2/influxd.bolt --engine-path /var/lib/influxdb2/engine --store di>    # Optional: Allow InfluxDB to be accessed from the outside.
    # ports:
    #   - 8086:8086
    restart: always
 db:
    image: postgres:15-alpine
    environment:
      - POSTGRES_PASSWORD=${POSTGRES_PASSWORD}
    volumes:
      - ${DB_VOLUME_PATH}:/var/lib/postgresql/data
    restart: always

  redis:
    image: redis:7-alpine
    volumes:
      - ${REDIS_VOLUME_PATH}:/data
    restart: always

  senec-collector:
    image: ghcr.io/solectrus/senec-collector:latest
    depends_on:
      - influxdb
    links:
      - influxdb
    environment:
      - SENEC_HOST
      - SENEC_INTERVAL
      - SENEC_SCHEMA
      - INFLUX_HOST
      - INFLUX_SCHEMA
 - INFLUX_PORT
      - INFLUX_TOKEN=${INFLUX_TOKEN_WRITE}
      - INFLUX_ORG
      - INFLUX_BUCKET
    restart: always

  forecast-collector:
    image: ghcr.io/solectrus/forecast-collector:latest
    depends_on:
      - influxdb
    links:
      - influxdb
    environment:
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_WRITE}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - FORECAST_LATITUDE
      - FORECAST_LONGITUDE
      - FORECAST_DECLINATION
      - FORECAST_AZIMUTH
      - FORECAST_KWP
      - FORECAST_INTERVAL
    restart: always
```

### Dein vollständiges `.env`

```text
##################################################################
# Dashboard application (the main part)
#
# Domain name or IP address of your host
APP_HOST=192.168.2.20:3000
#
# SSL redirect: Use "true" (which is the default) if you want to auto-redirect to https
FORCE_SSL=false
#
# Secret token to secure cookies, 128 chars long hexadecimal encoded string (don't use this example, make some random c># Currently there are no cookies in Soletrus, but this may change in the future
SECRET_KEY_BASE=xxx>#
# Date of commissioning of your photovoltaic system
INSTALLATION_DATE=2022-05-01
#
# Price you pay for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
ELECTRICITY_PRICE=0.3714
#
# Price you get for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
FEED_IN_TARIFF=0.0623
#
# Password for the PostgreSQL database, used by the app to comunicate internally with the database
POSTGRES_PASSWORD=my-secret-db-password
#
# Password to login as administrator, required to manage settings like historical prices
ADMIN_PASSWORD=xxxxxxx

##################################################################
# Influx database settings
#
# Influx host (to access from SOLECTRUS dashboard and collectors)
INFLUX_HOST=influxdb
INFLUX_SCHEMA=http
INFLUX_PORT=8086
#
# Credentials for the Influx database
INFLUX_ORG=solectrus
INFLUX_USERNAME=admin
INFLUX_PASSWORD=ExAmPl3PA55W0rD
INFLUX_ADMIN_TOKEN=geheimertoken
INFLUX_BUCKET=my-solectrus-bucket
#
# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=geheimertoken
INFLUX_TOKEN_READ=geheimertoken
#
# Volume path for storing the Influx data
INFLUX_VOLUME_PATH=/home/pi/solectrus/influxdb

##################################################################
# REDIS settings
#
REDIS_VOLUME_PATH=/home/pi/solectrus/redis

##################################################################
# PostgreSQL database settings
#
DB_VOLUME_PATH=/home/pi/solectrus/postgresql

##################################################################
# SENEC Collector
#
# Change this to your own SENEC IP address!
SENEC_HOST=192.168.2.118
SENEC_INTERVAL=5
SENEC_SCHEMA=https
##################################################################
# Solar forecasting with https://forecast.solar
# API docs: https://doc.forecast.solar/doku.php?id=api:estimate
#
# Latitude of the plant location
FORECAST_LATITUDE=xxxxxxxx
#
# Longitude of the plant location
FORECAST_LONGITUDE=xxxxxxxx
#
# Plane declination: 0 (horizontal) - 90 (vertical)
FORECAST_DECLINATION=45
#
# Plane azimuth: -180 ... 180 (-180 = north, -90 = east, 0 = south, 90 = west, 180 = north)
FORECAST_AZIMUTH=-20
#
# Installed modules power in kilowatt peak (kWp)
FORECAST_KWP=11.25
#
# Update interval in seconds, 900s = 15 minutes, the public (and free) API allows a minimum of 900 seconds
FORECAST_INTERVAL=900
```

ledermann, 2024-09-11:
Die relevante Fehlermeldung ist:

```
Error response from daemon: Conflict. 
The container name "/fd4be9ece6e2_solectrus-redis-1" is already in use by container "fd4be9ece6e2fc78af738bea2174f4a9541b4b036ff1d88bb411261c7104fc10". 
You have to remove (or rename) that container to be able to reuse that name.
````

Hast du vielleicht mehrere SOLECTRUS-Instanzen, die sich gegenseitig in die Query kommen?
Was ergibt `docker ps` ?

Versuch mal bitte das:

```
docker compose down
docker compose up -d
```

ledermann, 2024-09-11:
Was außerdem auffällt, ist der Einrückungsfehler in der `compose.yaml` bei der Zeile mit `REDIS_URL`.

TheBrain8788, 2024-09-11:
Fehler behoben, es liefen wohl 2 Solectrus Instanzen. Der Einrückungsfehler kam durchs Kopieren und Einfügen zustande.

Vielen Dank für die schnelle Hilfe.

closed 2024-09-11 by ledermann