solectrus/solectrus #3484: FR: Zeit bis zur Anzeige von "Keine Verbindung" konfigurierbar machen

gereons, 2024-09-22:
### Beschreibung (auf Deutsch oder Englisch)

Gefühlt seit dem Einbau meiner LFP-Akkus zeigt mein Solectrus oft "Keine Verbindung" im Status an. Das liegt ziemlich offensichtlich daran, dass der `senec-collector` die Daten nicht schnell genug liefern kann, siehe zB den folgenden Log-Ausschnitt von heute morgen:

```
Got record #15548 at 2024-09-22 10:08:23 +0200 within 9 ms, LADEN, Inverter 2173 W, House 500 W, Wallbox 0 W
Got record #15549 at 2024-09-22 10:08:28 +0200 within 11 ms, LADEN, Inverter 2192 W, House 507 W, Wallbox 0 W
Got record #15550 at 2024-09-22 10:08:41 +0200 within 8072 ms, LADEN, Inverter 2209 W, House 503 W, Wallbox 0 W
Got record #15551 at 2024-09-22 10:08:57 +0200 within 11123 ms, LADEN, Inverter 2221 W, House 469 W, Wallbox 0 W
Got record #15552 at 2024-09-22 10:09:08 +0200 within 6661 ms, LADEN, Inverter 2222 W, House 479 W, Wallbox 0 W
Got record #15553 at 2024-09-22 10:09:21 +0200 within 7158 ms, LADEN, Inverter 2195 W, House 468 W, Wallbox 0 W
Got record #15554 at 2024-09-22 10:10:19 +0200 within 52997 ms, LADEN, Inverter 2042 W, House 473 W, Wallbox 0 W
Got record #15555 at 2024-09-22 10:10:29 +0200 within 5573 ms, LADEN, Inverter 1847 W, House 462 W, Wallbox 0 W
Got record #15556 at 2024-09-22 10:10:41 +0200 within 7044 ms, LADEN, Inverter 1760 W, House 446 W, Wallbox 0 W
Got record #15557 at 2024-09-22 10:10:47 +0200 within 121 ms, LADEN, Inverter 1761 W, House 444 W, Wallbox 0 W
Got record #15558 at 2024-09-22 10:10:52 +0200 within 8 ms, LADEN, Inverter 1796 W, House 451 W, Wallbox 0 W
Got record #15559 at 2024-09-22 10:10:57 +0200 within 8 ms, LADEN, Inverter 1887 W, House 506 W, Wallbox 0 W
```

und das liegt wiederum ziemlich offensichtlich daran dass die Senec-Kiste immer mal wieder unfassbar lahm bei der Beantwortung der Requests des collectors ist - auf dem Raspi 5 auf dem das ganze läuft ist jedenfalls zu diesen Zeiten nichts anderes los was irgendwie diese Verzögerungen erklären könnte.

Wenn ich den Code richtig verstanden habe wird "Keine Verbindung" dann angezeigt wenn seit mehr als `INFLUX_POLL_INTERVAL * 2` Sekunden keine Daten mehr angekommen sind - und genau hier würde ich mir wünschen den Faktor 2 konfigurieren zu können.

ledermann, 2024-09-22:
Hm, das ist nicht so schön. Sicherlich wäre es denkbar, das Timeout höher setzen zu können, aber das löst das eigentliche Problem ja nicht. Denn wenn der SENEC-Speicher 50 Sekunden (!) zum Antworten benötigt, dann halt ich es für gerechtfertigt, dass SOLECTRUS ein Problem signalisiert.

Und ja, es ist genau, wie du schreibst: Wenn der jüngste Messwert älter ist als `INFLUX_POLL_INTERVAL` (d.h. älter als 10 Sekunden, das Intervall ist per Default 5s), dann wird "Keine Verbindung" angezeigt.

Ich denke, dass sowohl der Raspi als auch der SENEC-Collector unschuldig sind. Auch das Haus-Netzwerk ist vermutlich nicht die Ursache. Es dürfte allein am Speicher liegen, wenn dieser so lange zum Antworten benötigt. Bei mir sind es auch fast immer um die 8ms, was ich übrigens höllisch schnell finde. Das liegt insbesondere auch daran, dass der Collector mit `keep-alive` arbeitet, was u.a. den aufwändigen TLS-Handshake einspart.

Kannst du vielleicht (über die Speicher-Logs) herausfinden, was zwischen 10:08 und 10:10 auf dem Speicher los war? Ich tippe auf NPU-Probleme.

Noch ein Hinweis: Der SENEC-Collector speichert die Antwortzeit auch in InfluxDB ab (im Field `response_duration`). Du kannst also über das InfluxDB-Admin-UI herausfinden, wie oft solche Verzögerungen überhaupt auftreten.

Bei mir sahen die letzten 30 Tage so aus:
[image]

Das Problem tritt also auch bei mir manchmal auf - ganz ohne LFP-Akkus. Stichproben zeigen, dass NPU-Probleme die Ursache sind.

Mit jetzigem Kenntnisstand neige ich dazu, in SOLECTRUS und im Collector gar nichts zu verändern. Die Probleme liegen im Speicher und werden von SOLECTRUS korrekt angezeigt. Man kann nur hoffen, dass SENEC die NPU-Probleme irgendwann in Griff bekommt.

ledermann, 2024-09-22:
Da hat sich - vermutlich mit der neuen Firmware `0828` - ziemlich viel getan. Bei mir (ohne LFP-Akkus und mit Firmware `0826`) gibt es das Stichwort "MQTT" nirgendwo in den Logs. Und der Connect findet (laut meinem Pi-Hole) zu `machine.mein-senec.de` statt.

Das Diagramm sieht tatsächlich erschreckend aus. Würde mich interessieren, ob sich das durch ein Reboot ändert. Und ob andere User, die auch bereits die neuen Akkus haben, von ähnlichen Verbindungsproblemen berichten.

Wenn das nämlich normal wäre, würde mich das sehr ärgern. Auch wenn sich die Warnung "Keine Verbindung" unterdrücken ließe, wäre von einer Live-Anzeige nicht mehr viel übrig.

gereons, 2024-09-22:
Also der Reboot hat auf jeden Fall schon mal kurzfristig geholfen, das hier sind die Daten der letzten Stunde:

[image]

Reboot war um 16:19, seit dem nur vereinzelte Peaks um 250ms, alles andere liegt unter 20ms. Senec-Logs gibt's leider keine, weil zum ersten Mal das Logfile vollgelaufen ist... :-(

Dank dem Logging von `response_duration` werd ich das aber jetzt mal engmaschig beobachten können, vielleicht tritt das Phänomen ja doch nochmal auf.

Und BTW gab's nach dem Boot dann auch noch ein paar DNS-Queries, ich habe `machine.mein-senec.de`, `ntp3.senecops.com` und `ns.eg-services.net` in den pihole-Logs gefunden. Letzteres klingt als könnte es ein privater Nameserver sein, aber auf normale Anfagen mit `dig` antwortet er nicht.

Jedenfalls hab ich jetzt erstmal eher das Gefühl dass das ein wie auch immer getriggerter Ausreisser meiner Anlage war und es hier ausser meinem Monitoring erstmal nix weiter zu tun gibt.

closed 2024-09-22 by gereons