solectrus/solectrus #3494: Forecast nachträglich anpassen

NonBlackstation, 2024-09-25:
### Beschreibung (auf Deutsch oder Englisch)

Hallo Gemeinde, 
ich hab irgendwie nen Blindflug. Ich habe soweit schon durchsucht, doch nichts gefunden.
Den Solectrus habe ich auf einer Synology im Docker laufen. Gestern ein Update von 0.14.... auf die neuste Version 0.17.0 über ssh manuell durchgeführt und dann erst die News mal durchgelesen...*ja haut mich*

Nun hatte ich auf Grund der Infos aus den News das Wollen, den Angaben zum Forecast anzupassen. Aktuell habe ich nur eine Fläche angegeben und es sind aber zwei. Auch die Ausrichtung habe ich nicht so genau eingetragen bei der Erstinstallation.
Dazu sollte ich erstmal den Container stoppen und dann die .env und die yml bearbeiten oder austauschen?
Würdet Ihr mir bitte helfen? Weil so in dem Thema bin ich dann doch nicht
Vielen Dank

### Welche Version von SOLECTRUS verwendest du?

0.17.0

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Synology DS718+

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
hier auf Arbeit kein Zugriff auch die compose.yml
```

### Dein vollständiges `.env`

```text
hier auf Arbeit kein Zugriff auch die .env
```

ledermann, 2024-09-25:
Der [Forecast-Collector](https://github.com/solectrus/forecast-collector) unterstützt grundsätzlich zwei verschiedene Anbieter: [solcast.com](https://www.solcast.com/) und [forecast.solar](https://forecast.solar/). Du musst dich für einen entscheiden. 

Beispiele zur Konfiguration:

- solcast.com: https://github.com/solectrus/forecast-collector/blob/develop/.env.solcast.example
- forecast.solar: https://github.com/solectrus/forecast-collector/blob/develop/.env.example

Beim Konfigurieren darauf achten, ENV-Variablen auch in der compose-Datei aufzuführen. Nach jeder Änderung in `.env` oder `docker-compose.yml` muss `docker-compose up -d` ausgeführt werden, um die Änderungen zu übernehmen. Hilfreich ist es, die Logs der Container im Auge zu behalten.

Bei anhaltenden Problemen bitte `docker-compose.yml` **und** `.env` hier posten.

NonBlackstation, 2024-09-25:
Guten Abend Herr Ledermann, 

ich habe die beiden Dateien nun nachgetragen.

Wenn ich es jetzt richtig interpretiere, die beiden Dateien ersetzen (siehe Screenshot / Ordnerstruktur NAS)
[image]
und dann `docker-compose up-d `durchführen?

ledermann, 2024-09-25:
Bitte unbedingt beachten: Der Konfigurator ist für **Neuinstallationen** gedacht, nicht zum Bearbeiten/Ergänzen bestehender Installationen. Der Grund ist, dass die generierte Konfiguration höchstwahrscheinlich andere Measurements/Fields für InfluxDB definiert, d.h. man verliert seine historischen Daten.

Es ist aber möglich, die Konfiguration über den Konfigurator zu erstellen und dann die Zeilen herauszupicken (aus `docker-compose.yml` und `.env`), die den Forecast-Collector betreffen. Das heißt, man übernimmt NICHT die vollständigen Dateien des Konfigurators.

NonBlackstation, 2024-10-01:
Guten Abend Zusammen, 

Entschuldigung, das es eine kleine Weile gedauert hat, bis ich mich hier weider melde. Ich habe jetzt mal die Daten rein nur vom Forcast aus dem Konfigurator in die bestehenden `.env` und `docker-compose.yml` eingefügt.

NonBlackstation, 2024-10-01:
Also irgendwie kann ich keine Files anhängen :(

Dann also hier reinkopieren:
---> .env

##################################################################
# Dashboard application (the main part)
#
# Domain name or IP address of your host
APP_HOST=192.168.178.XXX
#
# SSL redirect: Use "true" (which is the default) if you want to auto-redirect to https
FORCE_SSL=true
#
# Secret token to secure cookies, 128 chars long hexadecimal encoded string (don't use this example, make some random changes!)
# Currently there are no cookies in Soletrus, but this may change in the future
SECRET_KEY_BASE=f60debe97dcb73280a2cc83668fd60e8d0e8e48269036a7bce980ee53cfb312e377989a750b8c945a5f69b041289ecb4e2d9e40641b81257c65ac2d43e3c837f
#
# Date of commissioning of your photovoltaic system
INSTALLATION_DATE=2020-11-30
#
# Price you pay for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
ELECTRICITY_PRICE=0.xx
#
# Price you get for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
FEED_IN_TARIFF=0.xxxxx
#
# Password for the PostgreSQL database, used by the app to comunicate internally with the database
POSTGRES_PASSWORD=existiert
#
# Password to login as administrator, required to manage settings like historical prices
ADMIN_PASSWORD=existiert

##################################################################
# Influx database settings
#
# Influx host (to access from SOLECTRUS dashboard and collectors)
INFLUX_HOST=influxdb
INFLUX_SCHEMA=http
INFLUX_PORT=8086
#
# Credentials for the Influx database
INFLUX_ORG=solectrus
INFLUX_USERNAME=vorhanden
INFLUX_PASSWORD=existiert
INFLUX_ADMIN_TOKEN=my-super-secret-admin-token
INFLUX_BUCKET=my-solectrus-bucket
#
# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token
#
# Volume path for storing the Influx data
INFLUX_VOLUME_PATH=/volume1/docker/solectrus/influxdb

##################################################################
# REDIS settings
#
REDIS_VOLUME_PATH=/volume1/docker/solectrus/redis

##################################################################
# PostgreSQL database settings
#
DB_VOLUME_PATH=/volume1/docker/solectrus/postgresql

##################################################################
# SENEC Collector
[... 103 lines cut ...]

# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=/volume1/docker/solectrus/influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=/volume1/docker/solectrus/postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=/volume1/docker/solectrus/redis

# URL of the Redis database
REDIS_URL=redis://redis:6379/1

NonBlackstation, 2024-10-01:
Irgendwie kann ich keine Files mehr anhängen :(

also hier reinkopieren : die .env
`##################################################################
# Dashboard application (the main part)
#
# Domain name or IP address of your host
APP_HOST=192.168.178.XXX
#
# SSL redirect: Use "true" (which is the default) if you want to auto-redirect to https
FORCE_SSL=true
#
# Secret token to secure cookies, 128 chars long hexadecimal encoded string (don't use this example, make some random changes!)
# Currently there are no cookies in Soletrus, but this may change in the future
SECRET_KEY_BASE=f60debe97dcb73280a2cc83668fd60e8d0e8e48269036a7bce980ee53cfb312e377989a750b8c945a5f69b041289ecb4e2d9e40641b81257c65ac2d43e3c837f
#
# Date of commissioning of your photovoltaic system
INSTALLATION_DATE=2020-11-30
#
# Price you pay for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
ELECTRICITY_PRICE=0.xx
#
# Price you get for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
FEED_IN_TARIFF=0.xxxxx
#
# Password for the PostgreSQL database, used by the app to comunicate internally with the database
POSTGRES_PASSWORD=existiert
#
# Password to login as administrator, required to manage settings like historical prices
ADMIN_PASSWORD=existiert

##################################################################
# Influx database settings
#
# Influx host (to access from SOLECTRUS dashboard and collectors)
INFLUX_HOST=influxdb
INFLUX_SCHEMA=http
INFLUX_PORT=8086
#
# Credentials for the Influx database
INFLUX_ORG=solectrus
INFLUX_USERNAME=vorhanden
INFLUX_PASSWORD=existiert
INFLUX_ADMIN_TOKEN=my-super-secret-admin-token
INFLUX_BUCKET=my-solectrus-bucket
#
# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token
#
# Volume path for storing the Influx data
INFLUX_VOLUME_PATH=/volume1/docker/solectrus/influxdb

##################################################################
# REDIS settings
#
REDIS_VOLUME_PATH=/volume1/docker/solectrus/redis

##################################################################
# PostgreSQL database settings
#
DB_VOLUME_PATH=/volume1/docker/solectrus/postgresql

##################################################################
# SENEC Collector
#
# Change this to your own SENEC IP address!
[... 101 lines cut ...]

# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=/volume1/docker/solectrus/influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=/volume1/docker/solectrus/postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=/volume1/docker/solectrus/redis

# URL of the Redis database
REDIS_URL=redis://redis:6379/1`

NonBlackstation, 2024-10-01:
Ähm vergessen die Dateien mit anzuhängen...

[env_neu.txt](https://github.com/user-attachments/files/17213705/env_neu.txt)
[docker-compose.yml_neu.txt](https://github.com/user-attachments/files/17213706/docker-compose.yml_neu.txt)

ledermann, 2024-10-01:
Deine Konfiguration ist ziemlich durcheinander geraten. Ich habe daher der Einfachheit eine neue erstellt (mit dem Konfigurator) und dann manuell angepasst. 

Bitte selber nochmal durchgehen, da müssen noch Passwörter und IP-Adressen eingesetzt werden.

[NonBlackStation-neu.zip](https://github.com/user-attachments/files/17214183/NonBlackStation-neu.zip)

closed 2025-01-22 by ledermann