solectrus/solectrus #3495: Feature Request: Tages-/Monats-/Jahresauswahl per Klick

bennysweb, 2024-09-25:
### Beschreibung (auf Deutsch oder Englisch)

In Anlehnung an #1931 könnte man als ersten Schritt die Diagramme klickbar machen. Wenn ich also z.B. in der Wochenansicht auf den Balken vom Dienstag klicke, könnte SOLECTRUS auf die Tagesanzeige von diesem Dienstag springen. Ob das auch bei der mobilen Version hilfreich wäre, müsste sich zeigen. Am Desktop aber wäre das ein echter Mehrwert.

ledermann, 2024-09-25:
Ja, sehr sinnvoll. Und relativ leicht zu machen, daher habe ich das heute implementiert :)

Verfügbar in `develop`, zu sehen in der [Live-Demo](https://demo.solectrus.de/inverter_power/2024-W38).

Es funktioniert derzeitig aber **nur** auf dem Desktop, also nicht auf Mobile. Dort kollidiert es mit den Tooltips und ich habe noch keine gute Lösung dafür gefunden. Sinnvoll wäre vermutlich: Erster Klick zeigt Tooltip, zweiter Klick führt dann den Link aus. Das habe aber noch nicht hinbekommen.

gereons, 2024-09-25:
Zwei Taps wären sehr ungewöhnlich... vielleicht Tap um den Link zu aktivieren, Longpress um den Tooltip zu sehen?

bennysweb, 2024-09-25:
Sehr gut, das ging ja wirklich fix :-). Mobile wäre eher andersrum sinnvoll denke ich - also short press aktiviert Link, long press zeigt Tooltip. Oder was meint ihr?

ledermann, 2024-09-25:
Danke für die allseitigen Vorschläge! Die Umsetzung ist allerdings erstaunlich schwierig. Ich habe bislang keine Möglichkeit gefunden, das Tooltip-Verhalten von Chart.js abzuändern. Der Tooltip erscheint (auf Touch-Displays) beim ersten Klick. Und das wäre gleichzeitig mit dem Link. Man würde also kurz den Tooltip aufblitzen sehen, bevor das Diagramm dann wechselt.

Solange mir nichts Besseres einfällt, muss der Balken-Link auf Touch-Displays entfallen :-(

bennysweb, 2024-09-25:
Und wenn man statt des Balkens die Zahl auf der x-Achse klickbar macht um den Link zu öffnen?

ledermann, 2024-09-26:
@bennysweb Ein Klicken auf die Zahl bzw. den Monat finde ich auch nicht ideal. Zum einen ist die Zahl sehr klein und kaum zu treffen, zum anderen wäre das eine große Abweichung zum Verhalten auf dem Desktop, wo das Klicken auf den Balken ja das Sinnvollste ist.

Ich habe es daher nun so gemacht: Auf Touch-Geräten führt der **erste** Klick auf einen Balken zur Anzeige des Tooltips. Ein zweiter Klick (auf den gleichen Balken) führt dann den Link aus. 

Verfügbar in `develop` und zu sehen in der [Live-Demo](https://demo.solectrus.de/inverter_power/2024-W39).

Ja, das ist nicht sonderlich intuitiv, aber zumindest funktioniert es so. Eine andere Lösung (mit Longtap) habe ich leider nicht hinbekommen. Dazu müsste vermutlich zu tief in die Chart.js-Library eingegriffen werden.

closed 2024-09-26 by ledermann

bennysweb, 2024-09-26:
Das ist super so, für mich ein echter Mehrwert 👍🏻