solectrus/solectrus #3517: E-Auto: Verbrauch und Stromkosten pro Kilometer

ledermann, 2024-10-04:
Wer ein E-Auto hat, möchte wissen, wie viel Strom für einen gefahrenen Kilometer verbraucht wird, welche Kosten dafür entstehen und wie sich das im Laufe der Zeit verändert. 

Das E-Auto selbst zeigt den Verbrauch üblicherweise an, aber der Wert ist oft ungenau (enthält z.B. keine Ladeverluste) und wird auch nicht historisch gespeichert.

## Benötigte Informationen

Um dies berechnen zu können, werden folgende Informationen benötigt:

- Gefahrene **Kilometer** (aus der API des E-Autos)
- Geladene **Energie** für das E-Auto (in kWh). Aus der eigenen Wallbox oder extern geladen (z.B. an Ladesäulen)
- Entstandene **Kosten** der Energie (in EUR). Bei der eigenen Wallbox über den Power-Splitter berechnet, bei externen Ladevorgängen manuell einzugeben.

## Ausgabe

Bei der Wahl eines Betrachtungszeitraums (Tag, Woche, Monat, Jahr, Gesamt) soll SOLECTRUS die folgenden Werte berechnen und anzeigen:

- Fahrleistung (in km)
- Verbrauch (in kWh pro 100km oder Wh/km)
- Kosten (in EUR pro km und EUR gesamt)

## TODO

- [ ] Zusätzlicher Sensor für die Fahrleistung (km) des E-Autos
- [ ] Eingabemöglichkeit für externe Ladevorgänge: Datum, Energie in kWh, Kosten in EUR, Ort
- [ ] Eingabemöglichkeit für Ladevorgänge über die Wallbox, die nicht zu berücksichtigen sind, weil die Wallbox z.B. von Gästen benutzt wurde: Datum, Energie in kWh, Notiz 
- [ ] Darstellung der Fahrleistung in km für den gewählten Zeitraum
- [ ] Darstellung des Verbrauchs in kWh pro 100km für den gewählten Zeitraum
- [ ] Darstellung der Kosten in EUR (pro km und gesamt) für den gewählten Zeitraum
- [ ] Darstellung des Verbrauchs als Diagramm (Säulen mit kWh pro 100km)
- [ ] Darstellung der Kosten als Diagramm (Säulen mit EUR pro km bzw. EUR gesamt)

## Ausbaustufen

Wer die Fahrleistung nicht automatisiert abrufen kann, sollte die Möglichkeit einer manuellen Erfassung haben. Hier dürfte es genügen, einmal im Monat den Kilometerstand abzulesen und einzugeben.

ledermann, 2026-10-04:
An diesem Thema bin ich schon eine ganze Weile dran. Die wesentliche Dinge sind bereits gelöst:

- Gegenüberstellung von LADEN (kWh) und FAHREN (km). Dadurch lässt sich auch bei kürzen Zeiträumen (z.B. Tag oder Woche) ermitteln, was das Fahren gekostet hat. Zur Berechnung wird ein Zeitfenster +/- 14 Tage herangezogen.
- Laden über die eigene Wallbox und auswärts (an Ladesäulen etc.)
- Automatische Erkennung von Ladevorgängen an der eigenen Wallbox. Beladungen fremder Autos ("Gastladung") können gekennzeichnet werden, das wird dann in der Kostenermittlung heraus gerechnet
- Berechnung von Verbrauch und Kosten pro 100 km und gesamt
- Erkennung von Standorten und Fahrtzielen 

Eingeführt werden dafür folgende neue Sensoren:

```env
# Fahrleistung in km
INFLUX_SENSOR_CAR_MILEAGE_1=car:mileage

# Aktuelle Reichweite in km
INFLUX_SENSOR_CAR_RANGE_1=car:range

# Auto zum Laden verbunden ja/nein
INFLUX_SENSOR_CAR_CONNECTED_1=car:connected

# Geo-Koordinaten für den aktuellen Standort
INFLUX_SENSOR_CAR_LATITUDE_1=car:latitude
INFLUX_SENSOR_CAR_LONGITUDE_1=car:longitude

# Ladestand in Prozent (gibt es schon)
INFLUX_SENSOR_CAR_BATTERY_SOC_1=car:battery_soc
```

Mindestens erforderlich ist die Fahrleistung, die anderen sind optional - bei fehlenden Sensoren entfallen die davon abhängigen Berechnungen und Darstellungen. Je öfter die Messwerte kommen, desto besser. In meinem Fall bekomme ich alle aufgeführten Werte über die API des Auto-Herstellers im 15min-Takt. 

Ich empfehle jedem, das dieses Feature zukünftig nutzen möchte, sich schon mal darüber Gedanken zu machen, wo die Messwerte herkommen. Idealerweise sammelt man die schon lange und hat sie in seiner InfluxDB.

Noch läuft das Feature nur bei mir lokal und es hat noch ein paar raue Kanten. Dennoch hier ein paar Impressionen:

## Live-Ansicht

Zeigt den aktuellen Zustand des E-Autos mit Akkustand, Eingestöpselt ja/nein, Aktuelle Ladeleistung, Reichweite, maximaler Reichweite und Standort.

[image]

## Fahrleistung

Gefahrene Kilometer, Summe im gewählten Zeitraum, gruppiert nach Tagen, Monaten oder Jahren. Tages-Durchschnitt.

[image]

## Verbrauch pro 100 km

Der saisonale Effekt ist stark sichtbar. Im Winter verbraucht ein E-Auto mehr. Die Anzeige ist der echte Verbrauch, inklusive Ladeverlusten. Was das Auto selber anzeigt, ist oft ein geschönter Werte ohne Ladeverluste.

[image]

## Fahrkosten

Der saisonale Effekt schlägt noch mehr durch, Im Winter muss auch mehr vom teuren Netzstrom bezogen werden. Im Jahresschnitt ist mein E-Auto mit 3,63€ pro 100 km dennoch eine sehr preisgünstige Sache.

[image]

## Maximale Reichweite

Wie weit würde man kommen, wenn der Akku auf 100% geladen wäre? Ergibt sich aus `Reichweite / Ladestand`. Berechnet für jeden Moment, für den Messwerte vorliegen. Gut zu sehen ist die saisonale Schwankung: Im Sommer komme ich 80km weiter als im Winter.

[image]

## Standorte

Liefert das Auto auch Geokoordinaten, lassen sich aufgesuchten Orte automatisch erkennen und auf einer Karte darstellen.

[image]

## Ladevorgänge Wallbox

Ladevorgänge an der heimischen Wallbox werden erkannt, also mit Startzeit und Ende, geladener Energie PV/Netz-Anteil und den sich daraus ergebenen Kosten. Gastladungen können markiert werden (und dem Gast in Rechnung gestellt werden 😄)

[image]

## Ladevorgänge auswärts

Lädt man unterwegs an einer Ladesäule, kann man das manuell erfassen. Das ist notwendig, damit die Verbrauchsberechnungen stimmen. SOLECTRUS muss wissen, was insgesamt ins Auto geflossen ist. Ob man das zukünftig vielleicht über irgendeine API bekommen kann, ist denkbar. Das wäre aber Zukunftsmusik.

[image]

deThommy, 2026-10-04:
Wow 😮 

Sieht richtig gut aus schon. Freue mich, das zu testen!

Ich habe etliches davon in der DB von evcc (das nutze ich seit knapp 3,5 jahren). So wie ich das hier ab und an lese, nutzen das auch hier einige SOLECTRUS-User. 

Planst du dafür eine importmöglichkeit? Also irgendwie ein Template für ne CSV oder so?!

StevieC121176, 2026-10-04:
Kann mich nur anschließen. Schaut fantastisch aus. Eine importmöglichkeit wäre natürlich genial.

ledermann, 2026-10-04:
Danke für die Blumen :) Ich werde zusehen, dass ich das bald zum Testen bereitstellen kann.

Bezüglich Import: In evcc sind möglicherweise gar nicht so viele Messwerte enthalten, die für SOLECTRUS nützlich wären. Der km-Stand ist wird z.B. nur beim Laden abgefragt (über die Auto-Hersteller-Cloud), das ist also sehr lückenhaft. Und Ladevorgänge ermittelt SOLECTRUS nun selbst - aus den Wallbox-Werten, die man ja bereits hat.

Eine Übernahme einzelner Auto-Messwerte aus evcc ist natürlich besser als nix, wenn man sonst gar nichts historisch gespeichert hat. Eine Datenübernahme aus evcc können wir gerne mal separat diskutieren, vielleicht unterschätze ich das Potential.

StevieC121176, 2026-10-05:
@patricknitsch Wäre es möglich den iobroker Adapter schon um die neuen Sensoren zu erweitern?

patricknitsch, 2026-10-05:
Kann ich schon machen. Ich wollte aufgrund der ganzen Paketupdates eh eine neue Version rausbringen. Wird aber noch ein paar Tage dauern, bis mein Laptop zurückkommt. Wenn du nicht warten willst, kannst du diese Sensoren ja problemlos per Hand eintragen :)

Erstell dafür am besten ein Issue.

patricknitsch, 2026-10-09:
Die Sensoren sind jetzt drin und werden automatisch nachgetragen. Hab eine Funktion eingebaut, die auch für zukünftige neue Sensoren automatisch funktioniert. Den Adapter mal über GitHub installieren und prüfen. Wenn alles passt, mache Ich morgen ein neues Release.

StevieC121176, 2026-10-10:
Er hat bei mir 6 neue Sensoren für ein Fahrzeug angelegt, alle mit _1 am Ende. Diese hab ich eingetragen und werden auch befüllt
Müsste er nicht für 5 Fahrzeuge die Sensoren anlegen wie bei den custom Power Sensoren z.Bsp.?

patricknitsch, 2026-10-10:
Grundsätzlich schon. Wir reden dann aber von 30 neuen Sensoren in den Standards. Das macht das ganze schnell unübersichtlich. Zwei Ideen hätte ich dazu:

1. Ich erweitere die Sensorkonfiguration um **E-Auto ja/nein** und gehe dann ins Detail mit **Anzahl 1-5**. Entsprechend werden die Sensoren erweitert.
2. Ich orientiere mich an der Struktur in HELIOS und übernehme die Struktur als Standard, z.B. Wechselrichter, Stromnetz, Batterie, Wallbox, Elektroauto, usw... Damit könnte ich dann die Strukturen synchron halten und ähnlich wie in HELIOS können die Sensoren an und abgewählt werden.

Ich persönlich tendiere eher zu Lösung zwei, damit die ich Konfiguration sauber halte und das ganze verständlich bleibt.

ledermann, 2026-10-10:
@patricknitsch Ja, die HELIOS-Struktur (thematische Gruppierung) zu übernehmen wäre grandios!

Ich schlage vor, alles rund um den ioBroker-Adapter im dortigen Repo zu besprechen, ich habe dort vorhin auch ein Issue (patricknitsch/ioBroker.solectrus-influxdb#212) angelegt.

ledermann, 2026-10-11:
Ich habe den aktuellen Stand der Entwicklung nun auf die Live-Demo gestellt:
https://demo.solectrus.de/cars

Die Implementierung ist vollständig, aber noch nicht fehlerfrei. Und am UI wird sich auch noch etwas tun, gegenüber den obigen Screenshots haben sich auch bereits einige Dinge verändert.

Ich bemühe mich, möglichst bald eine Testversion bereitzustellen.

ledermann, 2026-10-12:
Implementiert in `develop` ab `v1.4-beta-120`.
Weitere Infos in https://github.com/orgs/solectrus/discussions/5966#discussioncomment-18882124

closed 2026-10-12 by ledermann

ledermann, 2026-10-13:
Ich möchte wirklich sehr darum bitten, hier keine individuellen Konfigurations-Probleme zu thematisieren. Das ist für alle anderen Leser störend und ich bekomme das nicht sauber bearbeitet. Bei jedem Problem bitte **UNBEDINGT** ein eigenes Issue anlegen.

ledermann, 2026-10-16:
Hier nochmal ein aktueller Screenshot, da ich noch einiges an der Darstellung verändert habe:

[image]

Die Berechnung von Kosten und Verbrauch pro 100km ist außerdem genauer geworden, da der Akku-Stand bei Beginn und Ende des betrachteten 14-Tage-Zeitfensters mit einbezogen wird. Beginnt oder endet das Zeitfenster mit einem halbvollen Akku, wird das bei der Fahrtkosten-Ermittlung herausgerechnet.

Die verschiedenen Updates in `develop` führen manchmal dazu, dass die Tageszusammenfassungen neu berechnet werden müssen. Ich weiß, dass das lästig ist, daher lege ich großen Wert darauf, dass das so effizient wie möglich abläuft und auch nur dann erfolgt, wenn es wirklich nötig ist. Ob es nötig ist, hängt mittlerweile sehr von der individuellen Konfiguration ab, also den aktiven Sensoren. Das heutige Update (`v1.4-beta-235`) macht nur dann einen Neuaufbau, wenn man mindestens einen `CAR_BATTERY_SOC` hat, andernfalls passiert es nicht.

Falls jemand bei sich in den berechneten Zahlen der Säule "FAHRTEN" unplausible Dinge sieht, bitte ich um ein separates Issue, um das genauer untersuchen zu können. Meine Tests basieren auf dem einen E-Auto, das ich zur Verfügung habe. Der von SOLECTRUS berechnete Verbrauch ist (natürlich) viel höher, als das Auto selbst anzeigt, weil der Auto-Hersteller die Ladeverluste unter den Tisch fallen lässt.