solectrus/solectrus #3534: Ladestand des Autos wird nach Datenunterbrechung nicht mehr angezeigt

gereons, 2024-10-11:
### Beschreibung (auf Deutsch oder Englisch)

In der Zeit vom 10.10. ca. 01:15h nachts bis zum 11.10. ca 08:30h morgens hat das API, mit dem ich den Ladestand meines Autos abfrage irgendwie den Geist aufgegeben. Mittlerweile habe ich dieses Problem durch einen Neustart des entsprechenden Containers beheben können und die Daten laufen wieder ein.

Solectrus zeigte mir entsprechend weder in der Wochenansicht für KW 41 noch in der Monatsansicht für Oktober den Ladestand für den 10. und den 11. an. So weit, so gut.

Da ich in dieser Zeit das Auto weder bewegt noch geladen habe, hab ich mir einfach einen CSV-Export des entsprechenden Zeitraums aus InfluxDB geladen, die fehlenden Daten manuell nachgetragen (immer den gleichen SOC mit passenden Timestamps) und wieder nach Influx hochgeladen - da sieht jetzt auch alles prima aus, die Lücke ist komplett mit Datenpunkten alle 15 Minuten gefüllt so wie meine normale Abfrage auch gelaufen wäre.

Leider sind die Wochen- und Monatsansicht nach wie vor leer, auch nachdem ich den Redis-Cache geleert habe. Werden diese SoC-Daten noch anderswo gecacht?

Wochenansicht:
[image]

Tagesansicht:
[image]

### Welche Version von SOLECTRUS verwendest du?

0.17.0-52 (develop)

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Cloud-Server

ledermann, 2024-10-11:
Ich denke, es handelt sich hier nur um ein Darstellungsproblem, dass mit deinem Re-Import nichts zu tun hat. Ich finde das Phänomen in meiner Instanz ebenfalls, wenn das Auto an einem Tag nicht bewegt wurde, der Akkustand also unverändert bliebt.

Im Wochendiagramm fehlt der zugehörige Min-Max-Balken dann komplett:
https://demo.solectrus.de/car_battery_soc/2024-W38

Das Tagesdiagramm selbst sieht hingegen korrekt aus:
https://demo.solectrus.de/car_battery_soc/2024-09-22

Das ausgelieferte JSON zeigt, dass der Tag korrekt enthalten ist (als `[77, 77]`), Min und Max aber identisch sind:

```
{
  "labels": [
    1726437600000,
    1726524000000,
    1726610400000,
    1726696800000,
    1726783200000,
    1726869600000,
    1726956000000
  ],
  "datasets": [
    {
      "label": "Ladestand E-Auto",
      "data": [
        [
          83,
          87
        ],
        [
          62,
          83
        ],
        [
          55,
          84
        ],
        [
          75,
          84
        ],
        [
          60,
          86
        ],
        [
          77,
          78
        ],
        [
          77,
          77
        ]
      ],
      ...
    }
  ]
}
```

Das liegt wohl an der grundsätzlichen Darstellung von Min/Max-Diagrammen. Es sollte in solchen Fällen eine dünne Linie erscheinen. Warum das nicht geschieht, werde ich klären.

gereons, 2024-10-11:
Danke für die Erklärung!

ledermann, 2024-10-11:
Ich habe Chart.js jetzt davon überzeugen können, die Balken mit einer Mindesthöhe darzustellen, das Problem ist somit korrigiert. Enthalten in `develop` und zu sehen in der Live-Demo am oben zitierten Beispiel:
https://demo.solectrus.de/car_battery_soc/2024-W38

gereons, 2024-10-11:
Super, danke!

closed 2024-10-11 by gereons