solectrus/solectrus #3538: Kapazitätsmessung über Auswahl von Datum und Uhrzeit

halmich, 2024-10-12:
### Beschreibung (auf Deutsch oder Englisch)

Ich möchte mich nochmal für die Auswahl von Daten aus einem bestimmten Zeitraum aussprechen. 
Damit könnte man u.a. überprüfen, ob der Senec Speicher noch die versprochenenen 100% Kapazität bringt,
Im Anhang ein Beispiel. Man sieht hier, dass der Speicher um 17 Uhr voll war, und dann um 04 Uhr am nächsten Tag leer war. In der Zeit dazwischen wurde er konstant entladen und die Akkuentnahme sind 4,75kWh. Der Speicher hat also nicht mehr die versprochenen 100% von 5kWh. 
[image]

ledermann, 2024-10-12:
Ja, guter Hinweis, in #3007 geht es nur um das Datum, aber auch die zusätzliche Angabe der Uhrzeit kann nützlich sein, wie man hier sehen kann.

Ansonsten könnte ich mir für Akkubetrachtungen vorstellen, die Darstellung eines Tages optional auf "Nacht" umzustellen, also von Sonnenuntergang bis Sonnenaufgang (möglichst minutengenau). Dann bräuchte man die Uhrzeit gar nicht manuell eingeben, das würde dann automatisch erfolgen.

Zur Kapazitätsmessung habe ich in #3540 etwas geschrieben.

halmich, 2024-10-12:
automatisch würde ich das nicht machen. 
Die Entladung fängt oft vor Sonnenuntergang an, also man produziert noch 500 Watt, benötigt aber 800W.
Daneben bringt es auch nur an Tagen etwas, wo durchgängig beladen wird (ohne zwischenentladung) bzw. wo durchgängig entladen wird (ohne zwischenbeladung).