solectrus/solectrus #3556: Umbenennung Akku

bennysweb, 2024-10-16:
### Beschreibung (auf Deutsch oder Englisch)

Der besseren Übersichtlichkeit halber schlage ich vor, die Bezeichnung "Akku" in der Auswahl über den Graphen in "Hausakku" und "Ladestand Akku" in "Ladestand Hausakku" umzubenennen. 

### Welche Version von SOLECTRUS verwendest du?

17.0

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Synology

ledermann, 2024-10-16:
Ja, guter Vorschlag, den ich gerne übernehme. Zu sehen in der Live-Demo und in `develop`.

closed 2024-10-16 by ledermann

bennysweb, 2024-10-16:
Super, wieder mal vielen Dank für die prompte Umsetzung!