solectrus/solectrus #3561: Hausverbrauch wird zyklisch nicht angezeigt.

deadrabbit87, 2024-10-18:
### Beschreibung (auf Deutsch oder Englisch)

Ich habe das Problem, dass Solectrus in der UI zyklisch keinen Hausverbrauch mehr anzeigt. 

Das sieht dann z,B. so aus: 
[image]

Die Werte kommen vom MQTT-Collector, der zu gegebenem Zeitpunkt auch Werte liefert. 

Hier die Ausgabe von `docker logs  solectrus-mqtt-collector` (gekürzt)
```
# Message from 2024-10-18 13:02:00 +0200
  topic = evcc/site/homePower
  message = 404.27
  => pv:house_power = 404
```
Und zur ungefähr gleichen Zeiten der Tooltip: 
[image]

Ich hab das immer wieder, z.B. hier von gestern: 

[image]

### Welche Version von SOLECTRUS verwendest du?

0.17.0

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Ubuntu 24.04.1 LTS auf HP Microserver

ledermann, 2024-10-18:
Ich vermute, du hast einen großen Verbraucher wie Wärmepumpe oder Wallbox konfiguriert, der aus dem Hausverbrauch herausgerechnet wird, hast also die Variable `INFLUX_EXCLUDE_FROM_HOUSE_POWER` aktiviert. Ich tippe darauf, dass in den Phasen mit einem 0-Verbrauch die Wallbox oder die Wärmepumpe gerade Strom verbrauchen.

Bei diesem Herausrechnen achtet SOLECTRUS darauf, dass das Resultat aus `Hausverbrauch - Großverbraucher` nicht negativ wird und zeigt dann stattdessen eine 0 an.

Viel machen kann man dagegen nicht. Du solltest aber prüfen, ob der Großverbraucher korrekt gemessen wird und die Messung möglichst im gleichen Intervall wie beim Hausverbrauch erfolgt.

Für genauere Untersuchen kannst du direkt in InfluxDB nachsehen, wie sich die echten Werte von Hausverbrauch und Großverbraucher verhalten. In SOLECTRUS selbst siehst du aufgrund der Einstellung von `INFLUX_EXCLUDE_FROM_HOUSE_POWER` an jeder Stelle immer nur die Differenz (limitiert auf 0).

Als Besitzer eines SENEC-Stromspeichers habe ich übrigens das gleiche Problem, wenn über die SENEC-Wallbox das Auto beladen wird. Dabei spielt sich die Berechnungen direkt im Stromspeicher ab, SOLECTRUS rechnet also nichts selbst heraus. Dabei kommt es auch oft zu dem gleichen Phänomen eines Hausverbrauchs von 0, ohne dass ich etwas dagegen tun könnte. Auch die SENEC-App zeigt das so an, mit SOLECTRUS hat das nichts zu tun.

deadrabbit87, 2024-10-18:
Es ist tatsächlich so, dass in dem Zeitpunkt das Fahrzeug geladen hat. Da meine Werte von evcc kommen, ist da aber die Wallbox schon heraus gerechnet. 

`INFLUX_EXCLUDE_FROM_HOUSE_POWER` ist aktiv, da die Werte der Wärmepumpe nicht von evcc kommen. 

Um das ganze zu beheben, müsste man also Solectrus den Hausverbrauch mit der Leistung der Wallbox übergeben?

ledermann, 2024-10-18:
Verstehe ich dein Setup richtig?

- Hausverbrauch kommt von evcc per MQTT, bereinigt um Wallbox, aber enthält die Wärmepumpe
- Wallbox kommt ebenfalls von evcc per MQTT
- Wärmepumpe kommt separat per MQTT oder Shelly-Collector

Es gibt diese Zeile in der `.env`:
`INFLUX_EXCLUDE_FROM_HOUSE_POWER=heatpump_power`

`INFLUX_EXCLUDE_FROM_HOUSE_POWER` enthält also insbesondere **nicht** `wallbox_power`, weil das Herausrechnen bereits in evcc stattgefunden hat. Bitte prüfe das, möglicherweise ist das der Fehler und es wird die Wallbox herausgerechnet, die in dem Wert gar nicht mehr enthalten ist.

Wenn das alles so ist, sehe ich momentan nicht, wie das Phänomen zustande kommt, zumindest sehe ich keinen Fehler in SOLECTRUS.

Was du wirklich machen solltest: Prüfe direkt in InfluxDB (siehe [Anleitung](https://docs.solectrus.de/referenz/influxdb/direktzugriff#zugriff-auf-die-weboberfl%C3%A4che-von-influxdb)), wie sich die Werte von Hausverbrauch, Wallbox und Wärmepumpe an den betroffenen Tagen verhalten. In InfluxDB kannst du vor allem den Hausverbrauch im Original sehen, also ohne dass irgendwas herausgerechnet wurde. Möglicherweise ergeben sich daraus neue Anhaltspunkte.

> Um das ganze zu beheben, müsste man also Solectrus den Hausverbrauch mit der Leistung der Wallbox übergeben?

Könntest du versuchen, aber einen zwingenden Grund sehe ich dafür momentan nicht.

deadrabbit87, 2024-10-18:
Ja, genau so ist mein Setup. Die Wärmepumpe kommt vom Shelly-Collector über einen Shelly PM. 

Ich denke, ich habe den Fehler gefunden. Bei mir ist `INFLUX_EXCLUDE_FROM_HOUSE_POWER` wie folgt gesetzt: 
`INFLUX_EXCLUDE_FROM_HOUSE_POWER=HEATPUMP_POWER ,WALLBOX_POWER`

Das ist natürlich Mist. 

Ist also, definitv kein Fehler in Solectrus. Der Fehler sitzt wie immer vor dem Bildschirm ;-) 

Danke an dieser Stelle für deine großartige Arbeit!

ledermann, 2024-10-18:
Prima, so ist es mir auch am liebsten ;-)