solectrus/solectrus #3588: Anzeige mobile Geräte (Android) neue Instanz auf Raspi5/Bookworm

Zeppelin-PV, 2024-10-25:
### Beschreibung (auf Deutsch oder Englisch)

Ich habe eine neue Instanz auf einem Raspi5 OS Bookworm installiert
Und soweit alles ok, Dank Deiner Hilfe, Georg.
Jetzt fällt mir auf, dass wie schon mal (schon ne Weile her) die Anzeige gestaucht dargestellt wird weil der Browser oben die URL Zeile einblendet
und unten die Systembuttons gezeigt werden.
Über die Einstellung Vollbild anzeigen geht es nur bis zu einem erneuten Aufruf.
Meine erste Instanz von Solectrus, auf einem Raspi4 OS Bullseye wird formatfüllend angezeigt.

Raspi5
[image]

Raspi4
[image]

Außerdem habe ich in der URL ein "Warndreieck"
[image]

inhaltlich heißt es da:
[image]

Die Icons (egal ob Firefox oder Chrome) habe ich genau gleich erstellt wie die des Raspi4, wo es top aussieht.

LG Ernst

### Welche Version von SOLECTRUS verwendest du?

17,1

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspberry 5 mit OS Bookworm

ledermann, 2024-10-25:
Seltsam. Die Ursache würde ich eher beim Browser suchen. Auch dass die Icons unterschiedlich beschriftet sind, ist merkwürdig. Man Icons natürlich umbenennen, die Vorgabe ist aber alles in Großbuchstaben.

Im Zweifelsfall beide Icons nochmal vom Gert löschen und dann neu anlegen.

Zeppelin-PV, 2024-10-25:
Du hast recht, Georg
Irgend was hat sich wohl bei den "großen" Browsern geändert.
Hab lange probiert und mit Kiwi einen gefunden der Solectrus so anzeigt wie ich es gewohnt bin.
-- Gewohnheiten sind das Rückgrat des Alltags