solectrus/solectrus #3599: PowerSplitter-Aufteilung fehlt im Diagramm

ledermann, 2024-10-28:
Beobachtet in #3533:

> Was mir aber in den letzten Monaten aufgefallen ist. In der Auswertung wird nicht alles anzeigt. Im Monat September und Oktober fehlen in der Auswertung Haus-Verbrauch der Netz Bezug.

[image]

Dies betrifft die monatliche und jährliche Auswertung.

[image]

Hast du hierzu eine Idee?

_Originally posted by @patricknitsch in https://github.com/solectrus/solectrus/discussions/3587#discussioncomment-11074953_

ledermann, 2024-10-28:
@patricknitsch: Ich kann das leider noch nicht reproduzieren, daher folgende Fragen:

- Tritt das Problem wirklich erst seit Umstellung auf `pr-3533` auf? War also mit Version `0.17.0` oder `0.17.1` noch alles ok?
- Sind wirklich nur die Monate September und Oktober betroffen, alle anderen also ok?
- Läuft dein Power-Splitter korrekt? Guck mal nach, ob der Container läuft und die Logs keine Fehler anzeigen.
- Verschwindet das Problem, wenn du zurück auf `0.17.0` oder `0.17.1` gehst?

patricknitsch, 2024-10-28:
Ja, das liegt auch daran, dass die Aufteilung des Verbrauchs in Bezug und Photovoltaik in der 0.17.1 noch nicht angezeigt wird. Deswegen ist es jetzt auch sichtbar.

Es sind nur Oktober und November. Die Monate, bzw. die Jahre davor werden richtig grafisch angezeigt.

ledermann, 2024-10-28:
Die Aufteilung in Netzbezug und PV-Anteil erfolgt seit Version 0.16, notwendig ist aber der Power-Splitter, der permanent in einem eigenen Container die Berechnungen durchführt.

Durch `pr-3533` hat sich in diesem Punkt gar nichts geändert. Die Aufteilung in Netz und PV erfolgte auch schon vor.

Ich muss jetzt nochmal nach dem Power-Splitter fragen. Läuft der bei dir und lief der auch die letzten Monate? Aufgrund der Fehlerbeschreibung vermute, dass der seit September nicht mehr läuft. 

Bei Unsicherheit könntest du folgendes machen:

1) Power-Splitter zurücksetzen: `docker compose kill --signal USR1 power-splitter`. Die dann folgende Neuberechnung wird eine Weile dauern. 
2) Anschließend, wenn das durchgelaufen ist (nicht vorher! - siehe Docker-Log), unter Einstellungen die Tageszusammenfassungen zurücksetzen. Auch die müssen dann erneut aufgebaut werden.

patricknitsch, 2024-10-28:
Stimmt. Anscheinend gibt es wirklich ein Problem mit dem Power Splitter. Dieser bleibt aktuell im Status Restarting hängen.

ledermann, 2024-10-29:
Ok, erst mal prüfen, ob der Power-Splitter-Container überhaupt läuft: `docker compose ps`
Dann die Logs vom Poewr-Splitter angucken: `docker compose logs power-splitter -n 100`

Bitte poste jeweils den Output dieser beiden Kommandos.

Je nachdem, was zu sehen ist, könnte es eventuell helfen, alles mal neu zu starten:
`docker compose down`, dann `docker compose up -d`

Auch dabei unbedingt die Logs im Auge behalten, alles von Relevanz steht nur dort.

patricknitsch, 2024-10-29:
```
NAME                             IMAGE                                         COMMAND                  SERVICE              CREATED        STATUS                          PORTS
solectrus-dashboard-1            ghcr.io/solectrus/solectrus:latest            "docker/entrypoint.s…"   dashboard            8 hours ago    Up 7 hours (healthy)            0.0.0.0:3000->3000/tcp, :::3000->3000/tcp
solectrus-forecast-collector-1   ghcr.io/solectrus/forecast-collector:latest   "bundle exec app/mai…"   forecast-collector   9 hours ago    Up 7 hours                      
solectrus-influxdb-1             influxdb:2.7-alpine                           "/entrypoint.sh infl…"   influxdb             3 days ago     Up 7 hours (healthy)            8086/tcp
solectrus-mqtt-collector-1       ghcr.io/solectrus/mqtt-collector:latest       "bundle exec app.rb"     mqtt-collector       9 hours ago    Up 7 hours                      
solectrus-postgresql-1           postgres:16-alpine                            "docker-entrypoint.s…"   postgresql           5 weeks ago    Up 7 hours (healthy)            5432/tcp
solectrus-power-splitter-1       ghcr.io/solectrus/solectrus:develop           "docker/entrypoint.s…"   power-splitter       8 hours ago    Restarting (1) 32 seconds ago   
solectrus-redis-1                redis:7-alpine                                "docker-entrypoint.s…"   redis                3 days ago     Up 7 hours (healthy)            6379/tcp
solectrus-senec-collector-1      ghcr.io/solectrus/senec-collector:latest      "bundle exec app.rb"     senec-collector      9 hours ago    Up 7 hours                      
solectrus-watchtower-1           containrrr/watchtower                         "/watchtower --scope…"   watchtower           4 months ago   Up 7 hours (healthy)            8080/tcp
```

```
power-splitter-1  | SOLECTRUS Photovoltaic Dashboard
power-splitter-1  | Copyright (C) 2020-2024 Georg Ledermann
power-splitter-1  | License: GNU AGPLv3 - https://www.gnu.org/licenses/agpl-3.0.html
power-splitter-1  | Version v0.17.1-33-g374501ca (develop), built on 2024-10-28T13:52:34Z
power-splitter-1  | Using ruby 3.3.5 (2024-09-03 revision ef084cc8f4) [x86_64-linux-musl]
power-splitter-1  | Error: One or more required environment variables (REDIS_URL, INFLUX_HOST, DB_HOST) are not set. Stopping...
power-splitter-1  | Based on Alpine Linux 3.20.3
```

Sind neue Variablen hinzugekommen? Die .env habe ich nicht verändert..

ledermann, 2024-10-29:
Bestens, somit können wir das Thema schließen.

closed 2024-10-29 by ledermann