solectrus/solectrus #3617: Dashboard startet nicht

mheitm, 2024-11-01:
### Beschreibung (auf Deutsch oder Englisch)

Nach Update von .env und compose.yaml  (Erstellung neuer Dateien per Konfigurator) auf die neue Konfigurationssystematik, und loesen des Problems der PostgreSQL Datenbankversion mit Konfig 15 statt 16, kommt leider  auf der Solectrus Startseite nur eine Fehlermeldung:
*********************
500
Internal Server Error
Sorry, something went wrong. See the server logs for more details.

docker compose logs dashboard
*********************

INFLUX_BUCKET=solectrus habe ich in der .env gefunden, aber weiter komme ich leider nicht.

Danke!

### Welche Version von SOLECTRUS verwendest du?

17.1

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspberry Pi 4, Debian

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
docker compose logs dashboard:
dashboard-1  | E, [2024-11-01T19:33:20.926060 #1] ERROR -- : [172.18.0.1]   
dashboard-1  | [172.18.0.1] ActionView::Template::Error (failed to initialize execute state: could not find bucket "solectrus"):
dashboard-1  | [172.18.0.1] 
dashboard-1  | Causes:
dashboard-1  | [172.18.0.1] InfluxDB2::InfluxError (failed to initialize execute state: could not find bucket "solectrus")
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
services:
  dashboard:
    image: ghcr.io/solectrus/solectrus:latest
    depends_on:
      postgresql:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - postgresql
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - TZ
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - WEB_CONCURRENCY
      - INSTALLATION_DATE
      - ADMIN_PASSWORD
      - FRAME_ANCESTORS
      - UI_THEME
      - CO2_EMISSION_FACTOR
      - DB_HOST=postgresql
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
[... 427 lines cut ...]
# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token

# Volume path for storing the InfluxDB data
INFLUX_VOLUME_PATH=./influxdb

##################################################################
###                    PostgreSQL database                     ###
##################################################################

# Volume path for storing the PostgreSQL data
DB_VOLUME_PATH=./postgresql

##################################################################
###                 Redis in-memory database                   ###
##################################################################

# Volume path for storing the database snapshot
REDIS_VOLUME_PATH=./redis

# URL of the Redis database
REDIS_URL=redis://redis:6379/1
```

ledermann, 2024-11-01:
Für die Umstellung auf die Sensorkonfiguration sollte eher nicht der Konfigurator verwendet werden, weil dieser Einstellungen vornimmt, die nicht gut zu dem vorhandenen Datenbestand passen. Möglich ist es zwar, erfordert aber ein etwas tiefer gehendes Verständnis der Datenhaltung in InfluxDB. 

Ein Beispiel ist der Bucket: In deiner bestehenden InfluxDB wurde vermutlich ein anderer verwendet, wie der genau hieß, steht in der alten `.env`. Wenn du das korrigiert hast, könnten möglicherweise weitere Anpassungen notwendig sein.

Meine Empfehlung ist daher, für Aktualisierungen bestehender Installationen **nicht** den Konfigurator zu verwenden, sondern einfach nur die Warnung im Log abzuarbeiten, siehe hier:
https://docs.solectrus.de/wartung/sensor-konfiguration

gereons, 2024-11-02:
@mheitm Wenn dein Original-Setup auf den älteren Beispiel-Files basiert heisst dein Bucket vermutlich nicht `solectrus` sondern `my-solectrus-bucket`.

mheitm, 2024-11-02:
Danke fuer den Tipp mit 'my-solectrus-bucket', jetzt laeuft es wieder. War leider nicht so schlau von mir die alte .env einfach zu ueberschreiben und vorher keine Sicherungskopie zu machen.
Ich schau mal ob es irgendwo Fehlermeldungen wegen Inkompatibilitaeten gibt, bisher habe ich allerdings keine Funktionseinschraenkungen bemerkt.
Da das Problem geloest ist mache ich dieses Ticket zu.
Danke Euch!

closed 2024-11-02 by mheitm

ledermann, 2024-11-02:
Prima! Bezüglich Inkompatibilitäten würde ich folgende Dinge im Auge behalten:

1) Fehler oder Warnungen im Log (`docker compose logs`, insbesondere im Dashboard und den Kollektoren)
2) Werden auch historische Messwerte korrekt angezeigt?
3) Erscheint die Vorhersage (Forecast) korrekt? 

Wenn die Punkte ok sind, dann ist das Thema durch :)