solectrus/solectrus #3646: Wärmepumpe: Erzeugte Wärme mit Anteilen aus Netzstrom/PV-Strom/Luft

ledermann, 2024-11-08:
### Beschreibung (auf Deutsch oder Englisch)

Wer eine Wärmepumpe besitzt, interessiert sich für deren **Arbeitszahl**. Diese gibt die Effizienz an, mit der aus Strom die benötigte Wärme erzeugt wurde. Eine Arbeitszahl bewegt sich meist zwischen 3 und 5, aus 1 kWh Strom wird etwa 3-5 kWh Wärme erzeugt. Die Arbeitszahl ergibt sich also aus:

```
Arbeitszahl = Erzeugte Wärme (in Wh) / Stromverbrauch (in Wh)
```

Hat man die erzeugte Wärme als Messwert vorliegen, könnte SOLECTRUS in einem neuen Diagramm die Arbeitszahl im Zeitverlauf darstellen. Damit habe ich experimentiert, das Ergebnis ist aber nicht so überzeugend. Die Arbeitszahl ist im Herbst höher als im Sommer (da kein Heizbedarf, nur Warmwassererzeugung)

Die Arbeitszahl allein hat also nur eine beschränkte Aussagekraft - interessanter wird es, wenn man die erzeugte Wärme dazu nennt. Eine niedrige Arbeitszahl ist weniger problematisch, wenn nur wenig Wärme erzeugt wurde (wie im Sommer). Die gleiche Arbeitszahl im Winter hat eine viel größere Auswirkung.

Unter Zuhilfenahme des Power-Splitters lässt sich ein interessantes Diagramm erzeugen, dass die Effizienz der Wärmepumpe im Zeitverlauf darstellt und dabei nach Herkunft der Energie (Netzstrom, PV-Strom und Umgebungsluft) differenziert:

[image]

Die Idee ist, die aus der Umgebung bezogene Energie rückwärts zu errechnen:

```
Aus der Umgebungsluft entnommene Energie (in Wh) = 
  Erzeugte Wärme (in Wh) 
  - Netzstrom (in Wh) 
  - PV-Strom (in Wh) 
```

Bei einer gut eingestellten Wärmepumpe freut man sich dann über:
- Viel Blau (weil diese Energie ist kostenlos)
- Viel Grün (weil diese Energie wurde selbst erzeugt)
- Wenig Rot (weil diese Energie musste zugekauft werden)

Die Implementierung dieses Diagramm in SOLECTRUS wäre recht einfach. Entscheidend ist, einen Sensor zu ergänzen, der die **erzeugte Wärme** angibt. Der Rest ist dann ein wenig rechnen und Diagramm zeichnen.

Wer die erzeugte Wärme nicht direkt aus seiner Wärmepumpe abgreifen kann, der kann sich mit Vorlauftemperatur, Rücklauftemperatur und Durchfluss helfen, denn:

```
Erzeugte Wärme (in W) = 
  Durchfluss (in l/min) * (Vorlauftemperatur - Rücklauftemperatur) * 60 * 1.163
```

Diese Formel lässt sich im MQTT-Collector [hinterlegen](https://docs.solectrus.de/referenz/mqtt-collector/topics#mapping_x_json_formula-optional).

halmich, 2024-11-09:
Das ist schon ziemlich cool! :)

q2doc, 2024-11-10:
Hi Georg, so ähnlich realisiere ich das aktuell in ioBroker - wäre toll wenn es hier direkt ginge. Das Thema Wasser vs Heizung habe ich im ioBroker so geregelt, dass je nach Modus der Wärmepumpe die Werte entweder auf Heizung oder Warmwasser "gebucht" werden (also faktisch ist die Wärmepumpe zwei Verbraucher). Das wäre ja auch denkbar in Solectrus, was mich immer wieder auf meine ;-) most wanted bringt:

- zwei Ladepunkte / Wallboxen
- Fahrzeugdifferenzierung

die Lösung stelle ich mir (vermutlich zu einfach) ähnlich vor, aber es ist am Ende mehrere Verbraucher and mehreren Endpunkten. Schönes WE!

spirit0607, 2024-11-12:
Super klasse feature! Ich wäre hätte wohl für meine Wärmepumpe die notwendigen Messwerte und wäre gerne bei einem Beta Test dabei. Ist das jetzt schon möglich?

ledermann, 2024-11-13:
@spirit0607 Danke für dein Interesse :) Noch ist dieses Feature am Anfang der Entwicklung. Ich bin momentan dabei, weitere Sensoren einzuführen. Es wird weitere Verbraucher geben (#3396), aber auch die Output-Seite sehe ich mir an (Erzeugte Wärme der Wärmepumpe, Fahrleistung des E-Autos). Noch ist nicht im Detail klar, worauf das hinauslaufen wird. Ich experimentiere noch.

Es gibt daher noch keine Beta-Version. Ich werde es natürlich ankündigen, sobald etwas ausprobiert und getestet werden kann.

ledermann, 2024-11-13:
An die Interessenten dieses Features: Bitte den Erstbeitrag oben mit einem 👍 versehen, die [Abstimmung](https://github.com/orgs/solectrus/discussions/3523) ist auf Dauer angelegt.

ledermann, 2024-12-06:
Hierzu war in den letzten Tagen auf der Demo-Instanz ein erster Implementierungsversuch zu sehen. Da ich damit aber im Detail noch nicht glücklich bin und nicht zu viele Töpfe gleichzeitig auf dem Herd haben möchte, habe ich das erst mal wieder zurückgestellt. Ich wende mich dem wieder zu, wenn die Verbraucher (#3396) abgehandelt sind.

unclej84, 2025-01-25:
Hallo zusammen, ich finde das Feature und auch die Darstellung (soweit ich sehen konnte) super und habe dann auch direkt angefangen die Werte schon einmal zu loggen (die Wärmeleistung gemäß der Formel oben zusätzlich zu den Einzelwerten). Als es die letzten Tage so um die 0°C war, fiel mir auf, dass die Wärmepumpe für das Abtauen auf "Rückwärtsgang" geschaltet hat, sodass die Wärmeleistung dann kurzzeitig (mathematisch korrekt) negativ war, weil die Vorlauftemperatur dann unter der Rücklauftemperatur war. Wie würdet ihr bzw. wie sollte man nach Solectrus-Logik damit umgehen? Habe die Formel für den Fall so geändert, dass hier 0 anstatt dem negativen berechneten Wert gespeichert wird.
Danke für eure Meinungen.
Gruß Jörg

ledermann, 2025-01-26:
@unclej84 Die von mir oben angegeben Formel habe ich aus den gleichen Gründen leicht erweitert. Meine Wärmepumpe liefert mir einen Status, d.h. ich kann zwischen Heizen und Abtauen (und noch ein paar anderen Dingen) unterscheiden. Dennoch kommt es manchmal vor, dass Rücklauf (geringfügig) wärmer ist als Vorlauf. Mir ist nicht ganz klar, warum. Möglicherweise sind das Messungenauigkeiten. Das behandle ich dann genau wie du, setze also in diesem Fall 0 an.

Hier meine Gesamtformel (in der Notation des MQTT-Collectors):

```
IF({Operation Mode} = 'Heating' 
   AND {Freeze Protection} = 'OFF' 
   AND {Leaving water temp. after BUH (R2T)} > {Inlet water temp.(R4T)}, 
{Flow sensor (l/min)} * 60.0 * 1.163 * ({Leaving water temp. after BUH (R2T)} - {Inlet water temp.(R4T)}), 
0)
```

Damit komme ich **ungefähr** auf die Werte, die die Wärmepumpe selbst berechnet.

ledermann, 2025-03-15:
@deThommy Der Scatterplot sieht gut aus, das ist auch eine interessante Idee.

Wann es hier mit diesem Feature weiter geht, ist noch etwas unklar. Aber es kommt auf jeden Fall.

ledermann, 2025-08-11:
@unclej84 Der `INFLUX_SENSOR_HEATPUMP_STATUS` entspricht dem `INFLUX_SENSOR_SYSTEM_STATUS`, nur eben bezogen auf die Wärmepumpe. Da kann ein beliebiger String rein, die fünf von dir genannten sind gute Kandidaten dafür. Der Wert wird nur angezeigt und nicht weiter verarbeitet. Die Anzeige erfolgt bei [AKTUELL](https://demo.solectrus.de/heatpump/heatpump_heating_power/now) rechts oben neben der Uhr.

ledermann, 2025-08-12:
@deThommy Das klären wir sehr gerne, aber bitte in einem separaten Issue, sonst wird das für alle anderen, die dieses (Haupt-)Issue verfolgen, äußerst unübersichtlich.

deThommy, 2025-08-13:
Bei mir wird im Übrigen in der Grafik die **Umgebungswärme** nicht in dem Balkendiagramm dargestellt. 

Muss ich dafür einen weiteren Sensor konfigurieren oder liegt das daran, dass die erzeugte Wärme hier kleiner als die eingesetzte (PV-)Energie ist? Das liegt in der Tageszusammenfassung für heute daran, dass heute Morgen (ca. 6 Uhr) schon mal Warmwasser produziert wurde, der eine Sensor jedoch nicht korrekt konfiguriert war, weshalb die Werte nicht ankamen.

Habe ich das irgendwo überlesen? Habe auf die Schnelle dazu (noch) nichts finden können...

[image]

gereons, 2025-08-15:
> @gereons wird es mit Viessmann vermutlich leichter haben.

Sieht leider bisher nicht so aus :-( Aus dem kostenlosen API-Zugriff kann ich (soweit ich das bisher sehe) nur Stromverbrauch und Aussentemperatur auslesen, und den Strom hab ich eh via Shelly 3EM. Was fehlt ist die erzeugte Wärmeenergie, die aber irgendwo herkommen muss, denn die Viessmann-App zeigt sie an. Ich teste jetzt mal ob die kostenpflichtige API das hergibt, aber  wenn - ob ich dauerhaft knapp €150/Jahr dafür bezahlen will werd ich mir gründlich überlegen.

ledermann, 2025-08-15:
@gereons €150 jährlich, damit man an seine eigenen Messwerte herankommt, ist schon eine Ansage :-( Adhoc habe ich den [Splitter for Viessmann Optolink](https://github.com/philippoo66/optolink-splitter) gefunden, vielleicht hilft das weiter. 

Sieht nicht aus, als wäre die Anbindung an Viessmann ein Spaziergang. Für weitere Details vielleicht besser ein separates Issue oder Diskussionsthema aufmachen.

ledermann, 2025-08-15:
Noch ein Hinweis zur Konfiguration: Für eine möglichst genaue Darstellung sollte darauf geachtet werden, Stromverbrauch und Wärmeerzeugung in **kurzen Zeitabständen** und möglichst im **gleichen Takt** nach InfluxDB zu bringen. Wenn es große Abweichungen gibt (z.B. Stromverbrauch im 5s-Takt, Wärmeerzeugung aber im 1m-Takt), dann wird der Saldo der erzeugten Wärme sowie die Arbeitszahl grob falsch berechnet.

Hintergrund ist, dass beide Leistungswerte (in W) von SOLECTRUS selbst in Energie (kWh) umgerechnet werden. Da es bei einer Wärmepumpe zu hohem Leistungsänderungen in kurzer Zeit kommen kann, ist eine kurze Abtastrate wichtig.

Bei mir kommt der Stromverbrauch (gemessen durch einen Shelly) im 5s-Takt, die erzeugte Wärme (berechnet per ESPAltherma) im 10s-Takt an - also beides ohne Cloud, sondern lokal. Damit ergeben sich gute Ergebnisse.

ledermann, 2025-08-24:
@q2doc Momentan wird erst mal nur die **insgesamt** erzeugte Wärme verarbeitet und dargestellt, also für Heizung **plus** Warmwasser. Also ohne Differenzierung. Ich behalte das aber im Hinterkopf, ggfs. ergänze ich das später noch. Das ist aber überhaupt nicht trivial, weil man dann womöglich auch die Arbeitszahl separieren möchte, also "Arbeitszahl für Heizung" und "Arbeitszahl für Warmwasser" - um das zu bestimmen, muss aber auch der Stromverbrauch getrennt ermittelt werden. Es wird also immer komplizierter, daher scheue ich das noch.

Nitrox912, 2025-08-31:
Hallo, vielen Dank für die tolle Arbeit. Ich habe gestern die Beta installiert und es läuft ganz gut. Allerdings eine Frage (evtl. eine Bug?):

Die Warmwassertemperatur in der Anzeige hat sich bei mir nicht geändert. Sie steht noch bei 41,7 Grad, obwohl im Diagramm der aktuelle Wert bereits vorhanden ist.

[image]

ledermann, 2025-10-16:
Das Feature ist jetzt in der Version `1.0-beta` enthalten, d.h. das Docker-Tag sollte angepasst werden, `pr-4588` wird nicht mehr aktualisiert. Alles weitere hier:
https://github.com/orgs/solectrus/discussions/4805

closed 2025-11-05 by ledermann