solectrus/solectrus #3682: Es wird die Version v0.18.1 nict gepulled

KMontenarh, 2024-11-15:
### Beschreibung (auf Deutsch oder Englisch)

ein doker Compose pull datet nicht von v0.18.0 auf o.18.1 up

### Welche Version von SOLECTRUS verwendest du?

v0.18.0

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Pi4 

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
keine infos im Log
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
# version: '3.7'

services:
  app:
    image: ghcr.io/solectrus/solectrus:latest
    depends_on:
      db:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - db
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - APP_HOST
      - FORCE_SSL
      - SECRET_KEY_BASE
      - INSTALLATION_DATE
      - ELECTRICITY_PRICE
      - FEED_IN_TARIFF
      - ADMIN_PASSWORD
      - DB_HOST=db
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
    healthcheck:
      test: ['CMD-SHELL', 'nc -z 127.0.0.1 3000 || exit 1']
    restart: always

  influxdb:
    image: influxdb:2.7-alpine
    volumes:
      - ${INFLUX_VOLUME_PATH}:/var/lib/influxdb2
    environment:
      - DOCKER_INFLUXDB_INIT_MODE=setup
      - DOCKER_INFLUXDB_INIT_USERNAME=${INFLUX_USERNAME}
      - DOCKER_INFLUXDB_INIT_PASSWORD=${INFLUX_PASSWORD}
      - DOCKER_INFLUXDB_INIT_ORG=${INFLUX_ORG}
      - DOCKER_INFLUXDB_INIT_BUCKET=${INFLUX_BUCKET}
      - DOCKER_INFLUXDB_INIT_ADMIN_TOKEN=${INFLUX_ADMIN_TOKEN}
    command: influxd run --bolt-path /var/lib/influxdb2/influxd.bolt --engine-path /var/lib/influxdb2/engine --store disk
    # Optional: Allow InfluxDB to be accessed from the outside.
    # ports:
    #   - 8086:8086
    restart: always
    healthcheck:
      test: ['CMD', 'influx', 'ping']
      interval: 30s
      timeout: 10s
      retries: 5
      start_period: 30s

  db:
    image: postgres:15-alpine
    environment:
      - POSTGRES_PASSWORD=${POSTGRES_PASSWORD}
    volumes:
      - ${DB_VOLUME_PATH}:/var/lib/postgresql/data
    restart: always
    healthcheck:
      test: ['CMD-SHELL', 'pg_isready -U postgres']
      interval: 10s
      timeout: 20s
      retries: 5
      start_period: 60s

  redis:
    image: redis:7-alpine
    volumes:
      - ${REDIS_VOLUME_PATH}:/data
    restart: always
    healthcheck:
      test: ['CMD', 'redis-cli', 'ping']
      interval: 10s
      timeout: 20s
      retries: 5
      start_period: 60s

  senec-collector:
    image: ghcr.io/solectrus/senec-collector:latest
    depends_on:
      influxdb:
        condition: service_healthy
    links:
      - influxdb
    environment:
      - SENEC_HOST
      - SENEC_SCHEMA
      - SENEC_INTERVAL
      - INFLUX_HOST
      - INFLUX_SCHEMA
      - INFLUX_PORT
      - INFLUX_TOKEN=${INFLUX_TOKEN_WRITE}
      - INFLUX_ORG
      - INFLUX_BUCKET
    restart: always

  forecast-collector:
    image: ghcr.io/solectrus/forecast-collector:latest
    depends_on:
      influxdb:
        condition: service_healthy
    links:
      - influxdb
    environment:
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_WRITE}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - FORECAST_LATITUDE
      - FORECAST_LONGITUDE
      - FORECAST_DECLINATION
      - FORECAST_AZIMUTH
      - FORECAST_KWP
      - FORECAST_INTERVAL
    restart: always
```

### Dein vollständiges `.env`

```text
PW und IP entfernt

# Dashboard application (the main part)
#
# Domain name or IP address of your host
APP_HOST=192.168.x.y
#
# SSL redirect: Use "true" (which is the default) if you want to auto-redirect to https
FORCE_SSL=false
#
# Secret token to secure cookies, 128 chars long hexadecimal encoded string (don't use this example, make some random changes!)
# Currently there are no cookies in Soletrus, but this may change in the future
SECRET_KEY_BASE=f60debe97dcb73280a2cc83668fd60e8d0e8e48269036a7bce980ee53cfb312e377989a750b8c945a5f69b041289ecb4e2d9e40641b81257c65ac2d43e3c837f
#
# Date of commissioning of your photovoltaic system
INSTALLATION_DATE=2023-06-12
#
# Price you pay for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
ELECTRICITY_PRICE=0.34
#
# Price you get for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
FEED_IN_TARIFF=0.0820
#
# Password for the PostgreSQL database, used by the app to comunicate internally with the database
POSTGRES_PASSWORD=
#
# Password to login as administrator, required to manage settings like historical prices
ADMIN_PASSWORD=

##################################################################
# Influx database settings
#
# Influx host (to access from SOLECTRUS dashboard and collectors)
INFLUX_HOST=influxdb
INFLUX_SCHEMA=http
INFLUX_PORT=8086
#
# Credentials for the Influx database
INFLUX_ORG=solectrus
INFLUX_USERNAME=admin
INFLUX_PASSWORD=ExAmPl3PA55W0rD
INFLUX_ADMIN_TOKEN=my-super-secret-admin-token
INFLUX_BUCKET=my-solectrus-bucket
#
# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token
#
# Volume path for storing the Influx data
INFLUX_VOLUME_PATH=/home/pi/solectrus/influxdb

##################################################################
# REDIS settings
#
REDIS_VOLUME_PATH=/home/pi/solectrus/redis

##################################################################
# PostgreSQL database settings
#
DB_VOLUME_PATH=/home/pi/solectrus/postgresql

##################################################################
# SENEC Collector
#
# Change this to your own SENEC IP address!
SENEC_HOST=192.168.179.7
SENEC_SCHEMA=http
SENEC_INTERVAL=5

##################################################################
# Solar forecasting with https://forecast.solar
# API docs: https://doc.forecast.solar/doku.php?id=api:estimate
#
# Latitude of the plant location
FORECAST_LATITUDE=
#
# Longitude of the plant location
FORECAST_LONGITUDE=
#
# Plane declination: 0 (horizontal) - 90 (vertical)
FORECAST_DECLINATION=43
#
# Plane azimuth: -180 ... 180 (-180 = north, -90 = east, 0 = south, 90 = west, 180 = north)
FORECAST_AZIMUTH=-5
#
# Installed modules power in kilowatt peak (kWp)
FORECAST_KWP=5.3
#
# Update interval in seconds, 900s = 15 minutes, the public (and free) API allows a minimum of 900 seconds
FORECAST_INTERVAL=3600
```

ledermann, 2024-11-15:
> ein doker Compose pull datet nicht von v0.18.0 auf o.18.1 up

Hast du vielleicht das nachfolgende `docker compose up -d` vergessen? Ein `docker compose pull`  allein holt nur das neueste Image, startet aber nichts neu.

Für ein Update ist also beides erforderlich:

```bash
docker compose pull
docker compose up -d
```

Um das zukünftig nicht immer manuell machen zu müssen, empfiehlt sich übrigens die Verwendung von Watchtower: https://docs.solectrus.de/referenz/watchtower/

KMontenarh, 2024-11-15:
> [Verfolgungsbild][https://github.com/notifications/beacon/BAJ5MVF6YV4L6WUVM2KPYFT2AX5OZA5CNFSM6AAAAABR3EUEOCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUTYFNF6.gif]

Hallo
Sorry für die Belästigung
Ich hatte gedacht ich hätte den PI rebooter da noch mehr Updates da waren.
Hatte ich nicht gemacht.
Ich sollte es doch Scripten, da ich es eh jede Woche mache.
Entschuldigung und schönes Wochenende

Gruß Karlheinz

ledermann, 2024-11-15:
Ok, alles gut :)

closed 2024-11-15 by ledermann