solectrus/solectrus #3750: Selbst definierte Verbraucher: Fehler 500 bei Tagesansicht

gereons, 2024-12-06:
### Beschreibung (auf Deutsch oder Englisch)

Mit PR #3747 bekomme ich in der Tages/Monats/Jahres/Gesamtansicht der Verbraucher einen Error 500. Das dashboard-Log sagt

```
I, [2024-12-06T19:37:42.442325 #1]  INFO -- : [172.18.0.1] method=GET path=/house/stats/house_power/2024-12-06 format=html controller=House::StatsController action=index status=500 allocations=5134 duration=13.29 view=0.00 db=3.45
E, [2024-12-06T19:37:42.444594 #1] ERROR -- : [172.18.0.1]   
[172.18.0.1] ActionView::Template::Error (undefined method `custom_01_costs_grid' for #<Calculator::Range:0x0000f7c7ff5c3900>):
[172.18.0.1] 
Causes:
[172.18.0.1] NoMethodError (undefined method `custom_01_costs_grid' for #<Calculator::Range:0x0000f7c7ff5c3900>)
[172.18.0.1]     53: 
[172.18.0.1]     54:           // Inner content
[172.18.0.1]     55:           .flex-1.flex
[172.18.0.1]     56:             = render SegmentContainer::Component.new tippy_placement: 'right',
[172.18.0.1]     57:                                                      calculator:,
[172.18.0.1]     58:                                                      timeframe: do |balance|
[172.18.0.1]     59:               ruby:
[172.18.0.1]   
[172.18.0.1] app/components/segment/component.rb:67:in `public_send'
[172.18.0.1] app/components/segment/component.rb:67:in `costs'
[172.18.0.1] app/components/segment/component.html.slim:65:in `block in call'
[172.18.0.1] app/components/segment/component.html.slim:15:in `call'
[172.18.0.1] app/components/segment_container/component.html.slim:5:in `block in call'
[172.18.0.1] app/components/segment_container/component.html.slim:4:in `each'
[172.18.0.1] app/components/segment_container/component.html.slim:4:in `call'
[172.18.0.1] app/views/house/stats/index.turbo_stream.slim:56
[172.18.0.1] app/controllers/house/stats_controller.rb:8:in `index'
[172.18.0.1] app/controllers/concerns/auto_locale.rb:14:in `switch_locale'
```

In den Einstellungen und auch in der "Aktuell"-Ansicht werden meine 3 Verbraucher problemlos angezeigt.

### Welche Version von SOLECTRUS verwendest du?

pr-3747

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Cloud

ledermann, 2024-12-06:
Danke für den Hinweis, ist bereits korrigiert, in `separate-consumers-63`. Der Fehler trat auf, wenn in den Einstellungen die Checkbox "Opportunitätskosten" deaktiviert war.

ledermann, 2024-12-06:
Link zu #3749

closed 2024-12-06 by ledermann

Zeppelin-PV, 2024-12-06:
ich habe den selben Fehler gepostet, irrtümlicherweise in Discussion #3749
dort ist auch mein Log-file und die geänderten env und compose zu sehen
trotz neuerlichem Update ist bei mir der Fehler immer noch
bei Tag, Woche etc sieht man ganz kurz die Grafik bevor der Fehler 500 zuschlägt

ledermann, 2024-12-07:
@Zeppelin-PV Das scheint ein anderes Problem zu sein, habe #3752 angelegt.

commit 31378f4e (2024-12-06): Fix costs calculation without opportunity_costs

Resolves #3750