solectrus/solectrus #3751: konfuse Anzeige "verbunden" bei Hausverbrauch in Beta pr-3747

Zeppelin-PV, 2024-12-06:
### Beschreibung (auf Deutsch oder Englisch)

neben dem Fehler 500 ist mir noch aufgefallen, dass im Hauptmenü "keine Verbindung" angezeigt wird, währen im neuen Menü 
gesplittete Verbraucher "verbunden" erscheint

[image]
[image]

### Welche Version von SOLECTRUS verwendest du?

Beta pr-3747

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Pi 4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
gepostet in #3749
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
gepostet in #3749
```

### Dein vollständiges `.env`

```text
gepostet in #3749
```

Zeppelin-PV, 2024-12-06:
keine Verbindung ist korrekt, da der SENEC unter seiner IP gerade nicht erreichbar ist, demnach ist "verbunden" nicht korrekt

ledermann, 2024-12-07:
Danke für den Hinweis, das Problem ist jetzt behoben in `custom-consumers-64`

commit e5d6666b (2024-12-07): Fix missing system_status

Resolves #3751
closed 2024-12-07 by ledermann