solectrus/solectrus #3752: TypeError bei Zeiträumen

ledermann, 2024-12-07:
### Beschreibung (auf Deutsch oder Englisch)

Beschrieben von @Zeppelin-PV in #3749:

> Meine ersten zwei Verbraucher werden angezeigt, allerdings nur im Aktuell
> Tag, Woche, etc. kommt Fehler 500

### Welche Version von SOLECTRUS verwendest du?

custom-consumers-63

### Auf welchem Gerät läuft bei dir SOLECTRUS?

Raspi 4

### Auszug aus dem Logfile, sofern zum Verständnis sinnvoll (z.B. bei einem Fehler 500)

```text
dashboard-1 | I, [2024-12-06T21:12:29.171896 #1] INFO -- : [192.168.1.110] method=GET path=/house/stats/custom_power_02/2024-12-06
format=html controller=House::StatsController action=index status=500 allocations=51866 duration=946.73 view=0.00 db=67.31
dashboard-1 | E, [2024-12-06T21:12:29.184009 #1] ERROR -- : [192.168.1.110]
dashboard-1 | [192.168.1.110] ActionView::Template::Error (nil can't be coerced into Float):
dashboard-1 | [192.168.1.110]
dashboard-1 | Causes:
dashboard-1 | [192.168.1.110] TypeError (nil can't be coerced into Float)
dashboard-1 | [192.168.1.110] 53:
dashboard-1 | [192.168.1.110] 54: // Inner content
dashboard-1 | [192.168.1.110] 55: .flex-1.flex
dashboard-1 | [192.168.1.110] 56: = render SegmentContainer::Component.new tippy_placement: 'right',
dashboard-1 | [192.168.1.110] 57: calculator:,
dashboard-1 | [192.168.1.110] 58: timeframe: do |balance|
dashboard-1 | [192.168.1.110] 59: ruby:
dashboard-1 | [192.168.1.110]
dashboard-1 | [192.168.1.110] app/services/calculator/range.rb:328:in `*'
dashboard-1 | [192.168.1.110] app/services/calculator/range.rb:328:in `block (2 levels) in <class:Range>'
dashboard-1 | [192.168.1.110] app/components/segment/component.rb:67:in `public_send'
dashboard-1 | [192.168.1.110] app/components/segment/component.rb:67:in `costs'
dashboard-1 | [192.168.1.110] app/components/segment/component.html.slim:65:in `block in call'
dashboard-1 | [192.168.1.110] app/components/segment/component.html.slim:15:in `call'
dashboard-1 | [192.168.1.110] app/components/segment_container/component.html.slim:5:in `block in call'
dashboard-1 | [192.168.1.110] app/components/segment_container/component.html.slim:4:in `each'
dashboard-1 | [192.168.1.110] app/components/segment_container/component.html.slim:4:in `call'
dashboard-1 | [192.168.1.110] app/views/house/stats/index.turbo_stream.slim:56
dashboard-1 | [192.168.1.110] app/controllers/house/stats_controller.rb:8:in `index'
dashboard-1 | [192.168.1.110] app/controllers/concerns/auto_locale.rb:14:in `switch_locale'
```

### Dein vollständiges `docker-compose.yml` oder `compose.yml`

```yaml
services:
  dashboard:
    image: ghcr.io/solectrus/solectrus:pr-3747
    depends_on:
      postgresql:
        condition: service_healthy
      influxdb:
        condition: service_healthy
      redis:
        condition: service_healthy
    links:
      - postgresql
      - influxdb
      - redis
    ports:
      - 3000:3000
    environment:
      - TZ
      - APP_HOST
      - FORCE_SSL
[... 320 lines cut ...]
### Dein vollständiges `.env`

```text
INFLUX_SENSOR_INVERTER_POWER=SENEC:inverter_power
INFLUX_SENSOR_HOUSE_POWER=SENEC:house_power
INFLUX_SENSOR_GRID_IMPORT_POWER=SENEC:grid_power_plus
INFLUX_SENSOR_GRID_EXPORT_POWER=SENEC:grid_power_minus
INFLUX_SENSOR_BATTERY_CHARGING_POWER=SENEC:bat_power_plus
INFLUX_SENSOR_BATTERY_DISCHARGING_POWER=SENEC:bat_power_minus
INFLUX_SENSOR_BATTERY_SOC=SENEC:bat_fuel_charge
INFLUX_SENSOR_WALLBOX_POWER=
INFLUX_SENSOR_CASE_TEMP=SENEC:case_temp
INFLUX_SENSOR_SYSTEM_STATUS=SENEC:current_state
INFLUX_SENSOR_SYSTEM_STATUS_OK=SENEC:current_state_ok
INFLUX_SENSOR_GRID_EXPORT_LIMIT=SENEC:power_ratio
INFLUX_SENSOR_INVERTER_POWER_FORECAST=Forecast:watt
INFLUX_SENSOR_HEATPUMP_POWER=
# Benutzer definierte Verbraucher
INFLUX_SENSOR_CUSTOM_POWER_01=AUFZUG:power
INFLUX_SENSOR_CUSTOM_POWER_02=THERME:power

# Optional: Exclude sensors from house power calculation
# Default is HEATPUMP_POWER (if defined)
# INFLUX_EXCLUDE_FROM_HOUSE_POWER=HEATPUMP_POWER,WALLBOX_POWER
```

ledermann, 2024-12-07:
@Zeppelin-PV Der Fehler ist möglicherweise behoben - ganz sicher bin ich mir nicht, da ich das bei mir nicht klar reproduzieren konnte. Bitte mal ein Update durchführen - auf `custom-consumers-68`.

closed 2024-12-07 by ledermann

Zeppelin-PV, 2024-12-07:
hallo Georg, natürlich habe ich zum ersten Kaffee am Morgen sofort ein Update geholt
[image]
sieht gut aus, alles erscheint perfekt - kein Fehler 500 bei Tag, Woche, Monat, Jahr, Gesamt
Super - DANKE !!!

commit 94e78754 (2024-12-07): Fix TypeError in house_costs calculation

Closes #3752